ShadowBlip/HandyGCCS #2: Add OneXPlayer support

pastaq, 2022-06-10:
Add support for OneXPlayer devices.
Known devices:
oxp-1s
oxp-amd
oxp-mini
oxp-mini-amd

Due to bad EDID/DMI data, we will need to identify a simple way to isloate a specific device.
[OneX Player Mini input capture](https://github.com/ShadowBlip/HandyGCCS/files/8881739/capture_file.txt)

ruineka, 2022-06-12:
OneXPlayer 1S 11957 Input Codes.
[Keymaps.txt](https://github.com/ShadowBlip/HandyGCCS/files/8886098/Keymaps.txt)

pastaq, 2022-06-12:
Looks like Show desktop, screenshot, and task manager all send key 125 first. That is incompatible with founders through 2021 models on the Neo as they use key 125 alone for "win". Will need to check model in the line for button1
psudocode: 
```
if ((active == [125] and system_type == "AYA_GEN1") or (active == [125, 99] and system_type == "OXP")) and button_on == 1 and button1 not in event_queue:

```
possible replacement for https://github.com/ShadowBlip/HandyGCCS/blob/2a17b6dcddfe63fc0298099ec67ec366cbf5bf8b/handycon.py#L181

Will also need to add to the section in __init__() where system_type is identified.

pastaq, 2022-06-20:
Marked as complete. If additional models need support, submit new issue.

closed 2022-06-20 by pastaq