ShadowBlip/HandyGCCS #3: Add gyro support

pastaq, 2022-06-10:
Utilize optional gyro support from BMI160 chips in OXP/AN devices.

pastaq, 2022-06-10:
Track progress in https://github.com/ShadowBlip/HandyGCCS/tree/gyro-control

Samsagax, 2022-07-22:
For iio sensors:
On the OXP Mini the accelerometer is rotated.
Using  https://gitlab.freedesktop.org/hadess/iio-sensor-proxy/#accelerometer-orientation

modalias: `acpi:BMI0160:BMI0160:`

DMI string: `dmi:bvnAmericanMegatrendsInternational,LLC.:bvrV1.08_P4C8M43L6:bd04/28/2022:br5.19:efr1.35:svnONE-NETBOOKTECHNOLOGYCO.,LTD.:pnONEXPLAYER:pvrDefaultstring:rvnONE-NETBOOKTECHNOLOGYCO.,LTD.:rnONEXPLAYER:rvrV01:cvnONE-NETBOOKTECHNOLOGYCO.,LTD.:ct30:cvrDefaultstring:skuONEXPLAYER:`

Add this content to file `/etc/udev/hwdb.d/61-sensor-local.hwdb`: 
````
##################
# One X Player Mini
##################
sensor:modalias:acpi:BMI0160:*:dmi:*:rnONEXPLAYER:rvrV01:*
 ACCEL_MOUNT_MATRIX=-1, 0, 0; 0, 1, 0; 0, 0, -1
````

Extra: 
pick it up with `iio-sensor-proxy` for autorotation of the screen in GNOME/KDE desktops.

pastaq, 2022-07-25:
The pypi package BMI160_i2c is a required dependency that doesn't have an AUR package. We should utilize pypi2pkgbuild to convert these dependencies to installable packages. It can build from git source or pypi release.

https://github.com/lefuturiste/BMI160-i2c
https://github.com/anntzer/pypi2pkgbuild

pastaq, 2022-08-19:
python-bmi160-i2c is now available in the aur and is a dependency in the hangygccs-git PKGBUILD.

closed 2022-08-19 by pastaq