ShadowBlip/HandyGCCS #12: Values to edit in handycon.py for Aya Neo Founder / 2021 for personal use in vanilla Linux

ghost573, 2022-07-12:
I know this is designed with HoloISO in mind however I would like to make a personal edit so I can map the extra buttons on my founders unit to the equivalent value such as to open the start menu etc but I don't know what values to change, I'm not a programmer just an ordinary user and want some help.

pastaq, 2022-07-12:
Sure. You will need to find the keycodes you want to emulate, then edit the file in /usr/local/bin/handycon.py (you will need to do it as sudo)

Here is the master list of event codes:
https://github.com/torvalds/linux/blob/master/include/uapi/linux/input-event-codes.h

The location in the file you can edit is here:
https://github.com/ShadowBlip/HandyGCCS/blob/2efa0c28246edb2219a8420cf18fb71f05f9176d/handycon.py#L24

Ad you own event in the same format, I.E. EVENT_DESKTOP = [[e.EV_KEY, e.KEY_LEFTMETA], [e.EV_KEY, e.KEY_D]]

Then find your device section:
https://github.com/ShadowBlip/HandyGCCS/blob/2efa0c28246edb2219a8420cf18fb71f05f9176d/handycon.py#L209
Aya Gen 1 for Founders.
Look for the button you want mapped to that, I.E "WIN", take note of the button# that it presses (Button1) and scroll back up to the button map:
https://github.com/ShadowBlip/HandyGCCS/blob/2efa0c28246edb2219a8420cf18fb71f05f9176d/handycon.py#L47

Replace the button map even with your custom event.
I.E.
```
button_map = {
        "button1": EVENT_DESKTOP,
        "button2": EVENT_QAM,
        "button3": EVENT_ESC,
        "button4": EVENT_OSK,
        "button5": EVENT_HOME,
        }
```

ghost573, 2022-07-13:
Thanks so much.

closed 2022-07-13 by ghost573