ShadowBlip/HandyGCCS #16: HandyGCCS closes duckstation and potentially other qt programs when volume button is pressed

ghost573, 2022-07-27:
I noticed strange behavior as when I try to change volume duckstation quits without warning, I tested this and it happens with almost all of the extra buttons except the standard xbox controls but only when the handycon service is active. When disabled duckstation remains open when changing voltume or opening the keyboard etc.

pastaq, 2022-07-28:
That is quite strange. Can you provide me more details about your current setup? OS, device, method if installing duckstation & version?

ghost573, 2022-07-28:
Sure thing.

Aya Neo FE
Arch on kernel 5.18
Either [appimage](https://github.com/stenzek/duckstation) or from aur as duckstation-git, it affects both.

I can try to get logs as well if you need them, pretty sure that's just the terminal output but if there is another method please tell me.

pastaq, 2022-07-28:
can you install the evtest package and test the following devices by pressing the VOL UP then VOL DOWN once each
With handycon service not running, AT Translated Set 2 keyboard (usually event4)
With handycon service running, Handheld Controller
Paste the output in a response message here please.

ghost573, 2022-07-28:
Sure thing.

Here's with handycon service disabled.

```
Event: time 1659004003.030062, type 4 (EV_MSC), code 4 (MSC_SCAN), value ae
Event: time 1659004003.030062, type 1 (EV_KEY), code 114 (KEY_VOLUMEDOWN), value 1
Event: time 1659004003.030062, -------------- SYN_REPORT ------------
Event: time 1659004003.133713, type 4 (EV_MSC), code 4 (MSC_SCAN), value ae
Event: time 1659004003.133713, type 1 (EV_KEY), code 114 (KEY_VOLUMEDOWN), value 0
Event: time 1659004003.133713, -------------- SYN_REPORT ------------
Event: time 1659004004.748340, type 4 (EV_MSC), code 4 (MSC_SCAN), value b0
Event: time 1659004004.748340, type 1 (EV_KEY), code 115 (KEY_VOLUMEUP), value 1
Event: time 1659004004.748340, -------------- SYN_REPORT ------------
Event: time 1659004004.841923, type 4 (EV_MSC), code 4 (MSC_SCAN), value b0
Event: time 1659004004.841923, type 1 (EV_KEY), code 115 (KEY_VOLUMEUP), value 0
Event: time 1659004004.841923, -------------- SYN_REPORT ------------
```

Here's with handycon service enabled

```
Event: time 1659004186.137900, type 1 (EV_KEY), code 115 (KEY_VOLUMEUP), value 1
Event: time 1659004186.137900, -------------- SYN_REPORT ------------
Event: time 1659004186.241121, type 1 (EV_KEY), code 115 (KEY_VOLUMEUP), value 0
Event: time 1659004186.241121, -------------- SYN_REPORT ------------
Event: time 1659004189.383017, type 1 (EV_KEY), code 114 (KEY_VOLUMEDOWN), value 1
Event: time 1659004189.383017, -------------- SYN_REPORT ------------
Event: time 1659004189.476250, type 1 (EV_KEY), code 114 (KEY_VOLUMEDOWN), value 0
Event: time 1659004189.476250, -------------- SYN_REPORT ------------
```

I did mention it but this happens with the volume buttons and I think all the extra buttons.

pastaq, 2022-08-01:
Does duckstation use pygame at all to capture key events?

ghost573, 2022-08-01:
I'm not sure.

I have similar issues with the qt build of pcxs2 so I assumed it's a qt thing.

pastaq, 2022-08-01:
I installed the flatpak duckstation but wasn't able to reproduce this issue. Does this happen in-game only or in the system menus?

ghost573, 2022-08-01:
Both and I used the appimage and the build from aur not flatpak.

pastaq, 2022-08-01:
Can you post some crash logs for each project? This sounds similar to an issue we discovered with Chimera.

pastaq, 2022-08-01:
that only seems to contain the first few lines of the kernel init. Try `journalctl -u duckstation`

ghost573, 2022-08-02:
Sorry about that, first time trying this.
[journalfix.txt](https://github.com/ShadowBlip/HandyGCCS/files/9239436/journalfix.txt)

pastaq, 2022-08-18:
I just pushed a major rewrite that fixes HandyGCCS crashes. I've not yet been able to reproduce this issue. Do you mind testing the current release to see if this issue remains?

ghost573, 2022-08-19:
Will do. I don't have my device at the moment, it's sent off for a repair, I'll definitely test once I get it back.

ghost573, 2022-10-06:
Finally got to test it, problem is fixed, thanks.

closed 2022-10-06 by ghost573