ShadowBlip/HandyGCCS #19: Error during installation on Aya Neo Next with Holoiso

thalastwon, 2022-08-05:
Using Holoiso, on an Aya Neo Next.

after typing sudo make install in konsole, getting this error.

'HandyGCCS-Controller.cfg' -> '/usr/share/libretro/autoconfig/udev/'
cp: cannot create regular file  '/usr/share/libretro/autoconfig/udev/': No such file or directory
make: *** [makefile:3: install] Error 1

Note: The only other things I've installed are Crankshaft and some related plugins as well as a couple games to try out. 
Let me know if more information is needed.

pastaq, 2022-08-05:
Thanks. Started working on this when I saw your Reddit response. Can you try two things?

1.) Install from AUR using yay
2.) If that works, remove the package and try the installer from the commit I just submitted.

thalastwon, 2022-08-05:
I'm still relatively new to LInux and finding out what AUR is. It might take me a bit to google it.

thalastwon, 2022-08-05:
Got it installed through AUR and seems to be working. Removing and trying through the commit.

thalastwon, 2022-08-05:
It worked through the manual process as well! Thanks!!

pastaq, 2022-08-05:
Cool. I'll mark as closed. I forgot to make sure a directory existed in the install script. Thanks for reporting and testing.

closed 2022-08-05 by pastaq