ShadowBlip/HandyGCCS #20: Ensure legacy support for OXP devices

pastaq, 2022-08-05:
Work with OXP to fix their controller and DMI firmware has resulted in a potential bifurcation of support for identical models with different BIOS versioning. We need to create a knowledge base of new and old BIOS parameters to ensure we can maintain support for updated and non-updated BIOS devices.

pastaq, 2022-08-05:
Original BIOS. AMD Mini 5800u
```
[gamer@chimeraos ~]$ cat /sys/class/dmi/id/bios_date
04/28/2022
[gamer@chimeraos ~]$ cat /sys/class/dmi/id/bios_version
V1.08_P4C8M43L6
[gamer@chimeraos ~]$ cat /sys/class/dmi/id/bios_release
5.19
[gamer@chimeraos ~]$ cat /sys/class/dmi/id/bios_vendor
American Megatrends International, LLC.
[gamer@chimeraos ~]$ cat /sys/class/dmi/id/product_name
ONE XPLAYER
[gamer@chimeraos ~]$ cat /sys/class/dmi/id/product_sku
ONE XPLAYER
[gamer@chimeraos ~]$ cat /sys/class/dmi/id/product_version
Default string
[gamer@chimeraos ~]$ cat /sys/class/dmi/id/sys_vendor
ONE-NETBOOK TECHNOLOGY CO., LTD.
[gamer@chimeraos ~]$ cat /sys/class/dmi/id/ec_firmware_release
1.35
```

Black screen fix BIOS. AMD Mini 5800u
```
[gamer@chimeraos ~]$ cat /sys/class/dmi/id/bios_date
04/28/2022
[gamer@chimeraos ~]$ cat /sys/class/dmi/id/bios_version 
V1.08_P4C8M43L6
[gamer@chimeraos ~]$ cat /sys/class/dmi/id/bios_release 
5.19
[gamer@chimeraos ~]$ cat /sys/class/dmi/id/bios_vendor 
American Megatrends International, LLC.
[gamer@chimeraos ~]$ cat /sys/class/dmi/id/product_name 
ONE XPLAYER
[gamer@chimeraos ~]$ cat /sys/class/dmi/id/product_sku
ONE XPLAYER
[gamer@chimeraos ~]$ cat /sys/class/dmi/id/product_version 
Default string
[gamer@chimeraos ~]$ cat /sys/class/dmi/id/sys_vendor 
ONE-NETBOOK TECHNOLOGY CO., LTD.
[gamer@chimeraos ~]$ cat /sys/class/dmi/id/ec_firmware_release 
1.35
```

pastaq, 2022-08-05:
DMI Table:

```
Device series | dmi manufactures name | dmi board name
AOKZOE | AOKZOE | AOKZOE A1 AR07
OneXPlayer | One-NetBook | OneXPlayer 1 T08
OneXPlayer | One-NetBook | OneXPlayer 1S T08
OneXPlayer | One-NetBook | OneXPlayer 1S A08
OneXPlayer | One-NetBook | OneXPlayer Mini A07
OneXPlayer | One-NetBook | OneXPlayer Mini GT72
OneXPlayer | One-NetBook | OneXPlayer Mini GA72
OneXPlayer | One-NetBook | OneXPlayer GUNDAM GA72
```

ruineka, 2022-08-06:
OneXPlayer 1S i7-1195G7 8.4" Full Size

Bios Date
09/01/2021
Bios Version
1AWHY9.19
Bios Release
5.19
Bios Vendor
American Megatrends International, LLC.
Product Name
ONE XPLAYER
Product SKU
Default string
Product Version
Default string
System Vendor
ONE-NETBOOK TECHNOLOGY CO., LTD.
Firmware Release
0.81

ruineka, 2022-08-06:
[dmi-cat-scan.zip](https://github.com/ShadowBlip/HandyGCCS/files/9275613/dmi-cat-scan.zip)

We can run this script on any device to easily grab this information and have it stored at ~/dmi.log.

ruineka, 2022-08-06:
Intel Gundam Mini 1200p 1TB 7"

Bios Date
04/19/2022
Bios Version
1AXFE3.43
Bios Release
5.25
Bios Vendor
American Megatrends International, LLC.
Product Name
ONE XPLAYER
Product SKU
Default string
Product Version
Default string
System Vendor
ONE-NETBOOK TECHNOLOGY CO., LTD.
Firmware Release
0.119

pastaq, 2022-08-18:
Based on initial testing this doesn't appear to be an actual issue.

closed 2022-08-18 by pastaq