ShadowBlip/HandyGCCS #21: Crash when system loses or transfers control of input devices.

pastaq, 2022-08-14:
HandyGCCS will crash if another program or system event takes control of the hardware controller or keyboard.

Known causes:
- Switching to mouse mode on OXP devices.
- Initializing a joystick in pygame.
- Resuming from an extended sleep.

We need a way to detect a missing device, await the return of that device, and to restart the event loop for that device.

AlishaHawkward, 2022-08-15:
I found this issue too. Ayaneo Air

AlishaHawkward, 2022-08-15:
```
Aug 16 00:27:16 Zero python3[453]: Task exception was never retrieved
Aug 16 00:27:16 Zero python3[453]: future: <Task finished name='Task-2' coro=<capture_keyboard_e>
Aug 16 00:27:16 Zero python3[453]: Traceback (most recent call last):
Aug 16 00:27:16 Zero python3[453]:   File "/usr/local/bin/handycon.py", line 509, in capture_key>
Aug 16 00:27:16 Zero python3[453]:     await do_rumble(0, 150, 1000, 0)
Aug 16 00:27:16 Zero python3[453]:   File "/usr/local/bin/handycon.py", line 415, in do_rumble
Aug 16 00:27:16 Zero python3[453]:     effect_id = controller_device.upload_effect(effect)
Aug 16 00:27:16 Zero python3[453]:   File "/usr/lib/python3.10/site-packages/evdev/device.py", l>
Aug 16 00:27:16 Zero python3[453]:     ff_id = _input.upload_effect(self.fd, data)
Aug 16 00:27:16 Zero python3[453]: OSError: [Errno 19] No such device
```

pastaq, 2022-08-18:
Solved with #25

closed 2022-08-18 by pastaq