ShadowBlip/HandyGCCS #26: Aya neo air pro (5825) gyroscope support?

chnlzh2, 2022-08-26:
Aya neo air pro (5825) gyroscope support? 
THANK you

pastaq, 2022-08-27:
Can you elaborate? Gyro support should work already. You need to press and release LC and RC at the same time. The console will provide 1 buzz for enable and 2 buzz for disable.

alphonsez1, 2022-09-05:
I have the same issue. Also the rumble/haptic is not working either. I find that the Steam UI still shows the Xbox controller logo when the service is running.

I tried to run the python script manually. It runs fine, with the original devices moved to the hidden folder.
The log says the Gryo is successfully captured. It seems to be the issue with Steam that didn't recognize the Handheld controller.

When I got it running, there is always a popup says that Xbox 360 controller is connected. If I look at the js devices, there is always a new js device created when I run this application. So my guess is that something else is trying to recover the device.

alphonsez1, 2022-09-06:
I did some further testing. First of all, I got wrong about what LC and RC are. I thought it was clicking both joysticks.
@chnlzh2 Would you try pressing both of the (...) buttons and check whether it works to you?

[image]

It still doesn't work for me, although I do hear the 1 buzz when enabling gyro and 2 buzz for disable. I printed the gyro events and the handycon.py is working as expected. It can also rumble properly. However it appears the games are not using this controller somehow.

alphonsez1, 2022-09-06:
I got the gyro worked once when the UI says detected steam controller rather than xbox 360 controller. However I can't reproduce it. 
I'll also try `SDL_JOYSTICK_DEVICE` env var this weekend.

alphonsez1, 2022-09-07:
Update: Somehow my controller gyro suddenly starts to work. It still shows the Xbox button. 

Now I can't figure out how to make it not work any more. Even if I use a controller layout that says not supporting motion control, it still works on my Ayaneo Air.

EricFL1998, 2022-09-26:
my gyro on the air is not working either it shows the xbox button and detect the controller as xbox-360, pressing lc, rc and release does not do anything, no buzz either.

pastaq, 2022-11-03:
What OS are you both using?

ghost573, 2022-11-04:
> What OS are you both using?

Arch linux with kernel 6.0.6-arch1-1.
Gyro doesn't seem to work for me either, not a big deal right now so just contributing.

EricFL1998, 2022-11-04:
holoiso> What OS are you both using?

pastaq, 2022-11-22:
I found the issue. The AUR package wasn't properly installing the gyro config. Fixed in the latest. package.

ghost573, 2022-11-23:
Sorry if this sounds dumb but is there a trick to activating gyro?
Been trying to get it to work with Little Witch Nobeta and no luck.

pastaq, 2022-11-23:
should just be holding the LC & RC buttons until the device does a little buzz, one for on and two for off.

ghost573, 2022-11-23:
Strange, it doesn't seem working for me.
Is there a problem I can run to help debug this?

pastaq, 2022-11-23:
If you disable the handycon.service and run it manually you should get some debug output. running the `evtest` program should tell you if the gyro is sending events or not.

ghost573, 2022-11-23:
Here's my results.

```
Event: time 1669187582.208484, type 1 (EV_KEY), code 316 (BTN_MODE), value 1
Event: time 1669187582.208484, -------------- SYN_REPORT ------------
Event: time 1669187582.298663, type 1 (EV_KEY), code 311 (BTN_TR), value 1
Event: time 1669187582.298663, -------------- SYN_REPORT ------------
Event: time 1669187582.388847, type 1 (EV_KEY), code 316 (BTN_MODE), value 0
Event: time 1669187582.388847, -------------- SYN_REPORT ------------
Event: time 1669187582.479018, type 1 (EV_KEY), code 311 (BTN_TR), value 0
Event: time 1669187582.479018, -------------- SYN_REPORT ------------
```

pastaq, 2022-11-23:
> Here's my results.
>
> ```
> Event: time 1669187582.208484, type 1 (EV_KEY), code 316 (BTN_MODE), value 1
> Event: time 1669187582.208484, -------------- SYN_REPORT ------------
> Event: time [...]
This would seem to indicate only the LC button is detecting a press. What output do you get when you press just RC?

If nothing happens with RC only and only LC shows up when you press both, there could be an issue with the key map on some of these. Stop handycon.service with systemctl and run `sudo evtest` to capture the raw data for the RC button under the AT translated keyboard device.  If you still get nothing then you might have a hardware issue.

ghost573, 2022-11-23:
```
Event: time 1669190910.441148, type 1 (EV_KEY), code 316 (BTN_MODE), value 1
Event: time 1669190910.441148, -------------- SYN_REPORT ------------
Event: time 1669190910.531484, type 1 (EV_KEY), code 307 (BTN_NORTH), value 1
Event: time 1669190910.531484, -------------- SYN_REPORT ------------
Event: time 1669190910.622459, type 1 (EV_KEY), code 316 (BTN_MODE), value 0
Event: time 1669190910.622459, -------------- SYN_REPORT ------------
Event: time 1669190910.713144, type 1 (EV_KEY), code 307 (BTN_NORTH), value 0
Event: time 1669190910.713144, -------------- SYN_REPORT ------------
Event: time 1669190916.049937, type 1 (EV_KEY), code 316 (BTN_MODE), value 1
Event: time 1669190916.049937, -------------- SYN_REPORT ------------
Event: time 1669190916.140666, type 1 (EV_KEY), code 307 (BTN_NORTH), value 1
Event: time 1669190916.140666, -------------- SYN_REPORT ------------
Event: time 1669190916.231511, type 1 (EV_KEY), code 316 (BTN_MODE), value 0
Event: time 1669190916.231511, -------------- SYN_REPORT ------------
Event: time 1669190916.321958, type 1 (EV_KEY), code 307 (BTN_NORTH), value 0
Event: time 1669190916.321958, -------------- SYN_REPORT ------------
Event: time 1669190925.063344, type 1 (EV_KEY), code 316 (BTN_MODE), value 1
Event: time 1669190925.063344, -------------- SYN_REPORT ------------
Event: time 1669190925.153710, type 1 (EV_KEY), code 311 (BTN_TR), value 1
Event: time 1669190925.153710, -------------- SYN_REPORT ------------
Event: time 1669190925.244272, type 1 (EV_KEY), code 316 (BTN_MODE), value 0
Event: time 1669190925.244272, -------------- SYN_REPORT ------------
Event: time 1669190925.334615, type 1 (EV_KEY), code 311 (BTN_TR), value 0
Event: time 1669190925.334615, -------------- SYN_REPORT ------------

```

I hit LC first, then RC, then both together.
I know the button works because I mapped it before.

pastaq, 2022-11-23:
Strange, it appears to be detecting the same butotn for LC and RC. You remapped in windows with the AyaSpace app? What did you map it to?

I'll need to see what buttons the system is detecting from the raw input. Can you get an evtest capture of the AT Translated Keyboard with handycon.service sopped?

ghost573, 2022-11-24:
> Strange, it appears to be detecting the same butotn for LC and RC. You remapped in windows with the AyaSpace app? What did you map it to?

Nah, I remapped it by modifying constants.py.

> I'll need to see what buttons the system is detecting from the raw input. Can you get an evtest capture of the AT Translated Keyboard with handycon.service sopped?

Sure thing.

pastaq, 2022-11-24:
> Nah, I remapped it by modifying constants.py.

Uh, don't do that. There should be a file, $HOME/.config/handygccs.conf where you can change the assignments of each button. If you mess with the expected constants then its going to behave differently.

pastaq, 2022-11-24:
I also haven't tested the results of having two buttons mapped to the same thing, but I'm pretty sure that will cause issues.

ghost573, 2022-11-24:
> Uh, don't do that. There should be a file, $HOME/.config/handygccs.conf where you can change the assignments of each button. If you mess with the expected constants then its going to behave differently.

Already reinstalled and restored it back to how it was before testing.

pastaq, 2022-11-24:
Can you provide the console output from manually starting the service?

ghost573, 2022-11-24:
> Can you provide the console output from manually starting handycon.py

Sure thing.

pastaq, 2022-11-24:
Run it as sudo

ghost573, 2022-11-24:
Sure thing. Sorry about that.

pastaq, 2022-11-24:
Hmm, I'll need to research that error some. 

Can you run 

`sudo i2cdetect -l` and `sudo i2cdetect -y -r 1`

ghost573, 2022-11-24:
`sudo i2cdetect -l`
```
i2c-0   i2c             Synopsys DesignWare I2C adapter         I2C adapter
i2c-1   i2c             Synopsys DesignWare I2C adapter         I2C adapter
i2c-2   smbus           SMBus PIIX4 adapter port 0 at 0b00      SMBus adapter
i2c-3   smbus           SMBus PIIX4 adapter port 2 at 0b00      SMBus adapter
i2c-4   smbus           SMBus PIIX4 adapter port 1 at 0b20      SMBus adapter
i2c-5   i2c             AMDGPU DM i2c hw bus 0                  I2C adapter
i2c-6   i2c             AMDGPU DM i2c hw bus 1                  I2C adapter
i2c-7   i2c             AMDGPU DM i2c hw bus 2                  I2C adapter
i2c-8   i2c             AMDGPU DM aux hw bus 0                  I2C adapter
i2c-9   i2c             AMDGPU DM aux hw bus 1                  I2C adapter
i2c-10  i2c             AMDGPU DM aux hw bus 2                  I2C adapter
```
`sudo i2cdetect -y -r 1`
```
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:                         -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- --  
```

pastaq, 2022-11-24:
Hmmm, it's not on the same bus as normal. In that last command replace the 1 with 0.

ghost573, 2022-11-24:
`sudo i2cdetect -y -r 0`
```
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:                         -- -- -- -- -- -- -- -- 
10: -- -- -- -- UU -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- -- 
```

pastaq, 2022-11-24:
try this
` lsmod | grep i2c`

ghost573, 2022-11-24:
`lsmod | grep i2c`
```
bmi160_i2c             16384  0
bmi160_core            20480  2 bmi160_i2c,bmi160_spi
i2c_piix4              36864  0
i2c_dev                24576  0
```

pastaq, 2022-11-24:
Drivers appear to be loading but for some reason its not available at bus 1 address 68. I wonder if they connected it differently?

Can you post the output of that command for the busses 2-7 please? 

For reference, this is what my Non pro AIR shows for the two you provided already.

```
[gamer@chimeraos ~]$ sudo i2cdetect -y -r 0
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:                         -- -- -- -- -- -- -- -- 
10: -- -- -- -- UU -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- --                         
[gamer@chimeraos ~]$ sudo i2cdetect -y -r 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:                         -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- UU -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- -- 
```

ghost573, 2022-11-24:
`sudo i2cdetect -y -r 2`

```
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:                         -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- 2b 2c -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- 44 -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- --                         
```
`sudo i2cdetect -y -r 3`
```
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:                         -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- --                         
```
`sudo i2cdetect -y -r 4`
```
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:                         -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- 41 -- -- -- 45 -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- 53 -- -- -- 57 -- -- 5a -- -- -- 5e -- 
60: -- -- -- -- 64 65 -- -- 68 -- -- 6b 6c -- 6e 6f 
70: 70 -- 72 73 74 -- 76 77      
```                   
`sudo i2cdetect -y -r 5`
```
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:                         -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- --   
```                      
`sudo i2cdetect -y -r 6`
```
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:                         -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- --     
```                    
`sudo i2cdetect -y -r 7`
```
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:                         -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- --   
```

pastaq, 2022-11-24:
You might want to test in windows if your device is fully working. They have use 1x68 for all their other devices, including the non pro version of the air. I'll leave this open for now in the hopes we can get another pro user to test it.

ghost573, 2022-11-24:
Guess I have to pull it apart again because I don't want to touch my Linux installation.

pastaq, 2022-11-28:
Updated the name for this issue. It seems to affect the following devices:
- Aya Neo AIR Pro
- AOKZOE A1
- OXP Mini Pro

I also briefly had this issue on my Aya Neo 2021 Pro but a reboot resolved it.

I looked at the DSDT for the Mini Pro and the AOK ZOE and they both have the address 0x68 as expected configured in the firmware.

@ghost573 can you upload your DSDT as well? I'd like to confirm the same for the AIR Pro.

Instructions:
```
sudo pacman -Sy acpica
cd ~/
sudo  cat /sys/firmware/acpi/tables/DSDT > dsdt.dat
iasl -d dsdt.dat
```
Then upload dsdt.dsl here
If I can confirm the i2c address then it will have to be an issue with the driver in some way. I'll put together a gyro test script to see if we can't get it to initiate in a more simple environment.

ghost573, 2022-11-28:
@pastaq no problem. 

[dsdt.dsl.txt](https://github.com/ShadowBlip/HandyGCCS/files/10100851/dsdt.dsl.txt)

pastaq, 2022-11-28:
> @pastaq no problem.
> 
> [dsdt.dsl.txt](https://github.com/ShadowBlip/HandyGCCS/files/10100851/dsdt.dsl.txt)

Awesome, thanks. It appears to be the same as the other devices I've seen. I'll whip up a test script in the next few days.

pastaq, 2022-11-29:
journalctl from AOEZOE A1 revealed this interesting error:
```
Nov 29 12:28:00 chimeraos kernel: bmi160_i2c i2c-BMI0160:00: supply vdd not found, using dummy regulator
Nov 29 12:28:00 chimeraos kernel: bmi160_i2c i2c-BMI0160:00: supply vddio not found, using dummy regulator
Nov 29 12:28:00 chimeraos kernel: bmi160_i2c i2c-BMI0160:00: mounting matrix not found: using identity...
Nov 29 12:28:00 chimeraos kernel: bmi160_i2c i2c-BMI0160:00: Error reading chip id
Nov 29 12:28:00 chimeraos kernel: bmi160_i2c: probe of i2c-BMI0160:00 failed with error -121
```

ghost573, 2022-11-29:
Want me to run it on my Air Pro to compare?

pastaq, 2022-12-22:
An AOKZOE user noted that in windows the BMI160 returns a device ID of 27 vice 0xD1 as specified in the kernel driver. If anyone else tracking this issue is able to probe the device ID and post the value that would be great. I'd like to confirm if this is the same issue across all devices.

The GPD WIN4 uses a BMI260 which currently is not supported by the kernel driver.

Both of these issues will require a kernel patch.

ghost573, 2022-12-22:
Not sure how to do this in windows but I'll be happy to open my Air Pro to reinstall the stock windows SSD again tomorrow.

pastaq, 2022-12-22:
> Not sure how to do this in windows but I'll be happy to open my Air Pro to reinstall the stock windows SSD again tomorrow.

Sorry, I misspoke. Seems the user gets the ID in dmesg on Linux, not in windows.

Nik123456789010, 2022-12-23:
Hi! I came from HoloISO to Chimera OS, at HoloISO Gyro works (with stutter), i have a buzz by activating. Now at Chimera there where no buzz, may i can help you with with some logs from my Aya Neo Air? A question, how get HandyGCCS updates (in Chimera OS) Thank you!

pastaq, 2022-12-23:
> Hi! I came from HoloISO to Chimera OS, at HoloISO Gyro works (with stutter), i have a buzz by activating. Now at Chimera there where no buzz, may i can help you with with some logs from my Aya Neo Air? A question, how get HandyGCCS updates (in Chimera OS) Thank you!

Sure. I'll need the output of the following commands:
`lsmod | grep i2c`
`journalctl -u handycon`
`journalctl -b | grep i2c`
`sudo i2cdetect -y -r 1`

Nik123456789010, 2022-12-23:
[gamer@chimeraos ~]$ lsmod | grep i2c
bmi160_i2c             16384  0
bmi160_core            20480  2 bmi160_i2c,bmi160_spi
i2c_piix4              36864  0

[gamer@chimeraos ~]$ journalctl -u handycon
Dec 23 21:46:37 chimeraos systemd[1]: Started Handheld Game Console Controller >

[gamer@chimeraos ~]$ journalctl -b | grep i2c
Dec 23 21:46:36 chimeraos kernel: Goodix-TS i2c-GDIX1002:00: supply AVDD28 not found, using dummy regulator
Dec 23 21:46:36 chimeraos kernel: Goodix-TS i2c-GDIX1002:00: supply VDDIO not found, using dummy regulator
Dec 23 21:46:36 chimeraos kernel: Goodix-TS i2c-GDIX1002:00: Unexpected ACPI resources: gpio_count 1, gpio_int_idx 0
Dec 23 21:46:36 chimeraos kernel: Goodix-TS i2c-GDIX1002:00: ID 927, version: 0040
Dec 23 21:46:36 chimeraos kernel: Goodix-TS i2c-GDIX1002:00: Invalid config (0, 0, 0), using defaults
Dec 23 21:46:36 chimeraos kernel: input: Goodix Capacitive TouchScreen as /devices/platform/AMDI0010:00/i2c-0/i2c-GDIX1002:00/input/input5
Dec 23 21:46:36 chimeraos kernel: bmi160_i2c i2c-BMI0160:00: supply vdd not found, using dummy regulator
Dec 23 21:46:36 chimeraos kernel: bmi160_i2c i2c-BMI0160:00: supply vddio not found, using dummy regulator
Dec 23 21:46:36 chimeraos kernel: bmi160_i2c i2c-BMI0160:00: mounting matrix not found: using identity...
Dec 23 21:46:37 chimeraos iio-sensor-prox[512]: Could not find trigger name associated with /sys/devices/platform/AMDI0010:03/i2c-1/i2c-BMI0160:00/iio:device0
Dec 23 21:52:44 chimeraos dbus-daemon[453]: [system] Activating via systemd: service name='org.freedesktop.home1' unit='dbus-org.freedesktop.home1.service' requested by ':1.92' (uid=0 pid=9995 comm="sudo i2cdetect -y -r 1")
Dec 23 21:52:44 chimeraos sudo[9995]:    gamer : TTY=pts/0 ; PWD=/home/gamer ; USER=root ; COMMAND=/usr/bin/i2cdetect -y -r 1
Dec 23 21:56:50 chimeraos dbus-daemon[453]: [system] Activating via systemd: service name='org.freedesktop.home1' unit='dbus-org.freedesktop.home1.service' requested by ':1.94' (uid=0 pid=10066 comm="sudo i2cdetect -y -r 1")
Dec 23 21:56:55 chimeraos sudo[10066]:    gamer : TTY=pts/0 ; PWD=/home/gamer ; USER=root ; COMMAND=/usr/bin/i2cdetect -y -r 1

[gamer@chimeraos ~]$ sudo i2cdetect -y -r 1
[sudo] password for gamer: 
Error: Could not open file `/dev/i2c-1' or `/dev/i2c/1': No such file or directory

pastaq, 2022-12-23:
@Nik123456789010 seems the i2c_dev driver isn't loading

`sudo vim /etc/modules-load.d/handycon.conf`
add a single line to that file `i2c_dev`
Reboot

pastaq, 2022-12-24:
I think these devices may have a BMI260. The WIN4 has a BMI260. I found a driver in ChromiumOS we might be able to modify. It correlates to the device ID in the AOKZOE.

https://chromium.googlesource.com/chromiumos/platform/ec/+/master/driver/accelgyro_bmi260.h

https://chromium.googlesource.com/chromiumos/platform/ec/+/b123ecf67cb1f5becc10b9b829dd5dcf66e5edd5/driver/accelgyro_bmi260.c

https://chromium.googlesource.com/chromiumos/platform/ec/+/master/driver/accelgyro_bmi_common.h

https://chromium.googlesource.com/chromiumos/platform/ec/+/master/driver/accelgyro_bmi_common.c

Nik123456789010, 2022-12-24:
> @Nik123456789010 seems the i2c_dev driver isn't loading
> 
> `sudo vim /etc/modules-load.d/handycon.conf` add a single line to that file `i2c_dev` Reboot

Yes, and there was the buzz, thank you! I testet it at Aperture Desk Job, there was also gyro stutters.

pastaq, 2022-12-25:
> there was also gyro stutters.

Yes, I need to refractor it a bit. It's on the Todo list.

jcfrosty, 2022-12-31:
I finally got the system to tell me the chip ID in journalctl! it doesn't always work and its very random, but it finally tells you the chip_ID is 27 and expecting D1 !

```
Dec 31 00:32:32 chimeraos kernel: bmi160_i2c i2c-BMI0160:00: supply vdd not found, using dummy regulator
Dec 31 00:32:32 chimeraos kernel: bmi160_i2c i2c-BMI0160:00: supply vddio not found, using dummy regulator
Dec 31 00:32:32 chimeraos kernel: bmi160_i2c i2c-BMI0160:00: mounting matrix not found: using identity...
Dec 31 00:32:32 chimeraos kernel: bmi160_i2c i2c-BMI0160:00: Wrong chip id, got 27 expected d1
```

[journalctl.log](https://github.com/ShadowBlip/HandyGCCS/files/10326699/journalctl.log)

mkoelbaek-skyspecs, 2023-05-25:
I think I have this same issue. The output of my journalctl is: `May 25 20:11:51 chimeraos python3[582]: [230525_20:11:51 | handycon.py:504:get_gyro] [Errno 121] Remote I/O error | Gyro device not initialized. Ensure bmi160_i2c and i2c_dev modules are loaded. Skipping gyro device setup.
May 25 20:20:01 chimeraos kernel: i2c_dev: i2c /dev entries driver
May 25 20:20:01 chimeraos kernel: bmi160_i2c i2c-BMI0160:00: supply vdd not found, using dummy regulator
May 25 20:20:01 chimeraos kernel: bmi160_i2c i2c-BMI0160:00: supply vddio not found, using dummy regulator
May 25 20:20:01 chimeraos kernel: bmi160_i2c i2c-BMI0160:00: mounting matrix not found: using identity...
May 25 20:20:01 chimeraos kernel: bmi160_i2c i2c-BMI0160:00: Error reading chip id
May 25 20:20:01 chimeraos kernel: bmi160_i2c: probe of i2c-BMI0160:00 failed with error -121
May 25 20:20:00 chimeraos systemd-modules-load[407]: Inserted module 'i2c_dev'
May 25 20:20:00 chimeraos systemd-modules-load[407]: Inserted module 'bmi160_i2c'
May 25 20:20:01 chimeraos kernel: Goodix-TS i2c-GDIX1002:00: supply AVDD28 not found, using dummy regulator
May 25 20:20:01 chimeraos kernel: Goodix-TS i2c-GDIX1002:00: supply VDDIO not found, using dummy regulator
May 25 20:20:01 chimeraos kernel: Goodix-TS i2c-GDIX1002:00: Unexpected ACPI resources: gpio_count 1, gpio_int_idx 0
May 25 20:20:01 chimeraos kernel: Goodix-TS i2c-GDIX1002:00: ID 927, version: 0040
May 25 20:20:01 chimeraos kernel: Goodix-TS i2c-GDIX1002:00: Invalid config (0, 0, 0), using defaults
`

Did you ever find a fix? Awesome work btw :).

pastaq, 2023-06-16:
> @pastaq will it work out-of-box on winmax2 with bmi160 (knowing there is wrong id)?

There are 2 version of the wm2, where they switched to a BMI260 later in production. I thin the 160 ran out of stock as every company switched out around the same time. The later one's won't work, but the ones with the 160 and a bad ACPI ID will work with the python version of the drivers.

pastaq, 2023-07-17:
Due to the fractured state of gyro support in Linux currently I'm going to close this for now. I'm pulling gyro support until it can be refactored to work in a universal way.

closed 2023-07-17 by pastaq