ShadowBlip/HandyGCCS #28: Ayaneo Air no force feedback

alphonsez1, 2022-09-07:
Hi, I find there is no force feedback in Ayaneo Air. I tried the python script. The `do_rumble()` function is working as expected. Do you have something in mind that I can try?

alphonsez1, 2022-09-08:
I used `fftest` to verify and it seems to work. I made some changes to make the effects reusable. (Haven't polished yet)

https://github.com/alphonsez1/HandyGCCS/commit/8bbddd63f33d4339e5d0f6f9c9719766f470ac52

However there is still no rumble in the games. I did some search and find this might be related.
https://github.com/ValveSoftware/steam-for-linux/issues/8590

alphonsez1, 2022-09-08:
I tried to disable Steam Input in Ori and the Blind Forest. Then I have rumble.

It only works with games to support controllers by themselves. I tried Portal and it doesn't work. 

To disable Steam Input, on the right side of the Play button there is a button with controller logo. Click that button and there is a settings button with gear logo. Click this one and at the bottom you should find "Disable Steam Input".

pastaq, 2022-11-03:
A recent update to the steam beta client appears to have fixed the steam input rumble issues. Are you still experiencing this issue?

closed 2022-11-09 by pastaq