ShadowBlip/HandyGCCS #34: Aya Neo Air analog stick LEDs

ghost573, 2022-10-06:
My 2021 seems to be stuck in limbo but thankfully my Air Pro came so was wondering if there will be support to modify the Air's analog stick LEDs, personally I want to turn them off cause it's quite bright.

RustyStriker, 2022-10-21:
I would love that as well, if i knew what device(i think its just writing to the game controller itself) and what protocol(what to write and how to write to it) i could set up a cli tool, so if anyone got any leads it  will  be very useful

ghost573, 2022-10-21:
> also might be a better idea to have this as a different plugin maybe?

I would prefer that, doesn't need to have a GUI, editing the values in a config file is good enough.
Anything to turn the LEDs off, at least when the device is on, playing it at night is difficult, the screen is great but the LEDs are just too bright.

pastaq, 2022-11-03:
This is on my TODO list, but will likely be a part of another project. I'd like to create a UI for the Aya Neo Air/2 and the AOK ZOE/OXP Pro for editing LEDs. AYA have been pretty closed mouthed about the LED controller though so I'm needing to reverse engineer it. This would be outside the scope of HandyGCCS though and I don't currently have an ETA.

closed 2022-11-03 by pastaq