ShadowBlip/HandyGCCS #35: Report new equipment AOKZOE A1 AR07

danyi, 2022-10-21:
I got a new device  AOKZOE A1 AR07
 
Report according to your instructions

System ID:
AOKZOE A1 AR07

X-Box 360 Device:

keyboard Device:
AT Translated Set 2 keyboard | isa0060/serio0/input0 | bustype: 17 vendor: 1 product: 1 version: 43907

All Devices:
Microsoft X-Box 360 pad | usb-0000:04:00.3-4/input0 | bustype: 3 vendor: 1118 product: 654 version: 41474
AT Translated Set 2 keyboard | isa0060/serio0/input0 | bustype: 17 vendor: 1 product: 1 version: 43907

Captured Keymaps:

danyi, 2022-10-21:
I haven't been able to find you lately

I'll try to fix it on my fork first

danyi, 2022-10-22:
sudo python capture-system.py
Scanning system and creating device profile.
Gathering system info...
Unable to identify compatible controller. Additional steps may be required after uploading your capture file to fully integrate your device.
Capture complete. Please upload the file titled "capture_file.txt" in a new GitHub issue to https://github.com/ShadowBlip/aya-neo-fixes/issues and any additional information you have.
Exception ignored in: <function InputDevice.__del__ at 0x7f3c828c0ee0>
Traceback (most recent call last):
  File "/usr/lib/python3.10/site-packages/evdev/device.py", line 160, in __del__
  File "/usr/lib/python3.10/site-packages/evdev/device.py", line 305, in close
  File "/usr/lib/python3.10/site-packages/evdev/eventio_async.py", line 55, in close
  File "/usr/lib/python3.10/asyncio/events.py", line 739, in get_event_loop_policy
  File "/usr/lib/python3.10/asyncio/events.py", line 732, in _init_event_loop_policy
ImportError: sys.meta_path is None, Python is likely shutting down
Exception ignored in: <function InputDevice.__del__ at 0x7f3c828c0ee0>
Traceback (most recent call last):
  File "/usr/lib/python3.10/site-packages/evdev/device.py", line 160, in __del__
  File "/usr/lib/python3.10/site-packages/evdev/device.py", line 305, in close
  File "/usr/lib/python3.10/site-packages/evdev/eventio_async.py", line 55, in close
  File "/usr/lib/python3.10/asyncio/events.py", line 739, in get_event_loop_policy
  File "/usr/lib/python3.10/asyncio/events.py", line 732, in _init_event_loop_policy
ImportError: sys.meta_path is None, Python is likely shutting down
Exception ignored in: <function InputDevice.__del__ at 0x7f3c828c0ee0>
Traceback (most recent call last):
  File "/usr/lib/python3.10/site-packages/evdev/device.py", line 160, in __del__
  File "/usr/lib/python3.10/site-packages/evdev/device.py", line 305, in close
  File "/usr/lib/python3.10/site-packages/evdev/eventio_async.py", line 55, in close
  File "/usr/lib/python3.10/asyncio/events.py", line 739, in get_event_loop_policy
  File "/usr/lib/python3.10/asyncio/events.py", line 732, in _init_event_loop_policy
ImportError: sys.meta_path is None, Python is likely shutting down
Exception ignored in: <function InputDevice.__del__ at 0x7f3c828c0ee0>
Traceback (most recent call last):
  File "/usr/lib/python3.10/site-packages/evdev/device.py", line 160, in __del__
  File "/usr/lib/python3.10/site-packages/evdev/device.py", line 305, in close
  File "/usr/lib/python3.10/site-packages/evdev/eventio_async.py", line 55, in close
  File "/usr/lib/python3.10/asyncio/events.py", line 739, in get_event_loop_policy
  File "/usr/lib/python3.10/asyncio/events.py", line 732, in _init_event_loop_policy
ImportError: sys.meta_path is None, Python is likely shutting down
Exception ignored in: <function InputDevice.__del__ at 0x7f3c828c0ee0>
Traceback (most recent call last):
  File "/usr/lib/python3.10/site-packages/evdev/device.py", line 160, in __del__
  File "/usr/lib/python3.10/site-packages/evdev/device.py", line 305, in close
  File "/usr/lib/python3.10/site-packages/evdev/eventio_async.py", line 55, in close
  File "/usr/lib/python3.10/asyncio/events.py", line 739, in get_event_loop_policy
  File "/usr/lib/python3.10/asyncio/events.py", line 732, in _init_event_loop_policy
ImportError: sys.meta_path is None, Python is likely shutting down
Exception ignored in: <function InputDevice.__del__ at 0x7f3c828c0ee0>
Traceback (most recent call last):
  File "/usr/lib/python3.10/site-packages/evdev/device.py", line 160, in __del__
  File "/usr/lib/python3.10/site-packages/evdev/device.py", line 305, in close
  File "/usr/lib/python3.10/site-packages/evdev/eventio_async.py", line 55, in close
  File "/usr/lib/python3.10/asyncio/events.py", line 739, in get_event_loop_policy
  File "/usr/lib/python3.10/asyncio/events.py", line 732, in _init_event_loop_policy
ImportError: sys.meta_path is None, Python is likely shutting down
Exception ignored in: <function InputDevice.__del__ at 0x7f3c828c0ee0>
Traceback (most recent call last):
  File "/usr/lib/python3.10/site-packages/evdev/device.py", line 160, in __del__
  File "/usr/lib/python3.10/site-packages/evdev/device.py", line 305, in close
  File "/usr/lib/python3.10/site-packages/evdev/eventio_async.py", line 55, in close
  File "/usr/lib/python3.10/asyncio/events.py", line 739, in get_event_loop_policy
  File "/usr/lib/python3.10/asyncio/events.py", line 732, in _init_event_loop_policy
ImportError: sys.meta_path is None, Python is likely shutting down
Exception ignored in: <function InputDevice.__del__ at 0x7f3c828c0ee0>
Traceback (most recent call last):
  File "/usr/lib/python3.10/site-packages/evdev/device.py", line 160, in __del__
  File "/usr/lib/python3.10/site-packages/evdev/device.py", line 305, in close
  File "/usr/lib/python3.10/site-packages/evdev/eventio_async.py", line 55, in close
  File "/usr/lib/python3.10/asyncio/events.py", line 739, in get_event_loop_policy
  File "/usr/lib/python3.10/asyncio/events.py", line 732, in _init_event_loop_policy
ImportError: sys.meta_path is None, Python is likely shutting down
Exception ignored in: <function InputDevice.__del__ at 0x7f3c828c0ee0>
[... 94 lines cut ...]
ImportError: sys.meta_path is None, Python is likely shutting down
Exception ignored in: <function InputDevice.__del__ at 0x7f3c828c0ee0>
Traceback (most recent call last):
  File "/usr/lib/python3.10/site-packages/evdev/device.py", line 160, in __del__
[... cut ...]
Traceback (most recent call last):
  File "/usr/lib/python3.10/site-packages/evdev/device.py", line 160, in __del__
  File "/usr/lib/python3.10/site-packages/evdev/device.py", line 305, in close
  File "/usr/lib/python3.10/site-packages/evdev/eventio_async.py", line 55, in close
  File "/usr/lib/python3.10/asyncio/events.py", line 739, in get_event_loop_policy
  File "/usr/lib/python3.10/asyncio/events.py", line 732, in _init_event_loop_policy
ImportError: sys.meta_path is None, Python is likely shutting down

ruineka, 2022-10-22:
@danyi  Could you test this? https://github.com/ShadowBlip/HandyGCCS/tree/aokzoe-test

Gawah, 2022-10-27:
> @danyi Could you test this? https://github.com/ShadowBlip/HandyGCCS/tree/aokzoe-test I updated the capture script as well as the controller script itself, please let me know if this works for you.
I tried, but the script failed occasionally. I need to press it many times to call out the performance menu. And when I press the button at the bottom left corner, sometimes the script will call the event of pressing key A. But the first time you press it, you can always call out the menu smoothly.

ruineka, 2022-10-27:
> I tried, but the script failed occasionally. I need to press it many times to call out the performance menu. And when I press the button at the bottom left corner, sometimes the script will call the [...]

Ah so it would appear there are some logic changes that will need to be done to get this working properly. I don't have an Aokzoe sadly so without one doing this would be rather difficult.

Gawah, 2022-10-27:
Can you map the bottom left button to the steam key? Because when pressing the steam key for a long time, you can use the combination key. For example, steam+RB is the screenshot function, and steam+x is the display keyboard function. In this way, you can only need a custom button to implement all functions on aok or onexplayer.

> @danyi Could you test this? https://github.com/ShadowBlip/HandyGCCS/tree/aokzoe-test I updated the capture script as well as the controller script itself, please let me know if this works for you.

Can you map the bottom left button to the steam key? Because when pressing the steam key for a long time, you can use the combination key. For example, steam+RB is the screenshot function, and steam+x is the display keyboard function. In this way, you can only need a custom button to implement all functions on aok or onexplayer.
[image]

Gawah, 2022-10-27:
> Ah so it would appear there are some logic changes that will need to be done to get this working properly. I don't have an Aokzoe sadly so without one doing this would be rather difficult.

I'm sorry to hear that. If there's anything I can do to help you test, just let me know

ruineka, 2022-10-27:
> Can you map the bottom left button to the steam key? Because when pressing the steam key for a long time, you can use the combination key. For example, steam+RB is the screenshot function, and [...]

Sounds like it acts the same way the 8.4 versions of the OXP work. On long press it sends a home signal which is capable of combo keys, the minis behave differently.

To change this behavior of the short press etc I'll need results from an evtest output. If you can install evtest and select the keyboard/gamepad events and press all keys and log them for me I can see what I'm able to do.

Gawah, 2022-10-27:
> 

> Sounds like it acts the same way the 8.4 versions of the OXP work. On long press it sends a home signal which is capable of combo keys, the minis behave differently.
>
> To change this behavior of the [...]
I tested. There is no output when I press the custom button in event 6, and the output can be obtained in event 15. However, it is consistent with the previous situation in the deck UI. Sometimes, I can only output once after pressing several times.
[event6_log.txt](https://github.com/ShadowBlip/HandyGCCS/files/9881855/event6_log.txt)
[handheld controller_log.txt](https://github.com/ShadowBlip/HandyGCCS/files/9881857/handheld.controller_log.txt)

ruineka, 2022-10-27:
> I tested. There is no output when I press the custom button in event 6, and the output can be obtained in event 15. However, it is consistent with the previous situation in the deck UI. Sometimes, I [...]

Without the Handheld Controller (handyGCCS) running could you (without being in KB/mouse mode) try to find the event that reads out values from the left button, kb button, etc? It should be a keyboard event and it should have multiple readings that isn't (316).

Gawah, 2022-10-27:
In this test, whenever I press the key, there will be a corresponding output. But the turbo button has no output
[event2_log.txt](https://github.com/ShadowBlip/HandyGCCS/files/9882020/event2_log.txt)

ruineka, 2022-10-27:
> @ruineka In this test, whenever I press the key, there will be a corresponding output. But the turbo button has no output
> [event2_log.txt](https://github.com/ShadowBlip/HandyGCCS/files/9882020/event2_log.txt)
> 

The turbo button by itself is a firmware thing and doesn't sent keyboard inputs, this is normal. This is exactly what I needed. I'll take a look at it

Gawah, 2022-10-27:
> > @ruineka在这个测试中，每当我按下键时，都会有相应的输出。但是涡轮增压按钮没有输出[event2_log.txt](https://github.com/ShadowBlip/HandyGCCS/files/9882020/event2_log.txt)
> 
> 涡轮增压按钮本身就是固件，不会发送键盘输入，这是正常的。这正是我需要的。我来看看

Ok,, thank you very much.

Gawah, 2022-10-27:
https://user-images.githubusercontent.com/105637287/198367739-d2b7ab4d-2acb-4868-bade-fb69e073afb3.MP4

ruineka, 2022-10-27:
> > > @ruineka在这个测试中，每当我按下键时，都会有相应的输出。但是涡轮增压按钮没有输出[event2_log.txt](https://github.com/ShadowBlip/HandyGCCS/files/9882020/event2_log.txt)
> > 
> > 
> > 涡轮增压按钮本身就是固件，不会发送键盘输入，这是正常的。这正是我需要的。我来看看
> 
> Ok,, thank you very much.

I compared your log to my OXP Mini and the event codes are the same, so the timing must be different. I adjusted the delay time of the events, could you pull these changes and see if it improves the inputs or make them worse?

Gawah, 2022-10-28:
@ruineka No improvement. Sometimes you still need to click several times to call out the menu once. In addition, short pressing the left button will still trigger the A button event, but pressing the left button long will call homemenu without trigger A button.

ruineka, 2022-10-28:
> @ruineka No improvement. Sometimes you still need to click several times to call out the menu once. In addition, short pressing the left button will still trigger the A button event, but pressing the left button long will call homemenu without trigger A button.

The reason you are seeing "A" go through is because the Home + A inputs aren't being sent in the correct time to be counted as a combo key. Could you experiment with the delays from your end and see if you can find a good value?

https://github.com/ShadowBlip/HandyGCCS/blob/74ce266f5deba1368ba34d4b80a4f68f0cb2b5d0/handycon.py#L881

I guess the other thing I should ask is if the event signals are sent out from evtest during on press or release. The Mini always is on release. Meaning the inputd aren't sent until your remove your finger from the button, which was different from how the Aya Neo worked.

Gawah, 2022-10-28:
> > @ruineka No improvement. Sometimes you still need to click several times to call out the menu once. In addition, short pressing the left button will still trigger the A button event, but pressing the left button long will call homemenu without trigger A button.
> 
> The reason you are seeing "A" go through is because the Home + A inputs aren't being sent in the correct time to be counted as a combo key. Could you experiment with the delays from your end and see if you can find a good value?
> 
> https://github.com/ShadowBlip/HandyGCCS/blob/74ce266f5deba1368ba34d4b80a4f68f0cb2b5d0/handycon.py#L881
> 
> I guess the other thing I should ask is if the event signals are sent out from evtest during on press or release. The Mini always is on release. Meaning the inputs aren't sent until you remove your finger from the button, which was different from how the Aya Neo worked.

OK, I will test this value. I just tested the button. It sends the event when the button is released, just like mini.

Gawah, 2022-10-28:
@ruineka  0.05 seems to be the best value. If it is smaller, sometimes the home menu will be called and then the A button will be triggered, or the performance menu will be called after the A button is triggered. Setting to 0.05 has some improvement,but 0.05 still has a chance to trigger the a button. I think it is the best solution to map the left key to the steam key, and then call the combination key manually.

ruineka, 2022-10-28:
> @ruineka  0.05 seems to be the best value. If it is smaller, sometimes the home menu will be called and then the A button will be triggered, or the performance menu will be called after the A button [...]

We need the controller to have a "On Press" signal for an ideal situation. Since we only get the "On release" event it makes combo keys difficult. I'd like to have orange be home so you can hold it and press A or X etc.. not possible though.

Gawah, 2022-10-28:
On ayaneo next, it can press the aya key to call up a homemenu or form a key combination with other buttons. But on aokzoe, when I modify the code to map the steam button to the left button, I can only call out the home menu, and cannot form a combination key with other buttons. And it sometimes requires multiple clicks to call out a menu once.I can't find what caused it.

Gawah, 2022-10-28:
> We need the controller to have a "On Press" signal for an ideal situation. Since we only get the "On release" event it makes combo keys difficult. I'd like to have orange be home so you can hold it [...]
So it need aokzoe official to modify the handle firmware to obtain the button press event?

ruineka, 2022-10-28:
> So it need aokzoe official to modify the handle firmware to obtain the button press event?

Yup. AOKZOE needs to have an EC update to properly fix this issue. Same with One Net.

Gawah, 2022-10-28:
OK, I understand. Thank you.

danyi, 2022-10-28:
I'm back from a week-long business trip

I fixed the problem and I will submit the code later

danyi, 2022-10-28:
turbo button I know from the official

He deals directly from the bottom up

There is no mapping

So let's get rid of this button

closed 2022-10-28 by danyi

danyi, 2022-10-28:
https://github.com/ShadowBlip/HandyGCCS/compare/main...danyi:HandyGCCS:patch-1