ShadowBlip/HandyGCCS #36: Aya Neo Air analog sticks always shifted on Y axis

ghost573, 2022-10-26:
[image]

I have tried calibrating this both on linux through the KDE settings and through the in-built calibration on the Air Pro but it doesn't seem to fix it.

EricFL1998, 2022-10-27:
ya, it makes any fps game unplayable, and select using stick is hard AF

pastaq, 2022-11-03:
I think this issue is out of scope for this project. This appears to be a firmware/hardware issue that requires some sort of calibration. Everything in HandyGCCS assumes calibration is correct. Have you contacted Aya Neo support?

EricFL1998, 2022-11-03:
thats waht I thought，well ggs> I think this issue is out of scope for this project. This appears to be a firmware/hardware issue that requires some sort of calibration. Everything in HandyGCCS assumes calibration is correct. Have you contacted Aya Neo support?

closed 2022-11-03 by pastaq