ShadowBlip/HandyGCCS #38: GPD Win Max 2 support

ajwahab, 2022-11-03:
[capture_file.txt](https://github.com/ShadowBlip/HandyGCCS/files/9933577/capture_file.txt)

The WM2 also has a BMI160 IMU on the I2C bus that is accessible at /sys/bus/iio/devices/iio:device0.

pastaq, 2022-11-04:
I'm not super familiar with the GPD line, do they have extra controller buttons that need to be mapped as well? I imagine there could be compatibility issues with how we capture keyboard events and the built in keyboard. I'm definitely interested in adding support but the unique nature of the device could require a decent amount of back and forth testing. If you're willing to work with me through those issues I can provide you an updated capture script to start button mapping. I have a couple of PR's to validate first that should improve gryo and rumble latencies, so I'd be targeting inclusion after those are done.

danyi, 2022-11-04:
My GPD WINMAX2 6800U
It can be used without mapping
Because he has a normal home button

pastaq, 2022-11-04:
What about the two underside buttons?

ajwahab, 2022-11-04:
@pastaq, that all sounds fine to me - thank you for responding so quickly!  I've updated the issue to include two extra buttons on the bottom of the device.  I didn't call out the menu, start, and select buttons since they are presumably part of the integrated XBox360 controller.

danyi, 2022-11-04:
Oh, yes
This device provides two additional buttons on the back
But it's very easy to touch by mistake
So I wasn't going to use them so I ignored them
I'm sorry
But there are people who need it

pastaq, 2022-11-04:
> @pastaq, that all sounds fine to me - thank you for responding so quickly! I've updated the issue to include two extra buttons on the bottom of the device. I didn't call out the menu, start, and select buttons since they are presumably part of the integrated XBox360 controller.

Awesome, I think I have everything I need. I was thinking the OSK and QAM shortcuts would be best for those buttons, but LMK if you'd prefer a different functionality.

danyi, 2022-11-04:
[image]
Here they are

ajwahab, 2022-11-04:
> Awesome, I think I have everything I need. I was thinking the OSK and QAM shortcuts would be best for those buttons, but LMK if you'd prefer a different functionality.

Sure, let’s start with those.  Just as long as they aren’t bound to `rm -rf /` 😳

danyi, 2022-11-04:
[image]
I use this MEMU as a normal HOME button, and can also use key combinations

danyi, 2022-11-07:
> [image]他们在这里

GPD officially provides a program

In windows, you can set the mapping between these two keys

So all the buttons are usable

I've already tested it

danyi, 2022-11-07:
[image]
These keys are not mapped correctly
I have confirmed with GPD
It will be fixed in future firmware

pastaq, 2022-11-11:
I have a branch that needs testing with (what I hope) is support for this device.

https://github.com/ShadowBlip/HandyGCCS/tree/winmax2

To test, close this repository, `checkout winamax2` and run `sudo python3 handycon.py`

left should open QAM in GamepadUI, right OSK, and both together should toggle gyro on/off.

danyi, 2022-11-12:
I tested it. He doesn't have a job

I think that's the reason

A software is officially available under windows

You can customize the mapping of these two keys

I set the keys to map to R2 and L1

May result in getting a different value than you set

And I don't think it's a good idea

Because these two buttons are very soft

It's very easy to mistouch

danyi, 2022-11-12:
I can use the shortcut keys to open these things

So the machine might just need a gyroscope improvement

ajwahab, 2022-11-12:
> I have a branch that needs testing with (what I hope) is support for this device.
>
> https://github.com/ShadowBlip/HandyGCCS/tree/winmax2
>
> To test, close this repository, `checkout winamax2` and run [...]

Wow - thanks!  

I made a [fork](https://github.com/ajwahab/HandyGCCS) to work through some modifications.  The IMU isn't being detected, but that's because it's on a different I2C bus (i2c-2) and is using a different I2C slave address (0x69) than the other devices.  That said, even with that knowledge, I've been unable to coerce a response from the BMI160 when using basic I2C comms.  The IMU is, however, accessible on the iio bus, so I tossed together a stub of a sysfs-based python class for interfacing with it.  Consequently, I added some logic to the main script to select between BMI160_i2c classic and the sysfs interface.

Aside from the IMU not being detected, the handycon.py would fail after taking over the keyboard in such a manner that I would have to use an onscreen keyboard to attempt to kill the script (couldn't in some cases, so rebooted).  I've attached two screen captures of the error messages - apologies for not just copying the text.  
[image]
[image]

pastaq, 2022-11-12:
It seems like the xbox 360 pad device is "busy" and not letting you grab it. That is strange behavior that I haven't previously seen. If you install the `evtest` package does the controller show up when run as user?

One of those errors can be removed by adding the word await to this line. https://github.com/ajwahab/HandyGCCS/blob/73f2fa704354630ca1af0e857576863a17e27292/handycon.py#L583 That was my mistake.

ajwahab, 2022-11-12:
Thanks for taking a look.  

The controller shows up twice in evtest output as events 18 and 22.  Control inputs are only logged for 18, so I assume that’s the one we want.

The script now runs without error, thanks in part to your correction, however I am unable to terminate it with ctrl+c.  Putting it to sleep with ctrl+z works, but I can’t seem to kill or pkill it.  It also seems that something is wonky with the mapping, for example, pushing down on the left stick brings up the OSK and the keyboard arrow keys were mapped to alphabetical characters.

Is anything lost by omitting the keyboard grab? The Win Max 2 has a full hardware keyboard, so we could probably do without overriding it.

pastaq, 2022-11-12:
The second device is probably the steam input virtual controller. Nothing to worry about there.

The keyboard grab is there to allow us to transmit macros for the events we want (OSK and QAM) when the detected keys are pressed and not create double inputs on things. I had hoped emitting all other keyboard events unmodified would work, but that seems to cause some issue from your report. We could probably put in a bool that gets set to false in the GPD device detection that would determine if the keyboard is grabbed or not. We might still be able to read the keys in that case and just omit the general emit_events(), which would at least allow us to enable/disable the gyro. If ctrl+z is working then there's a logic bug somewhere in the async loop that is causing a race condition, or the c key could be wrong like the arrow keys. Honestly that's pretty strange behavior, as the left stick should be a standard controller input which we don't really touch, except to add gyro acceleration.

ajwahab, 2022-11-12:
A bool sounds good to me if you’re okay with it.  Removing the keyboard from the equation, should help to debug any issues with the controller and gyro.

I don’t mind poking around the system’s inputs and events if you give me pointers on the nature of those numbers used in the “active” conditionals and/or how the seed_event, event_queue, should works.

Shall I submit a PR with my changes so that we don’t diverge too much?

pastaq, 2022-11-12:
You can do a WIP PR against the branch if you want, but merging it into my repo at the moment wouldn't really be productive since I don't have hardware to test it on. I'm not currently working on any other major updates.

The "active" conditional buttons are just event codes. I thought about translating them for readability but I'm being lazy. You can find a full list here:

https://github.com/torvalds/linux/blob/master/include/uapi/linux/input-event-codes.h

There is a commented event code print statement at the top of the keyboard loop thay can be used for debugging as well, it prints a lot of data about each event.

If you want to hop on the ChimeraOS discord at some point I'm available.

pastaq, 2022-11-15:
I'm trying to figure out why the python driver isn't working for you since the module is the same as all the other devices. I assume you tried specifying the address and bus for the BMI160 at the driver init since your code supports that at device detection. Did it produce any errors when you tried that?

What is the output of  `lmsod | grep i2c`?
For the original aya neo devices I needed to `modprobe i2c_dev` due to an improper PCI Device ID. I'm curious if that will allow the BMI160 module to start working for the GPD as well?

ajwahab, 2022-11-19:
Indeed, it is confounding.  I did try specifying the bus (i2c-2) and address (0x69), but no dice.  When I attempted to read from the BMI160 directly with I2C tools, the returned results were all 0.

```shell
$ lsmod | grep i2c
bmi160_i2c             16384  0
bmi160_core            20480  2 bmi160_i2c,bmi160_spi
i2c_piix4              36864  0
i2c_hid_acpi           16384  0
i2c_hid                40960  1 i2c_hid_acpi
i2c_dev                24576  0
```
The WM2 had a similar issue with the BMI160 device ID.  Initially, we addressed the issue with a DKMS patch as was done with the Aya Neo devices.  We now use an ACPI override.

ajwahab, 2022-11-19:
Well, good news - all of a sudden, the IMU is responding to BMI160-i2c.py!  I’ll update my PR accordingly.

pastaq, 2022-11-19:
Awesome news.

pastaq, 2022-11-19:
Merged all your changes and added the grab flag we discussed. Can you test out #43 and let me know if you have any issues? In theory all your keyboard buttons should work but holding both underside buttons will enable/disable gyro input now.

pastaq, 2022-11-22:
I read on the GPD discord that the keyboard and the underside buttons are on two separate event devices and the underside buttons are actually detected as a mouse. What does the `evtest` package show for this when rus as sudo? If I can capture a different device and ignore the main keyboard that would solve a few problems.

pastaq, 2022-11-23:
Also, do you have the intel or AMD version?

ajwahab, 2022-11-23:
Thanks for merging!  I’ll give it a run tomorrow along with evtest.  

Should it help, I do know that the back buttons correspond to `  Mouse for Windows | usb-0000:74:00.3-4/input0 | bustype: 3 vendor: 12068 product: 309 version: 272 ` in the capture file posted above and `/dev/input/event18` on my system, which is the AMD version.

It may be worth noting that all of the production Win Max 2 are AMD.  Only a limited number of early-production and/or domestic Intel systems were made by GPD, so it may not be worth the development effort to support them.

pastaq, 2022-11-23:
> Should it help, I do know that the back buttons correspond to ` Mouse for Windows | usb-0000:74:00.3-4/input0 | bustype: 3 vendor: 12068 product: 309 version: 272` in the capture file posted above and `/dev/input/event18` on my system, which is the AMD version.

This is very interesting. I should be able to capture that device instead. This is the device that produces the keycodes 0 and 9? Grabbing that instead of the keyboard would solve the previous issues, as long it it generates button presses.

ajwahab, 2022-11-23:
Thanks for that.  Runs and exits cleanly after correcting a few minor bugs.  Just submitted another WIP PR.  Still need to find a means of testing the gyro functionality.  Motion inputs weren't recognized in dolphin controller configurator, and virtual controller wasn't picked up by steam...perhaps steam overlay was interfering with the latter?

Probably not so useful now, but here's that promised evtest dump.

[evtest_event18.txt](https://github.com/ShadowBlip/HandyGCCS/files/10076838/evtest_event18.txt)

pastaq, 2022-11-24:
Thanks for all your help with this. GPD support has been on my wishlist for a long time.

closed 2022-11-24 by pastaq