ShadowBlip/HandyGCCS #39: Extra button functionality such as long press, double press, triple press

ghost573, 2022-11-04:
Might sound like a weird request but would it be possible to add extra functionality to each button?

For example, I can configure in handycon.py so the on my Aya Neo Air Pro, I can press the small button once for the keyboard or long press for Ctrl+Alt+Esc.

Additionally can adjust the time between presses.

pastaq, 2022-11-04:
This is possible for buttons that expose three button states, such as the VOL UP/DOWN, and the ESC button on the 2021. Most of the buttons don't really work this way and I imagine would introduce quite a bit of input lag, as we would need to verify the users intent via the button delay before doing anything. I would like to implement an interface and/or configuration file to remap the functions of each button press to be more customizable in the future, but that would be an override vice an extension. If you have an idea on how best to implement this feature in a way that wouldn't result in a reduced user experience, I am more than willing to test out any pull request.

ghost573, 2022-11-04:
> I would like to implement an interface and/or configuration file to remap the functions of each button press to be more customizable in the future

That would be amazing and would be satisfactory for me.
I'll make a separate issue for that as I have just one request for it.

closed 2022-11-04 by ghost573