ShadowBlip/HandyGCCS #40: Add a new handheld device ONEXPLAYER Mini Pro

danyi, 2022-11-04:
His ID is ONEXPLAYER Mini Pro

danyi, 2022-11-04:
I need to get this device next week

danyi, 2022-11-05:
Oh, I've got the equipment

He used the exact same motherboard as AOK

The shape and the keys are exactly the same

But the more the mini

I think their buttons will have the same feedback

I need to run the test after work

danyi, 2022-11-06:
I got the machine, and sure enough it was the same as AOKZOE

But the bugs are the same

After pressing the button, it will trigger twice

Long press the button will not return to the position

When you release your finger, the button keeps firing

Luckily I was in China

I can contact them ONEXPLAYER to solve the problem

Please wait for my good news

pastaq, 2022-11-09:
AOK and One Net have been pretty silent lately about these devices. Can you run capture-system.py on this device and post the resulting text file? I need some DMI and PCI data for it.

desantienator, 2022-11-26:
Uploading capture-system.py capture file
[capture_file.txt](https://github.com/ShadowBlip/HandyGCCS/files/10095639/capture_file.txt)

pastaq, 2022-11-26:
> Uploading capture-system.py capture file [capture_file.txt](https://github.com/ShadowBlip/HandyGCCS/files/10095639/capture_file.txt)

Can you please run that as sudo? All the information I need is not accessible to the script otherwise.

desantienator, 2022-11-26:
Uploading second capture
[capture_file.txt](https://github.com/ShadowBlip/HandyGCCS/files/10095654/capture_file.txt)

pastaq, 2022-11-26:
I have a branch available for testing
#51 

Stop any current handycon.service, clone the repository, switch to the `mini-pro` branch, and manually run `sudo handycon.py`

danyi, 2022-11-26:
When I press the direction button or the waist bar for more than 5 seconds, it will keep firing
Sometimes it vibrates abnormally, it vibrates constantly, but I'm just using it normally, I'm not pressing the key combination

closed 2022-11-26 by pastaq

danyi, 2022-11-27:
[capture_file.txt](https://github.com/ShadowBlip/HandyGCCS/files/10098637/capture_file.txt)