ShadowBlip/HandyGCCS #47: [ENHANCEMENT] configuration request for AOKZOE A1 November batch

NikhilSaini38, 2022-11-13:
here is the data from capture_system.py (running latest holoiso build)
```
System ID:
AOKZOE A1 AR07

X-Box 360 Device:

keyboard Device:

All Devices:
Microsoft X-Box 360 pad 1 |  | bustype: 3 vendor: 10462 product: 4607 version: 1
Microsoft X-Box 360 pad 0 |  | bustype: 3 vendor: 10462 product: 4607 version: 1
Microsoft X-Box 360 pad | usb-0000:e4:00.3-4/input0 | bustype: 3 vendor: 1118 product: 654 version: 41474
Handheld Controller | py-evdev-uinput | bustype: 3 vendor: 1118 product: 654 version: 272
HD-Audio Generic Headphone | ALSA | bustype: 0 vendor: 0 product: 0 version: 0
HD-Audio Generic Mic | ALSA | bustype: 0 vendor: 0 product: 0 version: 0
HD-Audio Generic HDMI/DP,pcm=7 | ALSA | bustype: 0 vendor: 0 product: 0 version: 0
HD-Audio Generic HDMI/DP,pcm=3 | ALSA | bustype: 0 vendor: 0 product: 0 version: 0
GXTP7385:00 27C6:0118 Keyboard | i2c-GXTP7385:00 | bustype: 24 vendor: 10182 product: 280 version: 256
GXTP7385:00 27C6:0118 UNKNOWN | i2c-GXTP7385:00 | bustype: 24 vendor: 10182 product: 280 version: 256
GXTP7385:00 27C6:0118 Stylus | i2c-GXTP7385:00 | bustype: 24 vendor: 10182 product: 280 version: 256
GXTP7385:00 27C6:0118 | i2c-GXTP7385:00 | bustype: 24 vendor: 10182 product: 280 version: 256
PC Speaker | isa0061/input0 | bustype: 16 vendor: 31 product: 1 version: 256
Video Bus | LNXVIDEO/video/input0 | bustype: 25 vendor: 0 product: 6 version: 0
MM711 Lite Gaming Mouse COOLER MASTER | usb-0000:e5:00.0-1/input1 | bustype: 3 vendor: 6700 product: 45057 version: 273
Power Button | LNXPWRBN/button/input0 | bustype: 25 vendor: 0 product: 1 version: 0
Power Button | PNP0C0C/button/input0 | bustype: 25 vendor: 0 product: 1 version: 0

Captured Keymaps:

```

NikhilSaini38, 2022-11-13:
relates to #40

pastaq, 2022-11-13:
Can you rerun the script without handycon.service running?

NikhilSaini38, 2022-11-13:
I ran this one with sudo, do you want it with sudo or without?

NikhilSaini38, 2022-11-13:
service stopped, no sudo
```
System ID:
AOKZOE A1 AR07

X-Box 360 Device:

keyboard Device:
AT Translated Set 2 keyboard | isa0060/serio0/input0 | bustype: 17 vendor: 1 product: 1 version: 43907

All Devices:
AT Translated Set 2 keyboard | isa0060/serio0/input0 | bustype: 17 vendor: 1 product: 1 version: 43907
Microsoft X-Box 360 pad 1 |  | bustype: 3 vendor: 10462 product: 4607 version: 1
Microsoft X-Box 360 pad | usb-0000:e4:00.3-4/input0 | bustype: 3 vendor: 1118 product: 654 version: 41474

Captured Keymaps:

```

pastaq, 2022-11-14:
Can you run the attached version of the capture system script? It should detect both the controller and keyboard now. This is to make sure the keymaps are the same as the OXP devices. 

I'll need the following buttons pressed once it is detecting keypresses:
Yellow - Short press
Yellow - Long press (~2 seconds)
KB short press
Yellow + KB short press
Yellow + |||| (the one below keyboard)
|||| short/long shouldn't do anything, but you can try long and short presses for that. Make sure you don't leave it in low fan speed mode if that happens.

[capture_system.zip](https://github.com/ShadowBlip/HandyGCCS/files/10006931/capture_system.zip)

NikhilSaini38, 2022-11-15:
@pastaq did them in order you asked. yellow/home key responds as long if pressed for ~3seconds. turbo/|||| key doesnt respond to either short or long press
```
System ID:
AOKZOE A1 AR07

X-Box 360 Device:
Microsoft X-Box 360 pad | usb-0000:e4:00.3-4/input0 | bustype: 3 vendor: 1118 product: 654 version: 41474

keyboard Device:
AT Translated Set 2 keyboard | isa0060/serio0/input0 | bustype: 17 vendor: 1 product: 1 version: 43907

All Devices:
Microsoft X-Box 360 pad 0 |  | bustype: 3 vendor: 10462 product: 4607 version: 1
Microsoft X-Box 360 pad | usb-0000:e4:00.3-4/input0 | bustype: 3 vendor: 1118 product: 654 version: 41474
AT Translated Set 2 keyboard | isa0060/serio0/input0 | bustype: 17 vendor: 1 product: 1 version: 43907

Captured Keymaps:
[125]
[32, 125]
[34, 125]
[97]
[97, 125]
[24, 97, 125]
[100]
[97, 100]
[97, 100, 111]
[99, 125]
```

NikhilSaini38, 2022-11-15:
stdout of the script

```
Scanning system and creating device profile.
Gathering system info...
Successfully identified compatible controllers. Press each non-functioning button in succession. When complete press ctrl+c to end capture.
/home/deck/./capture_system.py:169: DeprecationWarning: There is no current event loop
  asyncio.ensure_future(capture_events(xb360))
/home/deck/./capture_system.py:170: DeprecationWarning: There is no current event loop
  asyncio.ensure_future(capture_events(keybd))
/home/deck/./capture_system.py:172: DeprecationWarning: There is no current event loop
  loop = asyncio.get_event_loop()

# short pressed home/yellow key 

[] [125] event at 1668504398.803637, code 04, type 04, val 219
[] [125] event at 1668504398.803637, code 125, type 01, val 01
[125] [125] event at 1668504398.803637, code 00, type 00, val 00
Identified new keymap:  [125]
[] [32, 125] event at 1668504398.806395, code 04, type 04, val 32
[] [32, 125] event at 1668504398.806395, code 32, type 01, val 01
[32, 125] [32, 125] event at 1668504398.806395, code 00, type 00, val 00
Identified new keymap:  [32, 125]
[] [32] event at 1668504398.813491, code 04, type 04, val 219
[] [32] event at 1668504398.813491, code 125, type 01, val 00
[] [32] event at 1668504398.813491, code 00, type 00, val 00

# long pressed home/yellow key

[] [125] event at 1668504422.556639, code 04, type 04, val 219
[] [125] event at 1668504422.556639, code 125, type 01, val 01
[125] [125] event at 1668504422.556639, code 00, type 00, val 00
[125] [34, 125] event at 1668504422.559520, code 04, type 04, val 34
[125] [34, 125] event at 1668504422.559520, code 34, type 01, val 01
[34, 125] [34, 125] event at 1668504422.559520, code 00, type 00, val 00
Identified new keymap:  [34, 125]
^X@sg[] [34] event at 1668504422.566544, code 04, type 04, val 219
[] [34] event at 1668504422.566544, code 125, type 01, val 00
[] [34] event at 1668504422.566544, code 00, type 00, val 00

# short pressed KB key

[] [97] event at 1668504428.348947, code 04, type 04, val 157
[] [97] event at 1668504428.348947, code 97, type 01, val 01
[97] [97] event at 1668504428.348947, code 00, type 00, val 00
Identified new keymap:  [97]
[] [97, 125] event at 1668504428.353738, code 04, type 04, val 219
[] [97, 125] event at 1668504428.353738, code 125, type 01, val 01
[97, 125] [97, 125] event at 1668504428.353738, code 00, type 00, val 00
Identified new keymap:  [97, 125]
[] [24, 97, 125] event at 1668504428.356806, code 04, type 04, val 24
[] [24, 97, 125] event at 1668504428.356806, code 24, type 01, val 01
[24, 97, 125] [24, 97, 125] event at 1668504428.356806, code 00, type 00, val 00
Identified new keymap:  [24, 97, 125]
^X@s^O[] [97, 125] event at 1668504428.361726, code 04, type 04, val 24
[] [97, 125] event at 1668504428.361726, code 24, type 01, val 00
[] [97, 125] event at 1668504428.361726, code 00, type 00, val 00
[] [97] event at 1668504428.368812, code 04, type 04, val 219
[] [97] event at 1668504428.368812, code 125, type 01, val 00
[] [97] event at 1668504428.368812, code 00, type 00, val 00

# short pressed yellow/home key + KB key

[] [100] event at 1668504435.473122, code 04, type 04, val 184
[] [100] event at 1668504435.473122, code 100, type 01, val 01
[100] [100] event at 1668504435.473122, code 00, type 00, val 00
Identified new keymap:  [100]
[] [97, 100] event at 1668504435.477901, code 04, type 04, val 157
[] [97, 100] event at 1668504435.477901, code 97, type 01, val 01
[97, 100] [97, 100] event at 1668504435.477901, code 00, type 00, val 00
Identified new keymap:  [97, 100]
[] [97, 100, 111] event at 1668504435.482927, code 04, type 04, val 211
[] [97, 100, 111] event at 1668504435.482927, code 111, type 01, val 01
[97, 100, 111] [97, 100, 111] event at 1668504435.482927, code 00, type 00, val 00
Identified new keymap:  [97, 100, 111]
[] [97, 100] event at 1668504435.490920, code 04, type 04, val 211
[] [97, 100] event at 1668504435.490920, code 111, type 01, val 00
[] [97, 100] event at 1668504435.490920, code 00, type 00, val 00
[] [100] event at 1668504435.497945, code 04, type 04, val 157
[] [100] event at 1668504435.497945, code 97, type 01, val 00
[] [100] event at 1668504435.497945, code 00, type 00, val 00

# short pressed yellow/home key + turbo/|||| key

[] [125] event at 1668504441.682611, code 04, type 04, val 219
[] [125] event at 1668504441.682611, code 125, type 01, val 01
[125] [125] event at 1668504441.682611, code 00, type 00, val 00
[125] [99, 125] event at 1668504441.687422, code 04, type 04, val 183
[125] [99, 125] event at 1668504441.687422, code 99, type 01, val 01
[99, 125] [99, 125] event at 1668504441.687422, code 00, type 00, val 00
Identified new keymap:  [99, 125]
[] [99] event at 1668504441.695431, code 04, type 04, val 219
[] [99] event at 1668504441.695431, code 125, type 01, val 00
[] [99] event at 1668504441.695431, code 00, type 00, val 00
```

pastaq, 2022-11-15:
The turbo key is handled by the BIOS directly and is OS independent. It is still making the changes to the SOC either way. The way they handle it is a little... sketchy. We've seen systems that completely shut off the fan when its pressed and the auto shutdown on overtemp fails, leading to dangerous heat levels.

NikhilSaini38, 2022-11-15:
> The turbo key is handled by the BIOS directly and is OS independent. It is still making the changes to the SOC either way. The way they handle it is a little... sketchy. We've seen systems that [...]

AOKZOE A1 and OXP mini pro ditched that button controlling hardware directly, now they use that button to toggle the overlay they made for hardware control. @pastaq

pastaq, 2022-11-15:
Some of us ChimeraOS devs are working with ON/AOK on a kernel module for this EC to expose fan controls. This button will also be exposable through writing a value to the EC. An updated BIOS will be required for older models but release is TBD. They haven't been responding to us lately however. Seems they took some of our advice at least for the new models, but It is entirely dependent on ON/AOK to provide us more information to finish this effort. We'll put this in the "we'll see" bin.

pastaq, 2022-11-16:
Can you test this branch?

https://github.com/ShadowBlip/HandyGCCS/tree/aokzoe

NikhilSaini38, 2022-11-16:
> Can you test this branch?
> 
> https://github.com/ShadowBlip/HandyGCCS/tree/aokzoe

Sure thing, what should i be looking for? because current release is working fairly fine too, other than showing 2 xbox controllers that is.

pastaq, 2022-11-16:
In GamepadUI the yellow button short press should open QAM, long press the steam menu, kb the keyboard, and yellow + kb should active the gyro. Every other button should work as normal.

NikhilSaini38, 2022-11-16:
> In GamepadUI the yellow button short press should open QAM, long press the steam menu, kb the keyboard, and yellow + kb should active the gyro. Every other button should work as normal.

Yellow button short press triggers QAM but also occassinally trigger whatever is highlighted in the either on QAM or main UI depending on whether QAM is opening or closing, long press of the same opens steam menu, but my impatient ass is unable time it right. KB button opens OSK but if you tap it quickly, it goes in mouse emulation mode (firmware issue i guess).

Cannot figure how to see gyro is enabled, it doesnt show in steam input overlay. @pastaq

pastaq, 2022-11-16:
The mouse mode is concerning because that's only supposed to happen on long press of KB. I will attempt to contact AOK about this. If possible, could you get a video of that happening so I can send it to them?

For the occasional A button passthrough on QAM, can you try adjusting the value in the following line up/down in .01 increments until you find the value that does it the least?

https://github.com/ShadowBlip/HandyGCCS/blob/a88026af0dd99c63325f7859ec9ac140fbff2d65/handycon.py#L154

For gyro, any game where the camera is controlled by the right stick will work. It should also do a single rumble for activate and double rumble for deactivate which you would feel. QAM should also do a rumble when it is pressed.

If you have any games that have a rumble effect like a racing game, that would be a good thing to test as well.

NikhilSaini38, 2022-11-18:
I clean installed the holoiso again yesterday, ran tests on handygccs aokzoe branch

yellow short press: QAM + rumble
In game rumble: Present
Gyro mode (Yellow + KB) : not working, occasionally if i try to do it the device starts rumbling non stop have to press yellow+KB many times to make it stop
KB -> OSK works in desktop mode, not in gamepadui (installed with fresh build of holoiso_install_main from PKGBUILD)

pastaq, 2022-11-18:
> I clean installed the holoiso (installed with fresh build of [holoiso_install_main](https://github.com/theVakhovskeIsTaken/holoiso_instal
> Gyro mode (Yellow + KB) : not working, occasionally if i try to do it the device starts rumbling non stop, I have to press yellow+KB many times to make it stop

I've had this happen before when testing other devices. I'll try seeing if I can replicate this behavior on my OXP Mini and maybe come up with a solution. Could be a couple of things causing that.

> KB -> OSK works in desktop mode, not in gamepadui

Hmmm... Might be a problem with the kb button codes. Seems like it's still sending the original key code. However, the above issue also uses that key.

Can you tell me what the output of the Handheld Controller is when running `evtest` for each of these keys/combos.

> `BUTTON_DELAY = 0.08` resolves the A button passthrough issue.

Great, thanks for testing this. I'll add it to the PR.

NikhilSaini38, 2022-11-18:
output from evtest (Desktop mode, holoiso 3.4, Handygccs service installed and running). UEFI - secure boot enabled with standard settings, fastboot enabled with network configuration, and redirect turned on.
```
[deck@steamos ~]$ evtest /dev/input/event17 
Input driver version is 1.0.1
Input device ID: bus 0x3 vendor 0x45e product 0x28e version 0x110
Input device name: "Handheld Controller"
Supported events:
  Event type 0 (EV_SYN)
  Event type 1 (EV_KEY)
    Event code 1 (KEY_ESC)
    Event code 2 (KEY_1)
    Event code 3 (KEY_2)
    Event code 4 (KEY_3)
    Event code 5 (KEY_4)
    Event code 6 (KEY_5)
    Event code 7 (KEY_6)
    Event code 8 (KEY_7)
    Event code 9 (KEY_8)
    Event code 10 (KEY_9)
    Event code 11 (KEY_0)
    Event code 12 (KEY_MINUS)
    Event code 13 (KEY_EQUAL)
    Event code 14 (KEY_BACKSPACE)
    Event code 15 (KEY_TAB)
    Event code 16 (KEY_Q)
    Event code 17 (KEY_W)
    Event code 18 (KEY_E)
    Event code 19 (KEY_R)
    Event code 20 (KEY_T)
    Event code 21 (KEY_Y)
    Event code 22 (KEY_U)
    Event code 23 (KEY_I)
    Event code 24 (KEY_O)
    Event code 25 (KEY_P)
    Event code 26 (KEY_LEFTBRACE)
    Event code 27 (KEY_RIGHTBRACE)
    Event code 28 (KEY_ENTER)
    Event code 29 (KEY_LEFTCTRL)
    Event code 30 (KEY_A)
    Event code 31 (KEY_S)
    Event code 32 (KEY_D)
    Event code 33 (KEY_F)
    Event code 34 (KEY_G)
    Event code 35 (KEY_H)
    Event code 36 (KEY_J)
    Event code 37 (KEY_K)
    Event code 38 (KEY_L)
    Event code 39 (KEY_SEMICOLON)
    Event code 40 (KEY_APOSTROPHE)
    Event code 41 (KEY_GRAVE)
    Event code 42 (KEY_LEFTSHIFT)
    Event code 43 (KEY_BACKSLASH)
    Event code 44 (KEY_Z)
    Event code 45 (KEY_X)
    Event code 46 (KEY_C)
    Event code 47 (KEY_V)
    Event code 48 (KEY_B)
    Event code 49 (KEY_N)
    Event code 50 (KEY_M)
    Event code 51 (KEY_COMMA)
    Event code 52 (KEY_DOT)
    Event code 53 (KEY_SLASH)
    Event code 54 (KEY_RIGHTSHIFT)
    Event code 55 (KEY_KPASTERISK)
    Event code 56 (KEY_LEFTALT)
    Event code 57 (KEY_SPACE)
    Event code 58 (KEY_CAPSLOCK)
    Event code 59 (KEY_F1)
    Event code 60 (KEY_F2)
    Event code 61 (KEY_F3)
[... 161 lines cut ...]

# Yellow long (triggers steam)

Event: time 1668761696.020247, type 1 (EV_KEY), code 316 (BTN_MODE), value 1
Event: time 1668761696.020247, -------------- SYN_REPORT ------------
Event: time 1668761696.020292, type 1 (EV_KEY), code 316 (BTN_MODE), value 0
Event: time 1668761696.020292, -------------- SYN_REPORT ------------

# triggers OSK in desktop mode

Event: time 1668761700.536027, type 1 (EV_KEY), code 316 (BTN_MODE), value 1
Event: time 1668761700.536027, -------------- SYN_REPORT ------------
Event: time 1668761700.606411, type 1 (EV_KEY), code 307 (BTN_NORTH), value 1
Event: time 1668761700.606411, -------------- SYN_REPORT ------------
Event: time 1668761700.676778, type 1 (EV_KEY), code 316 (BTN_MODE), value 0
Event: time 1668761700.676778, -------------- SYN_REPORT ------------
Event: time 1668761700.747046, type 1 (EV_KEY), code 307 (BTN_NORTH), value 0
Event: time 1668761700.747046, -------------- SYN_REPORT ------------

# |||| button doesn't respond to button press in any way

# Yellow + KB doesn't respond

# Yellow + |||| doesn't respond
```

pastaq, 2022-11-18:
That all looks normal at least. I'll pok around the code a bit and see if I can't reproduce the issue. Thanks a bunch.

NikhilSaini38, 2022-11-19:
@pastaq  i forgot to mention. KB doesn't trigger OSK in gamepadui session anymore in desktop mode it works fine.

pastaq, 2022-11-19:
I think that is a steam bug not responding to the key combo anymore. I tried with a standalone wireless XB360 controller and all the built in combos work except home + x, which is supposed to be the OSK. I also tried it on the Aya Neo 2021 with the same results. I'll submit a bug report to valve for that. I think the only remaining issue is the continuous rumble instead of gyro issue. I'm not sure how best to troubleshoot though. I need a lot more information about what is happening with the device when that happens, which normally I'd print a bunch of crap to the terminal at each stage to see if something is abnormal.

pastaq, 2022-11-19:
#48 for tracking

pastaq, 2022-11-19:
https://github.com/ValveSoftware/steam-for-linux/issues/8932 for tracking OSK issue, which won't hold up inclusion of this device.

pastaq, 2022-11-22:
I merged this into main. I think the gyro issue was caused by a mistake in the PKGBUILD not installing the gyro config properly. If there are furhter issues please open a new issue. Thanks for your assistance @NikhilSaini38.

NikhilSaini38, 2022-11-22:
@pastaq I manually removed all the related site-packages from github clone and built python-smbus2, bython-bmi160-i2c and handygccs-git from AUR. The issue with gyro still is there.

pastaq, 2022-11-22:
Well shoot. I think that may be the same firmware issue the OXP Mini Pro has. 

What is the output of the following command?

`sudo i2cdetect -y -r 1`

pastaq, 2022-11-22:
Hmm, it should have been in address 68. Another device had it installed on bus 2 address 69. Can you try `sudo i2cdetect -y -r 2` as well? That could be the entire issue.

NikhilSaini38, 2022-11-22:
@pastaq 
```
[deck@steamos ~]$ sudo i2cdetect -y -r 2
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:                         -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- 2b 2c -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- 44 -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- --                                  
```

pastaq, 2022-11-23:
Hmm, I imagine it is the same. I looked at the .inf for the windows driver and it appears to have the same ACPI ID as the BMI160 in the Aya Neo 2021's. In theory the modprobe for i2c_dev should have activated everything on the i2c bus.

What do you get for `sudo i2cdetect -l` ? 

for reference, the AMD Mini:
```
[gamer@chimeraos ~]$ sudo i2cdetect -l
i2c-0	i2c       	Synopsys DesignWare I2C adapter 	I2C adapter
i2c-1	i2c       	Synopsys DesignWare I2C adapter 	I2C adapter
i2c-2	smbus     	SMBus PIIX4 adapter port 0 at 0b00	SMBus adapter
i2c-3	smbus     	SMBus PIIX4 adapter port 2 at 0b00	SMBus adapter
i2c-4	smbus     	SMBus PIIX4 adapter port 1 at 0b20	SMBus adapter
i2c-5	i2c       	AMDGPU DM i2c hw bus 0          	I2C adapter
i2c-6	i2c       	AMDGPU DM i2c hw bus 1          	I2C adapter
i2c-7	i2c       	AMDGPU DM i2c hw bus 2          	I2C adapter
i2c-8	i2c       	AMDGPU DM aux hw bus 0          	I2C adapter
i2c-9	i2c       	AMDGPU DM aux hw bus 1          	I2C adapter
i2c-10	i2c       	AMDGPU DM aux hw bus 2          	I2C adapter
```
and Aya Neo 2021 Pro:
```
i2c-0	i2c       	Synopsys DesignWare I2C adapter 	I2C adapter
i2c-1	i2c       	Synopsys DesignWare I2C adapter 	I2C adapter
i2c-2	smbus     	SMBus PIIX4 adapter port 0 at 0b00	SMBus adapter
i2c-3	smbus     	SMBus PIIX4 adapter port 2 at 0b00	SMBus adapter
i2c-4	smbus     	SMBus PIIX4 adapter port 1 at 0b20	SMBus adapter
i2c-5	i2c       	AMDGPU DM i2c hw bus 0          	I2C adapter
i2c-6	i2c       	AMDGPU DM i2c hw bus 1          	I2C adapter
i2c-7	i2c       	AMDGPU DM i2c hw bus 2          	I2C adapter
i2c-8	i2c       	AMDGPU DM aux hw bus 0          	I2C adapter
i2c-9	i2c       	AMDGPU DM aux hw bus 1          	I2C adapter
i2c-10	i2c       	AMDGPU DM aux hw bus 2          	I2C adapter
```

also, what do you get when you `lsmod | grep i2c`?

NikhilSaini38, 2022-11-23:
> It might also be helpful to upload the file you get from running `udevadm info -t > udev_cap.txt`

invalid option -t

pastaq, 2022-11-23:
Ah shoot, you're on HoloISO right? I don't thing the -t option was added to systemd until a later version than is available on that OS.

NikhilSaini38, 2022-11-23:
any alternative?

> Ah shoot, you're on HoloISO right? I don't thing the -t option was added to systemd until a later version than is available on that OS.

danyi, 2022-11-24:
Now the short press opens QAM and the long press opens the menu
This results in the steamdeck combo being unavailable
For example, "Menu +R waist bar" is the mouse simulation, "menu LT" is the right mouse button, and so on
Can it be changed to be the same as the original one
A short press opens the menu, and a long press and steamdeck are prompt key combinations
The "Menu +A" button is a quick way to open QAM

pastaq, 2022-11-24:
Do the Mini Pro and the AZ A1 produce events on press or release? The older models all only did this on release which makes those key combo's difficult and has a tendency to break other key combos we want, like guide+kb. it will trigger a second "guide" on release and close the OSK.

danyi, 2022-11-24:
I have here the new controller firmware and bios from the authorities
Maybe you want to test it again with this latest one
[AOK-FW-Update-1.24bios-1.2.68EC-100W.zip](https://github.com/ShadowBlip/HandyGCCS/files/10081212/AOK-FW-Update-1.24bios-1.2.68EC-100W.zip)
[AOKZOE_A1-GA28-XBOX.zip](https://github.com/ShadowBlip/HandyGCCS/files/10081221/AOKZOE_A1-GA28-XBOX.zip)

danyi, 2022-11-24:
Theory can be used in ONEXPLAYER Mini Pro
But I don't have time to test it right now
About the turbo keys, their engineers said they don't plan to map them up
We need our own efforts to develop the corresponding drivers
They instruct you to access EC port 62/66

pastaq, 2022-11-26:
> About the turbo keys, their engineers said they don't plan to map them up We need our own efforts to develop the corresponding drivers They instruct you to access EC port 62/66

Yes, they added that at our request. The OXP platform driver will need to have this added, but we need some dedicated testers.

NikhilSaini38, 2022-11-27:
I am available for testing on aokzoe.

pastaq, 2022-11-27:
> I am available for testing on aokzoe.

Once the platform driver is ready I will update this post with instructions.

uplush, 2022-11-27:
I can test on aokzoe.

pastaq, 2022-11-28:
Basic support is added so I'm closing this issue. A new issue will be made for the updated turbo button via platform driver as that covers multiple devices.

closed 2022-11-28 by pastaq