ShadowBlip/HandyGCCS #58: [REQUEST] Anbernic Win600 support

theVakhovskeIsTaken, 2022-12-09:
```
System ID:
Win600

X-Box 360 Device:
Microsoft X-Box 360 pad | usb-0000:02:00.3-5/input0 | bustype: 3 vendor: 1118 product: 654 version: 272

keyboard Device:
AT Translated Set 2 keyboard | isa0060/serio0/input0 | bustype: 17 vendor: 1 product: 1 version: 43907

All Devices:
Microsoft X-Box 360 pad 0 |  | bustype: 3 vendor: 10462 product: 4607 version: 1
Microsoft X-Box 360 pad | usb-0000:02:00.3-5/input0 | bustype: 3 vendor: 1118 product: 654 version: 272
AT Translated Set 2 keyboard | isa0060/serio0/input0 | bustype: 17 vendor: 1 product: 1 version: 43907

Captured Keymaps:
[125] -- "Win Button" being pressed (Right side)
[34, 125] -- "Home" button being pressed (Left side)
```

Following request:
Map [34, 125] to normal home press, and [125] to QAM on gamepadui

pastaq, 2022-12-10:
Win600 support is actually done and will be included in the next major revision. If you'd like to use it now you can install this branch: https://github.com/ShadowBlip/HandyGCCS/pull/56

Win will bring up the main menu, home the QAM, kb the OSK (this is actually bugged in the current client), long press home will toggle performance mode if `ryzenadj-git` is installed, long press home will send and esc key, and long press KB will take a screenshot.

theVakhovskeIsTaken, 2022-12-12:
Mhm! Got it, thanks!

closed 2022-12-12 by theVakhovskeIsTaken

F1ReB4LL, 2023-08-26:
> Win will bring up the main menu, home the QAM, kb the OSK (this is actually bugged in the current client), long press home will toggle performance mode if `ryzenadj-git` is installed, long press home will send and esc key, and long press KB will take a screenshot.

I'm not sure why was such a weird set of actions chosen, but since the only linux build people are using on Win600 is HoloISO, it would be much more useful to see Home tied to Steam button (left menu on short press), Win tied to "..." button (right menu on short press) and KB tied to Steam + X (virtual keyboard on short press).