ShadowBlip/HandyGCCS #59: Gyro stutters

Alexmjacobs, 2022-12-10:
Hope I am doing this right, very new to this.

Installed on latest Holoiso on Aya Neo founder (1st IGG batch).

Used “esc” to enable and the gyro controls and they barely register or stutter and act unexpectedly in conjunction with the stick.

Is there a way tune this by adjusting one of the config files?

Had Gyro working relatively well in windows with Handheld Companion.

thank you so much creating this - button mapping is great.

pastaq, 2022-12-11:
Gyro could definitely use some work. There is a "gyro sensitivity" setting in the config file located at /etc/handygccs.conf that you can tweak. TBS I'm not an expert on the gyro stuff so I'm certainly looking for anyone who can improve on the current implementation.

Alexmjacobs, 2022-12-11:
Thank you so much for the info! I’ll take a poke around with that .conf file and see if that helps.Alex

Alexmjacobs, 2022-12-12:
> Gyro could definitely use some work. There is a "gyro sensitivity" setting in the config file located at /etc/handygccs.conf that you can tweak. TBS I'm not an expert on the gyro stuff so I'm certainly looking for anyone who can improve on the current implementation.

Alexmjacobs, 2022-12-12:
I did try sensitivity at 60 but had the same stuttering issues - Let me know if I can provide any more info.

Thanks.

pastaq, 2023-01-11:
I made a change that should improve gyro response. I was doing something that caused an unexpected delay for gyro but was necessary for other events. That has now been bypassed for gyro events. Latest AUR release includes the change. It will be present in the next ChimeraOS version.

Alexmjacobs, 2023-01-11:
Thank you much! Had gone back to windows, looking forward to giving this a try! Thank you for the update.Sent from my iPhone

closed 2023-01-31 by pastaq