ShadowBlip/HandyGCCS #60: Has ayaneo 2 been supported?

ixervice, 2022-12-11:
Hi, does HandyGCCS support Ayaneo 2 or not? Or what can I do to use HandyGCCS on Ayaneo 2? How  can I find its button mapping?

Thx a lot

pastaq, 2022-12-11:
They'll be sending me a dev unit here pretty quick, but I don't have an ETA yet. In the meantime you could run the `capture-system` script and post the output here. That will give me a head start.

ixervice, 2022-12-11:
Thx for your reply. As I’m not familiar with linux. Could you tell how to run the capture-system script in detail?  Thank you

pastaq, 2022-12-11:
Sure thing. What OS are you on?

ixervice, 2022-12-12:
Thank you.
I have installed Holoiso. So what should I do next? switch to desktop mode and open the Konsole, then?

pastaq, 2022-12-12:
Everything should be done with the console. Either from a desktop launch Konsole or use a TTY by pressing `CRTL+ALT+F3` 
Alternatively, set up SSH to another machine. That would be easiest and I'd recommend it. You'll need a keyboard attached if using the Aya directly

make sure git is installed
`sudo pacman -Sy git`

clone this repository:
`git clone https://github.com/ShadowBlip/HandyGCCS.git`

run the script
`cd HandyGCCS`
`sudo python capture-system.py`

if it detects your controller and keyboard automatically it will ask you to press the face buttons. Please press them in this order:
- Aya Button (Big face button)
- program button (small face button)
- LC (top left next to left bumper)
- RC (top right nect to right bumper)
- LC + RC together

Press `CTRL+C` to end and upload the generated `capture-file.txt`

ixervice, 2022-12-13:
Thank you.
Here is the progress content when I run the script

(steam@holoiso HandyGCCS)$ sudo python capture-system.py 
Scanning system and creating device profile.
Gathering system info...
Successfully identified compatible controllers. Press each non-functioning button in succession. When complete press ctrl+c to end capture.
/home/steam/HandyGCCS/capture-system.py:169: DeprecationWarning: There is no current event loop
  asyncio.ensure_future(capture_events(xb360))
/home/steam/HandyGCCS/capture-system.py:170: DeprecationWarning: There is no current event loop
  asyncio.ensure_future(capture_events(keybd))
/home/steam/HandyGCCS/capture-system.py:172: DeprecationWarning: There is no current event loop
  loop = asyncio.get_event_loop()
[] [97] event at 1670895658.113982, code 04, type 04, val 157
[] [97] event at 1670895658.113982, code 97, type 01, val 01
[97] [97] event at 1670895658.113982, code 00, type 00, val 00
Identified new keymap:  [97]
[] [97, 125] event at 1670895658.118684, code 04, type 04, val 219
[] [97, 125] event at 1670895658.118684, code 125, type 01, val 01
[97, 125] [97, 125] event at 1670895658.118684, code 00, type 00, val 00
Identified new keymap:  [97, 125]
[] [97, 125] event at 1670895658.121696, code 04, type 04, val 104
[] [97, 125] event at 1670895658.121696, code 00, type 00, val 00
[] [97, 125] event at 1670895658.345451, code 04, type 04, val 104
[] [97, 125] event at 1670895658.345451, code 00, type 00, val 00
[] [97] event at 1670895658.353251, code 04, type 04, val 219
[] [97] event at 1670895658.353251, code 125, type 01, val 00
[] [97] event at 1670895658.353251, code 00, type 00, val 00
[] [125] event at 1670895659.979224, code 04, type 04, val 219
[] [125] event at 1670895659.979224, code 125, type 01, val 01
[125] [125] event at 1670895659.979224, code 00, type 00, val 00
Identified new keymap:  [125]
[] [32, 125] event at 1670895659.981915, code 04, type 04, val 32
[] [32, 125] event at 1670895659.981915, code 32, type 01, val 01
[32, 125] [32, 125] event at 1670895659.981915, code 00, type 00, val 00
Identified new keymap:  [32, 125]
[] [32] event at 1670895660.307709, code 04, type 04, val 219
[] [32] event at 1670895660.307709, code 125, type 01, val 00
[] [32] event at 1670895660.307709, code 00, type 00, val 00
[] [125] event at 1670895662.175535, code 04, type 04, val 219
[] [125] event at 1670895662.175535, code 125, type 01, val 01
[125] [125] event at 1670895662.175535, code 00, type 00, val 00
[125] [32, 125] event at 1670895662.178689, code 04, type 04, val 32
[125] [32, 125] event at 1670895662.178689, code 32, type 01, val 01
[32, 125] [32, 125] event at 1670895662.178689, code 00, type 00, val 00
[32, 125] [32] event at 1670895662.379735, code 04, type 04, val 219
[32, 125] [32] event at 1670895662.379735, code 125, type 01, val 00
[32, 125] [32] event at 1670895662.379735, code 00, type 00, val 00
[32, 125] [97] event at 1670895691.600348, code 04, type 04, val 157
[32, 125] [97] event at 1670895691.600348, code 97, type 01, val 01
[97] [97] event at 1670895691.600348, code 00, type 00, val 00
[97] [97, 125] event at 1670895691.605044, code 04, type 04, val 219
[97] [97, 125] event at 1670895691.605044, code 125, type 01, val 01
[97, 125] [97, 125] event at 1670895691.605044, code 00, type 00, val 00
[97, 125] [97, 125] event at 1670895691.608077, code 04, type 04, val 102
[97, 125] [97, 125] event at 1670895691.608077, code 00, type 00, val 00
[97, 125] [97, 125] event at 1670895691.935202, code 04, type 04, val 102
[97, 125] [97, 125] event at 1670895691.935202, code 00, type 00, val 00
[97, 125] [97] event at 1670895691.943488, code 04, type 04, val 219
[97, 125] [97] event at 1670895691.943488, code 125, type 01, val 00
[97, 125] [97] event at 1670895691.943488, code 00, type 00, val 00
[97, 125] [97] event at 1670895693.065833, code 04, type 04, val 157
[97, 125] [97] event at 1670895693.065833, code 97, type 01, val 01
[97] [97] event at 1670895693.065833, code 00, type 00, val 00
[97] [97, 125] event at 1670895693.070152, code 04, type 04, val 219
[97] [97, 125] event at 1670895693.070152, code 125, type 01, val 01
[97, 125] [97, 125] event at 1670895693.070152, code 00, type 00, val 00
[97, 125] [97, 125] event at 1670895693.073218, code 04, type 04, val 103
[97, 125] [97, 125] event at 1670895693.073218, code 00, type 00, val 00
[97, 125] [97, 125] event at 1670895693.307020, code 04, type 04, val 103
[... 66 lines cut ...]
MOSART Semi. wireless dongle System Control | usb-0000:e4:00.4-2.3/input1 | bustype: 3 vendor: 6785 product: 4610 version: 272
MOSART Semi. wireless dongle Consumer Control | usb-0000:e4:00.4-2.3/input1 | bustype: 3 vendor: 6785 product: 4610 version: 272
MOSART Semi. wireless dongle Mouse | usb-0000:e4:00.4-2.3/input1 | bustype: 3 vendor: 6785 product: 4610 version: 272
MOSART Semi. wireless dongle | usb-0000:e4:00.4-2.3/input0 | bustype: 3 vendor: 6785 product: 4610 version: 272
MI Dongle MI Wireless Mouse Keyboard | usb-0000:e4:00.4-2.1/input2 | bustype: 3 vendor: 10007 product: 59 version: 273
MI Dongle MI Wireless Mouse Consumer Control | usb-0000:e4:00.4-2.1/input2 | bustype: 3 vendor: 10007 product: 59 version: 273
MI Dongle MI Wireless Mouse Consumer Control | usb-0000:e4:00.4-2.1/input1 | bustype: 3 vendor: 10007 product: 59 version: 273
MI Dongle MI Wireless Mouse | usb-0000:e4:00.4-2.1/input1 | bustype: 3 vendor: 10007 product: 59 version: 273
[... cut ...]
PC Speaker | isa0061/input0 | bustype: 16 vendor: 31 product: 1 version: 256
Video Bus | LNXVIDEO/video/input0 | bustype: 25 vendor: 0 product: 6 version: 0
Goodix Capacitive TouchScreen | input/ts | bustype: 24 vendor: 1046 product: 911 version: 4192
AT Translated Set 2 keyboard | isa0060/serio0/input0 | bustype: 17 vendor: 1 product: 1 version: 43907
Power Button | LNXPWRBN/button/input0 | bustype: 25 vendor: 0 product: 1 version: 0
Power Button | PNP0C0C/button/input0 | bustype: 25 vendor: 0 product: 1 version: 0

Captured Keymaps:
[97]
[97, 125]
[125]
[32, 125]

sshzz1, 2022-12-15:
we need ayaneo2！

ixervice, 2022-12-22:
> @ixervice these captures seem to be missing some button presses. Did you press all four buttons plus the LC/RC at the same time?

Yes， I did what you said. But it seems neither LC or RC has been captured.

pastaq, 2022-12-22:
It also seems to be reporting different values than I expected. A rep from Aya Neo gave me the codes a while back and they should be.
LC:  Ctrl+Win+F15
RC: Ctrl+Win+F16
AYA: Ctrl+Win+F17

ixervice, 2022-12-23:
Then what should I do next to help you？Or just wait until you get your AYANEO 2 dev unit？

pastaq, 2022-12-23:
Is it possible to get a clean version of the capture file with no other devices plugged in? Essentially I'd like to see the input devices list with only the native hardware included.

danyi, 2022-12-23:
System ID:
AYANEO 2

X-Box 360 Device:
Microsoft X-Box 360 pad | usb-0000:e4:00.3-4/input0 | bustype: 3 vendor: 1118 product: 654 version: 272

keyboard Device:
AT Translated Set 2 keyboard | isa0060/serio0/input0 | bustype: 17 vendor: 1 product: 1 version: 43907

All Devices:
Microsoft X-Box 360 pad 0 |  | bustype: 3 vendor: 10462 product: 4607 version: 1
Microsoft X-Box 360 pad | usb-0000:e4:00.3-4/input0 | bustype: 3 vendor: 1118 product: 654 version: 272
HD-Audio Generic Headphone | ALSA | bustype: 0 vendor: 0 product: 0 version: 0
HD-Audio Generic Mic | ALSA | bustype: 0 vendor: 0 product: 0 version: 0
HD-Audio Generic HDMI/DP,pcm=7 | ALSA | bustype: 0 vendor: 0 product: 0 version: 0
HD-Audio Generic HDMI/DP,pcm=3 | ALSA | bustype: 0 vendor: 0 product: 0 version: 0
PC Speaker | isa0061/input0 | bustype: 16 vendor: 31 product: 1 version: 256
Goodix Capacitive TouchScreen | input/ts | bustype: 24 vendor: 1046 product: 911 version: 4192
Video Bus | LNXVIDEO/video/input0 | bustype: 25 vendor: 0 product: 6 version: 0
Logitech M720 Triathlon | usb-0000:e4:00.4-2.4/input2:1 | bustype: 3 vendor: 1133 product: 16478 version: 273
BY Tech Usb Gaming Keyboard Mouse | usb-0000:e4:00.4-2.3/input1 | bustype: 3 vendor: 9610 product: 22 version: 272
BY Tech Usb Gaming Keyboard | usb-0000:e4:00.4-2.3/input1 | bustype: 3 vendor: 9610 product: 22 version: 272
BY Tech Usb Gaming Keyboard Consumer Control | usb-0000:e4:00.4-2.3/input1 | bustype: 3 vendor: 9610 product: 22 version: 272
BY Tech Usb Gaming Keyboard System Control | usb-0000:e4:00.4-2.3/input1 | bustype: 3 vendor: 9610 product: 22 version: 272
BY Tech Usb Gaming Keyboard | usb-0000:e4:00.4-2.3/input0 | bustype: 3 vendor: 9610 product: 22 version: 272
AT Translated Set 2 keyboard | isa0060/serio0/input0 | bustype: 17 vendor: 1 product: 1 version: 43907
Power Button | LNXPWRBN/button/input0 | bustype: 25 vendor: 0 product: 1 version: 0
Power Button | PNP0C0C/button/input0 | bustype: 25 vendor: 0 product: 1 version: 0

Captured Keymaps:
[97]
[97, 125]
[85]
[125]
[32, 125]
[32, 97, 125]
[305]

pastaq, 2022-12-23:
> Captured Keymaps: [97] [97, 125] [85] [85, 125] [32, 125] [32, 97, 125]
> 
> It does seem a little strange All of a sudden the value of the key changes

Hmm, the [85] through [32, 97, 125] was a single key? Was it held long or just a tap? I'm wondering if it has a long press state change.

Aya said they will send the dev unit in 1-2 weeks, so if we can't figure it out by then I'll have more hands on with it.

Can you post the raw output of  `sudo evtest` for the AT Translated Set Keyboard? You may have to unlock and install it if using ChimeraOS.

danyi, 2022-12-23:
I have been trying evtest today
but i have a problem
evtest: /usr/lib/libc.so.6: version `GLIBC_2.34' not found (required by evtest)

pastaq, 2022-12-23:
Are you on HoloISO?

danyi, 2022-12-24:
> Are you on HoloISO?

yes

pastaq, 2022-12-24:
> > Are you on HoloISO?
> 
> yes I'm up, I'm going to be testing with ChimeraOS today

Ah, yeah the package management for HoloISO can get messed up pretty easily.

hoaxlayer, 2023-01-21:
> Thank you. Here is the progress content when I run the script
>
> (steam@holoiso HandyGCCS)$ sudo python capture-system.py Scanning system and creating device profile. Gathering system info... [...]

The file generated on my end looks like this as well

pastaq, 2023-01-26:
If anyone would like to test, I was able to find a way to differentiate the LC/RC/AYASPACE buttons.

https://github.com/ShadowBlip/HandyGCCS/pull/71

Clone the repo, check out aya2_dev, and run usr/share/handyccs/handycon.py as sudo.

ixervice, 2023-01-27:
> 

Thx a lot. It works like a charm.

closed 2023-02-28 by pastaq