ShadowBlip/HandyGCCS #61: Configuration request for GPDWIN4

danyi, 2022-12-13:
[capture_file.txt](https://github.com/ShadowBlip/HandyGCCS/files/10218825/capture_file.txt)
It also has 2 buttons on the back
But after pressing, nothing catches

danyi, 2022-12-17:
W4 does not require an OSB button
Look at this
[image]

pastaq, 2022-12-17:
> W4 does not require an OSB button Look at this [image]

That's pretty cool. I still need the information I requested above.

pastaq, 2022-12-17:
Also, the systemID was blank in the capture file. Please also provide outputs of the following
`ls /sys/devices/virtual/dmi/id/`
`sudo cat /sys/devices/virtual/dmi/id/*`

pastaq, 2022-12-17:
Track progress here: https://github.com/ShadowBlip/HandyGCCS/tree/win4

danyi, 2022-12-17:
I used sudo evtest to test
/dev/input/event11: Mouse for Windows
/dev/input/event12: Mouse for Windows
/dev/input/event19: Microsoft X-Box 360 pad
/dev/input/event20: Microsoft X-Box 360 pad 0
Still no output

danyi, 2022-12-17:
> 
this problem has been solved
The official gave me a new bios
What I got is the internal test version, the official retail version, I don't think there will be this problem

danyi, 2022-12-17:
> Also, the systemID was blank in the capture file. Please also provide outputs of the following `ls /sys/devices/virtual/dmi/id/` `sudo cat /sys/devices/virtual/dmi/id/*`

this problem has been solved
The official gave me a new bios
What I got is the internal test version, the official retail version, I don't think there will be this problem

danyi, 2022-12-17:
[dsdt.zip](https://github.com/ShadowBlip/HandyGCCS/files/10251026/dsdt.zip)

danyi, 2022-12-17:
> > Also, the systemID was blank in the capture file. Please also provide outputs of the following `ls /sys/devices/virtual/dmi/id/` `sudo cat /sys/devices/virtual/dmi/id/*`
> 
> this problem has been solved The official gave me a new bios What I got is the internal test version, the official retail version, I don't think there will be this problem I will test the new BIOS as soon as possible and give you the obtained ID

[root@SteamOS ~]# cat /sys/devices/virtual/dmi/id/product_name
G1618-04

pastaq, 2022-12-17:
> [dsdt.zip](https://github.com/ShadowBlip/HandyGCCS/files/10251026/dsdt.zip)

Damn. Same problem with the 10EC5820 instead of BMI0160. Perhaps you could tell them to change it? The same problem is on the WinMax2. If this address is wrong Linux cannot identify the gyro correctly. For your testing see this: https://wiki.archlinux.org/title/GPD_Win_Max#IMU_(Accelerometer_+_Gyro)

This could be fixed for both models and improve the experience quite a bit. Allows for screen rotation by the user if desired as well. Also see this, it could be added in the same update:
https://wiki.archlinux.org/title/GPD_Win_Max#Keyboard
do you have issues with the keyboard on the WIN4?

danyi, 2022-12-17:
ok i'll keep testing
The keyboard works perfectly fine
The screen does not need to rotate
The screen itself is horizontal, 1080P

pastaq, 2022-12-17:
> 

This is very strange. One of the two Mouse for Windows device should have done something. Can you check the other event devices to see if anything outputs keycodes?

> The keyboard works perfectly fine

Perhaps not if the rear buttons aren't working with any device. They are keyboard buttons after all.

danyi, 2022-12-20:
I tested it all, and there is still no output, but it can be used under win

pastaq, 2022-12-20:
We'll need to figure out why that's happening. Also need to ID the location of the gyro still. According to the DSDT it should be on i2c register 0x68. Can you run the following for me and post the output of the second command?

`sudo modprobe i2c_dev`
`sudo i2cdetect -y -r 1`

You'll need i2c-tools

danyi, 2022-12-20:
> > [dsdt.zip](https://github.com/ShadowBlip/HandyGCCS/files/10251026/dsdt.zip)
> 
> Damn. Same problem with the gyro being labeled 10EC5820 instead of BMI0160. Perhaps you could tell them to change it? The same problem is on the WinMax2. If this address is wrong Linux cannot identify the gyro correctly. For your testing see this: https://wiki.archlinux.org/title/GPD_Win_Max#IMU_(Accelerometer_+_Gyro)
> 
> This could be fixed for both models and improve the experience quite a bit. Allows for screen rotation by the user if desired as well. Also see this, it could be added in the same update: https://wiki.archlinux.org/title/GPD_Win_Max#Keyboard do you have issues with the keyboard on the WIN4?

Regarding the gyroscope, I asked the official staff, and they replied that the ID here is correct, and if it is changed, it will cause failure under Windows. The model of win4 is bmi260, and it depends on whether the Linux driver is compatible. He said that the Windows driver The driver is matched based on this ID, and Linux is also matched by this ID, not by the model name

danyi, 2022-12-20:
> We'll need to figure out why that's happening. Also need to ID the location of the gyro still. According to the DSDT it should be on i2c register 0x68. Can you run the following for me and post the [...]

 sudo i2cdetect -y -r 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:                         -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- 68 -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- --

pastaq, 2022-12-20:
> Regarding the gyroscope, I asked the official staff, and they replied that the ID here is correct, and if it is changed, it will cause failure under Windows. The model of win4 is bmi0260, and it depends on whether the Linux driver is compatible. He said that the Windows driver The driver is matched based on this ID, and Linux is also matched by this ID, not by the model name bmi0260 is more accurate than bmi0160

Well that's just not true. Aya Neo's driver supports BMI0160, BMI0260, and the incorrect 10EC5280 for legacy support of the old versions. They were the first with this issue.

Download link:
https://cdn.ayaneo.com/ayaneo/downloads/UMDF2.0_BMI260_V1.0.23_5ID_signed_20H1.zip

You can see the list of support in the bmi160.inf file

10EC5280 is a Realtek ACPI ID that was initially used incorrectly by AyaNeo on the founders/2021 models. They have since changed to the correct ACPI ID of BMI0160, but the driver supports both

The issue is that it was attempted to add this ACPI ID to the Linux driver back in 2021 but it never happened because it's a Realtek ID, so it will never work in Linux unless changed. It will work in windows if the correct driver is used (as above) and also Linux.

See this thread from when they discuss adding this ID:

https://lore.kernel.org/all/CACAwPwatbMjd9CLdFCWnND3FqT5ZtPJqEVC-Ohze+GoJ6ODqqw@mail.gmail.com/

Without this change it will not work right in Linux (Same for GPD WinMax2 and Aya Neo Founders/2021) but will work in windows with the correct driver.

ixervice, 2023-03-11:
It seems the volume + and - cannot work properly in Holoiso, is there any plan to correct it?

pastaq, 2023-03-11:
> It seems the volume + and - cannot work properly in Holoiso, is there any plan to correct it?

Honestly I have no idea. These devices are pretty unique in how they work. Unless GPD send me one so I can get some hands on experience with it, it's unlikely this will move forward.

pastaq, 2023-04-02:
Some progress has been made with the back button keys. A udev rule that activates keyboard keys allows the "Mouse for Windows" devices to output keycodes.
`ACTION=="add|change", KERNEL=="event[0-9]*", ATTRS{idVendor}=="2f24", ATTRS{idProduct}=="0135", ENV{ID_INPUT_KEYBOARD}="1"`

Right Back:
Event: time 1680369326.921198, type 1 (EV_KEY), code 119 (KEY_PAUSE), value 1
Event: time 1680369326.921198, -------------- SYN_REPORT ------------
Event: time 1680369327.176241, type 1 (EV_KEY), code 119 (KEY_PAUSE), value 2
Event: time 1680369326.921198, -------------- SYN_REPORT ------------
Event: time 1680369327.176241, type 1 (EV_KEY), code 119 (KEY_PAUSE), value 0

Left Back:
Event: time 1680369326.921198, type 1 (EV_KEY), code 99 (KEY_SYSREQ), value 1
Event: time 1680369326.921198, -------------- SYN_REPORT ------------
Event: time 1680369327.176241, type 1 (EV_KEY), code 999 (KEY_SYSREQ), value 2
Event: time 1680369326.921198, -------------- SYN_REPORT ------------
Event: time 1680369327.176241, type 1 (EV_KEY), code 999 (KEY_SYSREQ), value 0

pastaq, 2023-04-02:
#74 Also uses the same "Mouse for windows" device and can add support for Win3 devices if this works.

pastaq, 2023-04-02:
I: Bus=0019 Vendor=0000 Product=0001 Version=0000
N: Name="Power Button"
P: Phys=PNP0C0C/button/input0
S: Sysfs=/devices/LNXSYSTM:00/LNXSYBUS:00/PNP0C0C:00/input/input0
U: Uniq=
H: Handlers=kbd event0 
B: PROP=0
B: EV=3
B: KEY=10000000000000 0

I: Bus=0019 Vendor=0000 Product=0001 Version=0000
N: Name="Power Button"
P: Phys=LNXPWRBN/button/input0
S: Sysfs=/devices/LNXSYSTM:00/LNXPWRBN:00/input/input1
U: Uniq=
H: Handlers=kbd event1 
B: PROP=0
B: EV=3
B: KEY=10000000000000 0

I: Bus=0018 Vendor=0416 Product=038f Version=1060
N: Name="Goodix Capacitive TouchScreen"
P: Phys=input/ts
S: Sysfs=/devices/platform/AMDI0010:03/i2c-2/i2c-GDIX1002:00/input/input3
U: Uniq=
H: Handlers=kbd event2 mouse0 
B: PROP=2
B: EV=b
B: KEY=400 0 0 0 2000000000000001 f800000000000000
B: ABS=265800000000003

I: Bus=0019 Vendor=0000 Product=0006 Version=0000
N: Name="Video Bus"
P: Phys=LNXVIDEO/video/input0
S: Sysfs=/devices/LNXSYSTM:00/LNXSYBUS:00/PNP0A08:00/device:1e/LNXVIDEO:00/input/input5
U: Uniq=
H: Handlers=kbd event3 
B: PROP=0
B: EV=3
B: KEY=3e000b00000000 0 0 0

I: Bus=0010 Vendor=001f Product=0001 Version=0100
N: Name="PC Speaker"
P: Phys=isa0061/input0
S: Sysfs=/devices/platform/pcspkr/input/input6
U: Uniq=
H: Handlers=kbd event4 
B: PROP=0
B: EV=40001
B: SND=6

I: Bus=0003 Vendor=258a Product=000c Version=0110
N: Name="HAILUCK CO.,LTD USB KEYBOARD"
P: Phys=usb-0000:73:00.3-3/input0
S: Sysfs=/devices/pci0000:00/0000:00:08.1/0000:73:00.3/usb2/2-3/2-3:1.0/0003:258A:000C.0001/input/input7
U: Uniq=
H: Handlers=sysrq kbd leds event5 
B: PROP=0
B: EV=120013
B: KEY=1000000000007 ff9f207ac14057ff febeffdfffefffff fffffffffffffffe
B: MSC=10
B: LED=7

I: Bus=0003 Vendor=258a Product=000c Version=0110
N: Name="HAILUCK CO.,LTD USB KEYBOARD Mouse"
P: Phys=usb-0000:73:00.3-3/input1
S: Sysfs=/devices/pci0000:00/0000:00:08.1/0000:73:00.3/usb2/2-3/2-3:1.1/0003:258A:000C.0002/input/input8
U: Uniq=
H: Handlers=event6 mouse1 
B: PROP=0
B: EV=17
B: KEY=70000 0 0 0 0
B: REL=903
B: MSC=10

I: Bus=0003 Vendor=258a Product=000c Version=0110
N: Name="HAILUCK CO.,LTD USB KEYBOARD System Control"
P: Phys=usb-0000:73:00.3-3/input1
S: Sysfs=/devices/pci0000:00/0000:00:08.1/0000:73:00.3/usb2/2-3/2-3:1.1/0003:258A:000C.0002/input/input9
U: Uniq=
H: Handlers=kbd event7 
B: PROP=0
B: EV=13
B: KEY=c000 10000000000000 0
B: MSC=10

I: Bus=0003 Vendor=258a Product=000c Version=0110
N: Name="HAILUCK CO.,LTD USB KEYBOARD Consumer Control"
P: Phys=usb-0000:73:00.3-3/input1
S: Sysfs=/devices/pci0000:00/0000:00:08.1/0000:73:00.3/usb2/2-3/2-3:1.1/0003:258A:000C.0002/input/input10
U: Uniq=
H: Handlers=kbd event8 
B: PROP=0
B: EV=13
B: KEY=28000000000 0 0 1000302000000 3878d8011001 e000000000000 0
B: MSC=10

I: Bus=0003 Vendor=258a Product=000c Version=0110
N: Name="HAILUCK CO.,LTD USB KEYBOARD Wireless Radio Control"
P: Phys=usb-0000:73:00.3-3/input1
S: Sysfs=/devices/pci0000:00/0000:00:08.1/0000:73:00.3/usb2/2-3/2-3:1.1/0003:258A:000C.0002/input/input11
U: Uniq=
H: Handlers=kbd event9 rfkill 
B: PROP=0
B: EV=13
B: KEY=80000000000000 0 0 0
B: MSC=10

I: Bus=0000 Vendor=0000 Product=0000 Version=0000
N: Name="HD-Audio Generic HDMI/DP,pcm=3"
P: Phys=ALSA
S: Sysfs=/devices/pci0000:00/0000:00:08.1/0000:73:00.1/sound/card0/input12
U: Uniq=
H: Handlers=event10 
B: PROP=0
B: EV=21
B: SW=140

I: Bus=0000 Vendor=0000 Product=0000 Version=0000
N: Name="HD-Audio Generic HDMI/DP,pcm=7"
P: Phys=ALSA
S: Sysfs=/devices/pci0000:00/0000:00:08.1/0000:73:00.1/sound/card0/input13
U: Uniq=
H: Handlers=event11 
B: PROP=0
B: EV=21
B: SW=140

I: Bus=0000 Vendor=0000 Product=0000 Version=0000
N: Name="HD-Audio Generic HDMI/DP,pcm=8"
P: Phys=ALSA
S: Sysfs=/devices/pci0000:00/0000:00:08.1/0000:73:00.1/sound/card0/input14
U: Uniq=
H: Handlers=event12 
B: PROP=0
B: EV=21
B: SW=140

I: Bus=0000 Vendor=0000 Product=0000 Version=0000
N: Name="HD-Audio Generic HDMI/DP,pcm=9"
P: Phys=ALSA
S: Sysfs=/devices/pci0000:00/0000:00:08.1/0000:73:00.1/sound/card0/input15
U: Uniq=
H: Handlers=event13 
B: PROP=0
B: EV=21
B: SW=140

I: Bus=0000 Vendor=0000 Product=0000 Version=0000
N: Name="HD-Audio Generic Mic"
P: Phys=ALSA
S: Sysfs=/devices/pci0000:00/0000:00:08.1/0000:73:00.6/sound/card1/input16
U: Uniq=
H: Handlers=event14 
B: PROP=0
B: EV=21
B: SW=10

I: Bus=0000 Vendor=0000 Product=0000 Version=0000
N: Name="HD-Audio Generic Front Headphone"
P: Phys=ALSA
S: Sysfs=/devices/pci0000:00/0000:00:08.1/0000:73:00.6/sound/card1/input17
U: Uniq=
H: Handlers=event15 
B: PROP=0
B: EV=21
B: SW=4

I: Bus=0003 Vendor=2f24 Product=0135 Version=0110
N: Name="  Mouse for Windows"
P: Phys=usb-0000:73:00.3-4.2/input0
S: Sysfs=/devices/pci0000:00/0000:00:08.1/0000:73:00.3/usb2/2-4/2-4.2/2-4.2:1.0/0003:2F24:0135.0003/input/input18
U: Uniq=
H: Handlers=event16 mouse2 
B: PROP=0
B: EV=17
B: KEY=1f0000 0 0 0 0
B: REL=903
B: MSC=10

I: Bus=0003 Vendor=2f24 Product=0135 Version=0110
N: Name="  Mouse for Windows"
P: Phys=usb-0000:73:00.3-4.2/input1
S: Sysfs=/devices/pci0000:00/0000:00:08.1/0000:73:00.3/usb2/2-4/2-4.2/2-4.2:1.1/0003:2F24:0135.0004/input/input19
U: Uniq=
H: Handlers=sysrq kbd leds event17 
B: PROP=0
B: EV=120013
B: KEY=1000000000007 ff9f207ac14057ff febeffdfffefffff fffffffffffffffe
B: MSC=10
B: LED=7

pastaq, 2023-04-02:
@danyi can you ask your GPD contact about the device hardware that should be activated from the volume keys? It doesn't appear to work in Linux. The win3 has a gpio_keys device that isn't present on the win4. I need to know if the device for win4 is detected in the above list of input devices, and if not, the specifics. They can contact me at derekjohn.clark@gmail.com if they'd like to respond directly. Thanks.

pastaq, 2023-04-08:
I've added the necessary codes to this PR: #72 
If anyone is available to test, checkout the `win4` branch and run `sudo ./install.sh`

danyi, 2023-04-15:
> @danyi can you ask your GPD contact about the device hardware that should be activated from the volume keys? It doesn't appear to work in Linux. The win3 has a gpio_keys device that isn't present on the win4. I need to know if the device for win4 is detected in the above list of input devices, and if not, the specifics. They can contact me at [derekjohn.clark@gmail.com](mailto:derekjohn.clark@gmail.com) if they'd like to respond directly. Thanks.

ok

pastaq, 2023-04-16:
> ok

Just FYI, we've made some progress here. It seems in kernel 5.19 the volume keys work but 6.1+ they detect as "PC Speaker". We're still trying to figure out why.

azq8oz, 2023-04-16:
> > ok
> 
> Just FYI, we've made some progress here. It seems in kernel 5.19 the volume keys work but 6.1+ they detect as "PC Speaker". We're still trying to figure out why.

Hello i have win 4 how i can install it. I tried volume keys on gaming mode for last stable version and it’s working with out handygccs.

danyi, 2023-04-22:
win4" volume button problem has been solved
Need to update "bios" to 3.0.3
match this command
"usermod -a -G input deck"
Elevate the privileges of the deck username

pastaq, 2023-04-24:
> win4" volume button problem has been solved Need to update "bios" to 3.0.3 match this command "usermod -a -G input deck" Elevate the privileges of the deck username

Notes for other users: replace "deck" with your username. This is only necessary if your user is not already in the "input" group, which is abnormal,

closed 2023-04-24 by pastaq

ghost573, 2023-04-26:
Updated bios to v3.0.4 and it's half solved.
On a fresh boot, volume works correctly, once device sleeps and wakes, volume stops working.
This is on normal arch 6.2.12 wayland.

pastaq, 2023-04-28:
> Updated bios to v3.0.4 and it's half solved. On a fresh boot, volume works correctly, once device sleeps and wakes, volume stops working. This is on vanilla arch 6.2.12.

This is true, but its beyond the scope of HandyGCCS. GPD Will need to solve this with another BIOS update.