ShadowBlip/HandyGCCS #62: Gyro as actual gyro ("virtual DS4" controller) possible?

jibsaramnim, 2022-12-15:
I just gave your project a try on my Win Max 2 running Fedora Silverblue. I had to make some changes here and there to get it to run (some IDs are different -- but different subject). Once I had it running, I realized that the currently implemented gyro support basically behaves like right joystick inputs. I was wondering; would it be possible for HandyGCCS to work in such a way that it basically behaves like a DualShock 4 controller, with "native" gyro support? If so, what do you think that would entail? 

Thank you!

pastaq, 2022-12-15:
> 

It's certainly possible. I'm hoping someone with more expertise in gyro will be willing to submit a change for that as better gyro support is frequently requested.

Can you explain what's going on with the ID's in a new issue? That shouldn't be occurring.

jibsaramnim, 2022-12-15:
> I'm hoping someone with more expertise in gyro will be willing to submit a change for that as better gyro support is frequently requested.

Ah, got it. Do you know if there's already any Python examples that might be worth looking into, or any information on how best to expose this information to the system for applications like Cemu to be able to make use of? I know that Cemu (even under Linux) natively supports a DS4 controller's gyro, but I am not sure how this is exposed and picked upon. 

> Can you explain what's going on with the ID's in a new issue? That shouldn't be occurring.

I could, but I am not sure if these differences are due to me using a different Linux flavor. This project is (right now) solely focused on Arch-like environments, I had to modify the installation script too. If there's plans to broaden support beyond Arch-like distros and you think it'll be useful to share these bits, I'm more than happy to do so :).

pastaq, 2022-12-15:
> 

I'm aware of some C programs for windows that manage to emulate a DS controller using a similar method. Tricking the system into thinking it is one is as simple as changing the virtual device attributes to a DS controller. The interaction of the gyro and how that should be applied, plus any methods or events it is expecting to be able to respond to are unknown to me at this time. If you're interested take a look at https://github.com/Valkirie/HandheldCompanion

As for other distro support, I'm all for it. Shouldn't be difficult. I think you may be the first user outside of an arch derivative so if you provide the install script differences I should be able to detect the host OS and manage it automatically.

As far as the device ID's go they should be pretty universal as that stuff is detected by the kernel drivers. If there's variation in a model then that's something I'd like to be aware of either way. If there is an OS specific reason then I'll need that info to get it working as well, but I won't really be able to determine it as one or the other unless I know what was different.

jibsaramnim, 2022-12-16:
> I'm aware of some C programs for windows that manage to emulate a DS controller using a similar method. Tricking the system into thinking it is one is as simple as changing the virtual device attributes to a DS controller.

It seems like `python-evdev` supports _reading_ gyro values of a DS4 controller (both absolute and relative values). Though as things are abstracted by this package, I am not sure how helpful it will be to help us create this as output in a way that other tools would expect. It might be wise to look at these other solutions and try to make sense of them, even though they're written in other languages yeah.

> As far as the device ID's go they should be pretty universal as that stuff is detected by the kernel drivers. If there's variation in a model then that's something I'd like to be aware of either way. 

I have pushed some of the (decidedly incomplete) changes I made to get things running on my system. The `install.sh` changes in particular are not complete, as I ended up just doing things manually instead as I was juggling between doing things directly on my host system as-well as within `toolbox` containers (I'm running Fedora Silverblue, which has an immutable root drive for the most part). 

I ran out of time to do anything more for this right now, but perhaps sharing this as-is might be of some help to you already as it includes the different devices IDs I had to have it look for on my particular system. It's [on my fork, right here](https://github.com/jibsaramnim/HandyGCCS/commit/1783ffc8be407317485e0ebff9a2191defebb124)

pastaq, 2022-12-16:
Interesting. Seems that some WM2 have the same addresses as the WIN4. Once I finish adding support for that device it should work fine.

I'll look into improving the installer as well. One of the initiatives we have is installing it as shipping as a python module, so that may be the best bet.

pastaq, 2023-05-08:
I haven't really looked into this at all yet, but you probably need to update the udev rules. I'm in the middle of the [refactor](https://github.com/ShadowBlip/HandyGCCS/tree/full_refactor), and I need to look at using the native sysfs file descriptors instead of relying on the bmi160_i2c module. After that is done it might be possible. That being said, HandyGCCS is transforming from the endpoint for controller support to the prototyping area for new devices when looking to add support for them into OpenGamepadUI. I'm not sure how much I'm going to be adding new features like that. I'm more than happy to take PR's though.

pastaq, 2023-07-17:
I'm pulling gyro support for now as it is in a really fractured state in Linux. If/when it gets added again I'll revisit  this idea.

closed 2023-07-17 by pastaq