ShadowBlip/HandyGCCS #63: Ship as a python module

Samsagax, 2022-12-15:
Refactor the project to be more like other python modules in a way that can be distributed more easily.

I can do some PRs to this if you agree. The ultimate goal is to convert it to a build/install wheel that can be managed by the python installer and hopefully can be easier to packagers of other distros or even the AUR.

pastaq, 2022-12-15:
Sure. Another user had already started a class refactor so you might collaborate or continue from where they left off.

https://github.com/ajwahab/HandyGCCS/tree/refactor

Samsagax, 2022-12-15:
> Sure. Another user had already started a class refactor so you might collaborate or continue from where they left off.
> 
> https://github.com/ajwahab/HandyGCCS/tree/refactor

Either it is a huge task I did not anticipate or the interest there has faded a little?

pastaq, 2022-12-15:
I last spoke to them on the 7th, but not progress has been made so far. I think they're distracted a bit as they have a WinMax2 and it has some sleep state issues that haven't been figured out yet.

Samsagax, 2022-12-15:
Ok. Seems fair. I'll try to restructure the project in that way, shuffle things around. Will most likely cause some merge conflicts but I'm willing to deal with them. Better now than later in this regard.

pastaq, 2023-07-17:
See progress here: https://github.com/ShadowBlip/HandyGCCS/pull/102

closed 2023-11-06 by pastaq