ShadowBlip/HandyGCCS #64: Update 3.4 breaks Controllerfunction with HandyGCCS on Aya Neo Air

Nik123456789010, 2022-12-16:
Is there a fix, or a possibility to disable Steam Deck Updates?

Samsagax, 2022-12-16:
Hello! @Nik123456789010 Can you provide some logs from handycon? How exactly does it break? As for the question. You should ask HoloISO develpers.

Samsagax, 2022-12-16:
try with:
`$ sudo journalctl --unit handycon`

Nik123456789010, 2022-12-16:
ng@holoiso ~)$ sudo journalctl ––unit handycon
Failed to add match '––unit': Invalid argument
(1)(ng@holoiso ~)$ sudo journalctl ––unit handyc
Failed to add match '––unit': Invalid argument
(1)(ng@holoiso ~)$
(1)(ng@holoiso ~)$ sudo journalctl --unit handycon
Dec 15 14:53:15 holoiso systemd[1]: Stopping Handheld Game Console Controller Service...
Dec 15 14:53:15 holoiso python3[522]: [221215_14:53:15 | handycon.py:1030:restore_all] Receved exit signal. R>
Dec 15 14:53:15 holoiso python3[522]: [221215_14:53:15 | handycon.py:1050:restore_all] Devices restored.
Dec 15 14:53:15 holoiso python3[522]: [221215_14:53:15 | handycon.py:1060:restore_all] Handheld Game Console >
Dec 15 14:53:16 holoiso systemd[1]: handycon.service: Deactivated successfully.
Dec 15 14:53:16 holoiso systemd[1]: Stopped Handheld Game Console Controller Service.
Dec 15 14:53:16 holoiso systemd[1]: handycon.service: Consumed 8.840s CPU time.
-- Boot 203c5383541f4aaeabe1c1a5bca6b300 --
Dec 15 22:31:55 holoiso systemd[1]: Started Handheld Game Console Controller Service.
Dec 15 22:31:55 holoiso python3[529]: [221215_22:31:55 | handycon.py:248:id_system] Identified host system as>
Dec 15 22:31:55 holoiso python3[529]: [221215_22:31:55 | handycon.py:260:get_config] Loading existing config:>
Dec 15 22:31:56 holoiso python3[529]: [221215_22:31:56 | handycon.py:1088:main] Handheld Game Console Control>
Dec 15 22:31:56 holoiso python3[529]: [221215_22:31:56 | handycon.py:833:capture_controller_events] Attemptin>
Dec 15 22:31:56 holoiso python3[529]: [221215_22:31:56 | handycon.py:346:get_controller] Controller device no>
Dec 15 22:31:56 holoiso python3[529]: [221215_22:31:56 | handycon.py:439:get_gyro] Found gyro device. Gyro su>
Dec 15 22:31:56 holoiso python3[529]: [221215_22:31:56 | handycon.py:782:capture_keyboard_events] Attempting >
Dec 15 22:31:56 holoiso python3[529]: [221215_22:31:56 | handycon.py:385:get_keyboard] Found AT Translated Se>
Dec 15 22:31:56 holoiso python3[529]: [221215_22:31:56 | handycon.py:905:capture_power_events] Attempting to >
Dec 15 22:31:56 holoiso python3[529]: [221215_22:31:56 | handycon.py:422:get_powerkey] Found Power Button. Ca>
Dec 15 22:31:56 holoiso python3[529]: [221215_22:31:56 | handycon.py:833:capture_controller_events] Attemptin>
Dec 15 22:31:56 holoiso python3[529]: [221215_22:31:56 | handycon.py:346:get_controller] Controller device no>
Dec 15 22:31:57 holoiso python3[529]: [221215_22:31:57 | handycon.py:833:capture_controller_events] Attemptin>
Dec 15 22:31:57 holoiso python3[529]: [221215_22:31:57 | handycon.py:346:get_controller] Controller device no>
Dec 15 22:31:57 holoiso python3[529]: [221215_22:31:57 | handycon.py:833:capture_controller_events] Attemptin>
Dec 15 22:31:57 holoiso python3[529]: [221215_22:31:57 | handycon.py:346:get_controller] Controller device no>
Dec 15 22:31:58 holoiso python3[529]: [221215_22:31:58 | handycon.py:833:capture_controller_events] Attemptin>
Dec 15 22:31:58 holoiso python3[529]: [221215_22:31:58 | handycon.py:346:get_controller] Controller device no>
lines 1-27...skipping...
Dec 15 14:53:15 holoiso systemd[1]: Stopping Handheld Game Console Controller Service...
Dec 15 14:53:15 holoiso python3[522]: [221215_14:53:15 | handycon.py:1030:restore_all] Receved exit signal. Restoring devices.
Dec 15 14:53:15 holoiso python3[522]: [221215_14:53:15 | handycon.py:1050:restore_all] Devices restored.
Dec 15 14:53:15 holoiso python3[522]: [221215_14:53:15 | handycon.py:1060:restore_all] Handheld Game Console Controller Service stopped.
Dec 15 14:53:16 holoiso systemd[1]: handycon.service: Deactivated successfully.
Dec 15 14:53:16 holoiso systemd[1]: Stopped Handheld Game Console Controller Service.
Dec 15 14:53:16 holoiso systemd[1]: handycon.service: Consumed 8.840s CPU time.
-- Boot 203c5383541f4aaeabe1c1a5bca6b300 --
Dec 15 22:31:55 holoiso systemd[1]: Started Handheld Game Console Controller Service.
Dec 15 22:31:55 holoiso python3[529]: [221215_22:31:55 | handycon.py:248:id_system] Identified host system as AIR and configured defaults for AYA_GEN2.
Dec 15 22:31:55 holoiso python3[529]: [221215_22:31:55 | handycon.py:260:get_config] Loading existing config: /etc/handygccs/handygccs.conf
Dec 15 22:31:56 holoiso python3[529]: [221215_22:31:56 | handycon.py:1088:main] Handheld Game Console Controller Service started.
Dec 15 22:31:56 holoiso python3[529]: [221215_22:31:56 | handycon.py:833:capture_controller_events] Attempting to grab controller device...
Dec 15 22:31:56 holoiso python3[529]: [221215_22:31:56 | handycon.py:346:get_controller] Controller device not yet found. Restarting scan.
Dec 15 22:31:56 holoiso python3[529]: [221215_22:31:56 | handycon.py:439:get_gyro] Found gyro device. Gyro support enabled.
Dec 15 22:31:56 holoiso python3[529]: [221215_22:31:56 | handycon.py:782:capture_keyboard_events] Attempting to grab keyboard device...
Dec 15 22:31:56 holoiso python3[529]: [221215_22:31:56 | handycon.py:385:get_keyboard] Found AT Translated Set 2 keyboard. Capturing input data.
Dec 15 22:31:56 holoiso python3[529]: [221215_22:31:56 | handycon.py:905:capture_power_events] Attempting to grab power device...
Dec 15 22:31:56 holoiso python3[529]: [221215_22:31:56 | handycon.py:422:get_powerkey] Found Power Button. Capturing input data.
Dec 15 22:31:56 holoiso python3[529]: [221215_22:31:56 | handycon.py:833:capture_controller_events] Attempting to grab controller device...
[... cut ...]
Dec 15 22:32:03 holoiso python3[529]: [221215_22:32:03 | handycon.py:346:get_controller] Controller device not yet found. Restarting scan.
Dec 15 22:32:04 holoiso python3[529]: [221215_22:32:04 | handycon.py:833:capture_controller_events] Attempting to grab controller device...
Dec 15 22:32:04 holoiso python3[529]: [221215_22:32:04 | handycon.py:346:get_controller] Controller device not yet found. Restarting scan.
Dec 15 22:32:04 holoiso python3[529]: [221215_22:32:04 | handycon.py:833:capture_controller_events] Attempting to grab controller device...
lines 1-52

pastaq, 2022-12-16:
I think I need to add a check for the old file structure to remove them. Seems to be causing conflicts on upgrade.
Disable handycon service
`sudo systemctl disable --now handcon`
Delete the old systemd unit
`sudo rm /etc/systemd/system/handycon.service`
Reenable handy con
`sudo systemctl enable --now handcon`

Nik123456789010, 2022-12-16:
Thank you, but have no luck with this. The Update 3.3.3, 3.4 break even more on the Aya, there is what like later Firmware update für the Steam Deck

pastaq, 2022-12-16:
Its difficult to assist without more feedback than it didn't work. Are there any specific errors you received when running those commands or change in specific behavior/symptoms? 

Lets see if your devices are properly being detected.

`sudo systemctl disable --now handcon`
`sudo reboot`
`sudo pacman -Sy evtest`
`sudo evtest`
select the item `Microsoft X-Box 360 pad` and verify the normal xbox360 buttons all work
ctrl+c to quit
select the item `AT Translated Set 2 keyboard` and verify the two face buttons work, as well as the RC and LC buttons on top (they are difficult to see, but if you look top down they are next to the shoulder buttons but closer to the center)

let me know if all those are being detected and all events exist while handygccs is off

luisho24, 2022-12-18:
Hello, 

I also have issues with Handycon on my Ayaneo Air Pro.  I follow your instructions, however, I do not even see "Microsoft X-Box 360 pad".  Only AT TRanslated Set 2 keyboard.  

The only keys it captures are the Ayaneo special keys. But non of the other controller keys actually work. 

$ sudo evtest
No device specified, trying to scan all of /dev/input/event*
Available devices:
/dev/input/event0:      Lid Switch
/dev/input/event1:      Power Button
/dev/input/event2:      Power Button
/dev/input/event3:      AT Translated Set 2 keyboard
/dev/input/event4:      Video Bus
/dev/input/event5:      Goodix Capacitive TouchScreen
/dev/input/event6:      PC Speaker
/dev/input/event7:      HD-Audio Generic HDMI/DP,pcm=3
/dev/input/event8:      HD-Audio Generic HDMI/DP,pcm=7
/dev/input/event9:      HD-Audio Generic Mic
/dev/input/event10:     HD-Audio Generic Headphone
Select the device event number [0-10]: 3
Input driver version is 1.0.1
Input device ID: bus 0x11 vendor 0x1 product 0x1 version 0xab83
Input device name: "AT Translated Set 2 keyboard"
Supported events:
  Event type 0 (EV_SYN)
  Event type 1 (EV_KEY)
    Event code 1 (KEY_ESC)
    Event code 2 (KEY_1)
    Event code 3 (KEY_2)
    Event code 4 (KEY_3)
    Event code 5 (KEY_4)
    Event code 6 (KEY_5)
    Event code 7 (KEY_6)
    Event code 8 (KEY_7)
    Event code 9 (KEY_8)
    Event code 10 (KEY_9)
    Event code 11 (KEY_0)
    Event code 12 (KEY_MINUS)
    Event code 13 (KEY_EQUAL)
    Event code 14 (KEY_BACKSPACE)
    Event code 15 (KEY_TAB)
    Event code 16 (KEY_Q)
    Event code 17 (KEY_W)
    Event code 18 (KEY_E)
    Event code 19 (KEY_R)
    Event code 20 (KEY_T)
    Event code 21 (KEY_Y)
    Event code 22 (KEY_U)
    Event code 23 (KEY_I)
    Event code 24 (KEY_O)
    Event code 25 (KEY_P)
    Event code 26 (KEY_LEFTBRACE)
    Event code 27 (KEY_RIGHTBRACE)
    Event code 28 (KEY_ENTER)
    Event code 29 (KEY_LEFTCTRL)
    Event code 30 (KEY_A)
    Event code 31 (KEY_S)
    Event code 32 (KEY_D)
    Event code 33 (KEY_F)
    Event code 34 (KEY_G)
    Event code 35 (KEY_H)
    Event code 36 (KEY_J)
    Event code 37 (KEY_K)
    Event code 38 (KEY_L)
    Event code 39 (KEY_SEMICOLON)
    Event code 40 (KEY_APOSTROPHE)
    Event code 41 (KEY_GRAVE)
    Event code 42 (KEY_LEFTSHIFT)
    Event code 43 (KEY_BACKSLASH)
[... 200 lines cut ...]
Event: time 1671379330.721989, -------------- SYN_REPORT ------------
Event: time 1671379330.731058, type 4 (EV_MSC), code 4 (MSC_SCAN), value db
Event: time 1671379330.731058, -------------- SYN_REPORT ------------
Event: time 1671379330.737198, type 4 (EV_MSC), code 4 (MSC_SCAN), value 20
Event: time 1671379330.737198, -------------- SYN_REPORT ------------
Event: time 1671379330.746440, type 4 (EV_MSC), code 4 (MSC_SCAN), value db
Event: time 1671379330.746440, -------------- SYN_REPORT ------------
Event: time 1671379330.752018, type 4 (EV_MSC), code 4 (MSC_SCAN), value 20
Event: time 1671379330.752018, -------------- SYN_REPORT ------------
Event: time 1671379330.761836, type 4 (EV_MSC), code 4 (MSC_SCAN), value db
Event: time 1671379330.761836, -------------- SYN_REPORT ------------
Event: time 1671379330.768011, type 4 (EV_MSC), code 4 (MSC_SCAN), value 20
Event: time 1671379330.768011, -------------- SYN_REPORT ------------
Event: time 1671379330.777487, type 4 (EV_MSC), code 4 (MSC_SCAN), value db
Event: time 1671379330.777487, -------------- SYN_REPORT ------------
Event: time 1671379330.783032, type 4 (EV_MSC), code 4 (MSC_SCAN), value 20
Event: time 1671379330.783032, -------------- SYN_REPORT ------------
Event: time 1671379330.792074, type 4 (EV_MSC), code 4 (MSC_SCAN), value db
Event: time 1671379330.792074, -------------- SYN_REPORT ------------
Event: time 1671379330.798023, type 4 (EV_MSC), code 4 (MSC_SCAN), value 20
Event: time 1671379330.798023, -------------- SYN_REPORT ------------
Event: time 1671379330.807079, type 4 (EV_MSC), code 4 (MSC_SCAN), value db
Event: time 1671379330.807079, -------------- SYN_REPORT ------------
Event: time 1671379330.813048, type 4 (EV_MSC), code 4 (MSC_SCAN), value 20
Event: time 1671379330.813048, -------------- SYN_REPORT ------------

pastaq, 2022-12-18:
@luisho24 

Hmmm, that would indicate that the xpad driver isn't loading the xb360 controller device initially. This should be reported on the HoloISO issues tracker. Nothing HandyGCCS does would preclude that device from initializing. I don't know if they changed anything over there recently, but the driver loads properly for me in ChimeraOS.

Nik123456789010, 2022-12-19:
Sorry, about my 2 Installation i recognize^it yet: different startoptions( not set by me): Steamdeck, and linux-holoiso  and there HandyGCCS is working. So yes, for me its a chase für HoloISO. Thank you!

closed 2022-12-22 by pastaq