ShadowBlip/HandyGCCS #65: Steam repository take down breaks Installation

Nik123456789010, 2022-12-22:
Is there a way to avoid this?

(ng@holoiso ~)$ cd HandyGCCS
(ng@holoiso HandyGCCS)$ sudo make install
./install.sh
Installing HandyGCCS...
:: Synchronizing package databases...
 holoiso-stable is up to date
 holostaging is up to date
 jupiter.db failed to download
 holo.db failed to download
 core.db failed to download
 extra.db failed to download
 community.db failed to download
error: failed retrieving file 'jupiter.db' from steamdeck-packages.steamos.cloud : The requested URL returned error: 404
error: failed retrieving file 'holo.db' from steamdeck-packages.steamos.cloud : The requested URL returned error: 404
error: failed retrieving file 'core.db' from steamdeck-packages.steamos.cloud : The requested URL returned error: 404
warning: too many errors from steamdeck-packages.steamos.cloud, skipping for the remainder of this transaction
error: failed retrieving file 'community.db' from steamdeck-packages.steamos.cloud : The requested URL returned error: 404
error: failed retrieving file 'extra.db' from steamdeck-packages.steamos.cloud : The requested URL returned error: 404
error: failed to synchronize all databases (failed to retrieve some files)
make: *** [makefile:3: install] Error 1

kyh96403, 2022-12-22:
That is holoiso's problem...
Valve remove old repo...

pastaq, 2022-12-22:
As @kyh96403 said, this is a problem for HoloISO to solve. I'm about to put a disclaimer on the README that I'll not be providing further support for HoloISO because of how frequently it has issues that are induced by bad management and design. There are lots of better options available that don't come with the major security issues, frequently broken packages, unstable installation script, etc. HoloISO is a deeply flawed imitation of an arch based OS and is quite frankly dangerous. I would highly encourage you to use another OS such as a customized EndeavourOS,or ab OOTB configured OS like ChimeraOS instead.

closed 2022-12-22 by pastaq