ShadowBlip/HandyGCCS #66: I can’t install HandyGCCS

McWalters00, 2022-12-28:
When I put the command in « Console » it puts that on me.
[image]

ruineka, 2022-12-28:
Holoiso has broken steam repo configurations. You'll have to work around this.

Enter this in the console in desktop mode:

`git clone https://github.com/ShadowBlip/HandyGCCS.git
cd ~/HandyGCCS
sudo cp -Rv usr/* /usr
sudo udevadm control -R
sudo systemctl enable handycon && sudo systemctl start handycon`

pastaq, 2022-12-28:
> Holoiso has broken steam repo configurations. You'll have to work around this.
>
> Enter this in the console in desktop mode:
>
> `git clone https://github.com/ShadowBlip/HandyGCCS.git cd ~/HandyGCCS sudo [...]

That will install the scripts but not the dependencies. There currently isn't a way that will get the necessary libraries to install otherwise.
@McWalters00 HoloISO is broken and currently not supported.

closed 2022-12-28 by pastaq

McWalters00, 2022-12-28:
> That will install the scripts but not the dependencies. There currently isn't a way that will get the necessary libraries to install otherwise.
>
> @McWalters00 HoloISO is broken and currently not [...]

How is it that nothing works on holoiso when it worked a few months ago?

ruineka, 2022-12-28:
> That will install the scripts but not the dependencies. There currently isn't a way that will get the necessary libraries to install otherwise.
> @McWalters00 HoloISO is broken and currently not [...]

The dependencies are on Holoiso to begin with so this method works at this current time. It bypasses the fact Holoiso has the broken databases at the moment.

ruineka, 2022-12-28:
> How is it that nothing works on holoiso when it worked a few months ago?

Valve changed their repos which Holoiso directly pulls from so the database is invalid and needs to be updated to match Valves changes.

McWalters00, 2022-12-29:
> Valve changed their repos which Holoiso directly pulls from so the database is invalid and needs to be updated to match Valves changes.

And if I go to chimera OS I would have the same problems?

pastaq, 2022-12-29:
> same problems?

No, we don't see the valve repository in our system image.