ShadowBlip/HandyGCCS #86: Please add AYANEO AIR Plus(6800u) device support

TwistTsai, 2023-04-23:
I have tried to use the configuration of AIR and AIR Pro, but it doesn't work, the OS is holoiso.

ruineka, 2023-04-23:
> I have tried to use the configuration of AIR and AIR Pro, but it doesn't work, the OS is holoiso.

I'll be getting the AIR PRO Plus hopefully soon and I'll work towards getting this device supported. A few things changed on this system compared to other devices that will require a system to work with in person to add support.

honjow, 2023-04-23:
Currently on Air Plus, kernel 6.0 and above cannot drive the "AT Translated Set 2 keyboard" properly. The highest version that can be used is kernel 5.19. Additionally, the six-axis sensor cannot be driven properly. If you can downgrade to kernel 5.19, you can try using my repository at https://github.com/honjow/HandyGCCS.

TwistTsai, 2023-04-24:
> Currently on Air Plus, kernel 6.0 and above cannot drive the "AT Translated Set 2 keyboard" properly. The highest version that can be used is kernel 5.19. Additionally, the six-axis sensor cannot be [...]

Thank you，it works.

pastaq, 2023-04-24:
I'm going to close this as a duplicate of #87 as that thread has more valuable information on getting this to work fully. Thanks for reporting this @TwistTsai