ShadowBlip/HandyGCCS #87: Some issues supported by AYANEO AIR Plus.

honjow, 2023-04-24:
Testing system holoiso. In kernel 6.0 and 6.1, the 'AT Translated Set 2 keyboard' device cannot be driven properly. However, it can be driven by rolling back to kernel 5.19.

The gyroscope sensor is not BMI160 but a device with HID BOSC0200.

AirPlus can use this test that I modified: https://github.com/honjow/HandyGCCS

DSDT file
[plus_dsdt.tar.gz](https://github.com/ShadowBlip/HandyGCCS/files/11306539/plus_dsdt.tar.gz)

honjow, 2023-04-24:
> Can you dump a copy of dmesg from 5.19 and 6.1? I'd like to start investigating ASAP. AYANEO are sending me a device for integration but it would be nice to get a head start on the research.
> 
> You have provided some good info. Its good to see the buttons are consistent with the 2/geek. I've been meaning to move to a sysfs implementation for gyro, looks like that will be necessary for the BOSC0200 to work at all. I also noticed you made a function to do on press events. This is also part of the full refactor that I'm doing right now. You can track progress here: https://github.com/ShadowBlip/HandyGCCS/tree/full_refactor currently only the original AYANEO Founders/2021 have been implemented so far. This is a large project but will solve a lot of things in the long term.

[plus_dmesg.tar.gz](https://github.com/ShadowBlip/HandyGCCS/files/11313549/plus_dmesg.tar.gz)
Thanks

pastaq, 2023-04-24:
Interesting

5.19:
```
[Tue Apr 25 02:21:48 2023] usb 3-1: new full-speed USB device number 2 using xhci_hcd
[Tue Apr 25 02:21:48 2023] serio: i8042 KBD port at 0x60,0x64 irq 1
[Tue Apr 25 02:21:48 2023] input: AT Translated Set 2 keyboard as /devices/platform/i8042/serio0/input/input3
[Tue Apr 25 02:21:48 2023] usb 3-1: New USB device found, idVendor=8087, idProduct=0032, bcdDevice= 0.00
[Tue Apr 25 02:21:48 2023] usb 3-1: New USB device strings: Mfr=0, Product=0, SerialNumber=0
```
6.1:
```
[Tue Apr 25 02:30:38 2023] usb 3-1: new full-speed USB device number 2 using xhci_hcd
[Tue Apr 25 02:30:38 2023] serio: i8042 KBD port at 0x60,0x64 irq 1
[Tue Apr 25 02:30:38 2023] usb 3-1: New USB device found, idVendor=8087, idProduct=0032, bcdDevice= 0.00
[Tue Apr 25 02:30:38 2023] usb 3-1: New USB device strings: Mfr=0, Product=0, SerialNumber=0
```
So the device is being detected by the i8042 driver as a keyboard, it's just not assigned to serio by input in 6.0+

kyh96403, 2023-05-01:
That is linux kernel problem..

ruineka, 2023-05-03:
Reverting the changes listed in my bug report fixes this issue. Follow it upstream here https://bugzilla.kernel.org/show_bug.cgi?id=217394

honjow, 2023-05-04:
> Reverting the changes listed in my bug report fixes this issue. Follow it upstream here https://bugzilla.kernel.org/show_bug.cgi?id=217394

Is there any way to modify and compile the kernel by myself for using version 6.1? I am eager to use it.

ruineka, 2023-05-04:
> > Reverting the changes listed in my bug report fixes this issue. Follow it upstream here https://bugzilla.kernel.org/show_bug.cgi?id=217394
> 
> Is there any way to modify and compile the kernel by myself for using version 6.1? I am eager to use it.

You have two options:
- Install the prebuilt kernel I compiled here https://github.com/ruineka/linux/releases
- Use a modified DSDT using CPIO following the instructions here https://wiki.archlinux.org/title/DSDT#Using_a_CPIO_archive
Fixed DSDT

[dsdt_fixed.tar.gz](https://github.com/ShadowBlip/HandyGCCS/files/11398024/dsdt_fixed.tar.gz)

If you choose to install a new kernel on ChimeraOS you will need to install DKMS first and make sure to adjust the bootloader entry to point to the new kernel. If you use the one I linked you to all you need to do is add `-mainline` to the `initramfs-linux.img` and `vmlinuz-linux` located at `/boot/loader/entries/frzr.conf`

It will look like `initrd /vmlinuz-linux-mainline` and `initrd /initramfs-linux-mainline.img`.

honjow, 2023-05-05:
> You have two options:
>
> * Install the prebuilt kernel I compiled here https://github.com/ruineka/linux/releases
> * Use a modified DSDT using CPIO following the instructions here [...]

Thank you. I used the DSDT method and now the keyboard is working properly.

pastaq, 2023-05-07:
AIR Plus code merged into main. It still requires aDSDT, kernel quirk, or BIOS update for the KB to properly function.

closed 2023-05-07 by pastaq