ShadowBlip/HandyGCCS #88: Strange behavior with power button on GPD Win 4

ghost573, 2023-05-02:
Just noticed strange behavior regarding the power button. 

When handycon is started, the power button is intermittent, requiring multiple presses to enter the power menu.
When handycon is stopped, the power button works as intended, only requiring one press to enter the power menu.

I don't think this is intended.

pastaq, 2023-05-02:
Thank you for reporting this. I think the Win4 might be unique, in that it uses a different power button device than most of these devices do so the event is getting lost. There is a part of the code that stops multiple events from firing on multiple power button uinput devices, which causes infinite sleep loops on other devices. Because of that I can't revert it without having a worse issue on those other devices. I'm working on a full refactor of HandyGCCS right now that will solve this issue by allowing me to more easily identify the primary power device per handheld, as well as a host of other improvements.

pastaq, 2023-07-17:
Does this still happen in the latest version?

ghosting3175, 2023-08-02:
The power button works but automatically sleeps the device instead of opening the standby menu which is the intended use.

pastaq, 2023-08-02:
> The power button works but automatically sleeps the device instead of opening the standby menu which is the intended use.
Is this on the desktop? The power button has never been intended to bring up the power menu by default. If you're in steam you can change the config file to say shutdown instead of sleep and it will open the steam power menu that way.

ghosting3175, 2023-08-03:
>Is this on the desktop?

Yes, I don't use front ends, I test this by booting in to KDE and then hit the power button.

ghosting3175, 2023-08-04:
Then how are users meant to have working gyro on their handheld PC?

I want to have working gyro controls.

pastaq, 2023-08-04:
Gyro was removed From the latest version due to issues with the implementation. It will be added back at a later time once it works with a universal interface.

closed 2023-11-06 by pastaq