ShadowBlip/HandyGCCS #90: GPD Win 4 not working?

DocMAX, 2023-05-19:
i have a win 4 and getting this all the time
```
Mai 20 00:25:40 gpdwin4 python3[681]: [230520_00:25:40 | handycon.py:999:capture_controller_events] Attempting to grab controller device...
Mai 20 00:25:40 gpdwin4 python3[681]: [230520_00:25:40 | handycon.py:399:get_controller] Controller device not yet found. Restarting scan.
Mai 20 00:25:41 gpdwin4 python3[681]: [230520_00:25:41 | handycon.py:948:capture_keyboard_events] Attempting to grab keyboard device...
Mai 20 00:25:41 gpdwin4 python3[681]: [230520_00:25:41 | handycon.py:434:get_keyboard] Keyboard device not yet found. Restarting scan.
Mai 20 00:25:41 gpdwin4 python3[681]: [230520_00:25:41 | handycon.py:999:capture_controller_events] Attempting to grab controller device...
Mai 20 00:25:41 gpdwin4 python3[681]: [230520_00:25:41 | handycon.py:399:get_controller] Controller device not yet found. Restarting scan.
Mai 20 00:25:42 gpdwin4 python3[681]: [230520_00:25:42 | handycon.py:948:capture_keyboard_events] Attempting to grab keyboard device...
Mai 20 00:25:43 gpdwin4 python3[681]: [230520_00:25:43 | handycon.py:434:get_keyboard] Keyboard device not yet found. Restarting scan.

```

pastaq, 2023-05-21:
This is the second report I've had of this. It's this in ChimeraOS or did you install it yourself?

DocMAX, 2023-05-21:
It's from Arch Linux AUR repo.

pastaq, 2023-07-17:
Is this still happening?

lucianjones, 2023-07-28:
Hey, just installed ChimeraOS the other day and I'm running into the same issue:

```
Jul 27 23:45:40 cyberdeck python3[640]: [230727_23:45:40 | handycon.py:948:capture_keyboard_events] Attempting to grab keyboard device...
Jul 27 23:45:41 cyberdeck python3[640]: [230727_23:45:41 | handycon.py:434:get_keyboard] Keyboard device not yet found. Restarting scan.
Jul 27 23:45:41 cyberdeck python3[640]: [230727_23:45:41 | handycon.py:999:capture_controller_events] Attempting to grab controller device...
Jul 27 23:45:41 cyberdeck python3[640]: [230727_23:45:41 | handycon.py:399:get_controller] Controller device not yet found. Restarting scan.
```

On version: `handygccs-git 23.05.r136.a8d4057-1`.

One thing I noticed is that the version of HandyGCCS that is in the AUR, the one installed by ChimeraOS, identifies my GPD Win 4 as `GPD_GEN2` instead of `GPD_GEN3` as indicated by the log: 
`handycon.py:298:id_system] Identified host system as G1618-04 and configured defaults for GPD_GEN2.`

That being said, I did build from your latest commit, and the device not yet found errors seemed to remain, although I'm not certain it was properly built. Let me know if there's any other information I can provide to help solve this. Not being able to bind these buttons is about the only thing keeping my system from being perfect haha.

pastaq, 2023-07-29:
That's very odd. It seems to differ from the address we used in providing initial support, which was done by a user of the device. https://github.com/ShadowBlip/HandyGCCS/pull/50/files#diff-d33894a27091b41cedffe66e686d7ae21591bd89920ed64baa0f4536c9d1d54fR325

I think we need to investigate the difference here because that code is known to work on *some* WM2 devices. I would hate to fix this for some folks and break it for others.

Can you run the system profiler and upload the resulting file? There's a more up to date one available in the OpenGamepadUI repo.  https://github.com/ShadowBlip/OpenGamepadUI/blob/main/rootfs/usr/share/opengamepadui/scripts/system_profiler.py

Please also run `sudo cat /sys/class/dmi/id/* > dmi_full.log` and provide that. 

@ajwahab are you able to provide similar logs? It seems there is a difference between your system and @lucianjones

lucianjones, 2023-07-29:
Here are those files @pastaq. Just to make sure there's no confusion - I'm on a Win 4, not a WM2. I did upgrade my system to the latest BIOS when I received it, I wonder if that could result in the differences in systems.

[dmi_full.log](https://github.com/ShadowBlip/HandyGCCS/files/12207760/dmi_full.log)
[G1618-04_system_profile.txt](https://github.com/ShadowBlip/HandyGCCS/files/12207761/G1618-04_system_profile.txt)

pastaq, 2023-07-29:
Oh, right... My mistake.  This would be the relevant PR then. https://github.com/ShadowBlip/HandyGCCS/pull/72/files

ajwahab, 2023-08-04:
> That's very odd. It seems to differ from the address we used in providing initial support, which was done by a user of the device. [...]

Sorry for the delay - do you logs still want logs?

pastaq, 2023-08-05:
@ajwahab na, I'll get another win4 user to confirm.

pastaq, 2023-08-10:
@lucianjones  can you test if this branch fixes your issue?
https://github.com/ShadowBlip/HandyGCCS/pull/113

aarron-lee, 2023-09-06:
I'm assuming this issue can now be closed. After #121 was resolved, as far as I can tell the controls are now fully working on the latest HandyGCCS.

closed 2023-09-06 by pastaq