ShadowBlip/HandyGCCS #91: GPD WIN4

Tendysencer, 2023-06-07:
There are two universal buttons on the back of gpd win4. They can customize the key value to be stored in ec through wincontrol under Windows, which makes the mapping of win4 custom back button in Windows also in Linux. If I want to map one of them to the QAM button, how should I change it? 

I'm sorry that I'm not familiar with Linux.

pastaq, 2023-06-07:
WIN4 support was added in #72. If the mapping was changed in the EC then the default keymap might not be working correctly. Can you use the `evtest` program and let me know what EV_KEY events are triggered on the `"  Mouse for Windows" ` device? (There will be two of them). You'll need to disable the handycon service during the test `systemctl stop handycon`

Tendysencer, 2023-06-09:
By default, the left and right keys on the back correspond to PRTSC and PAUSE respectively. And their desktop mode under SteamOS also takes effect. Their key values are the same as ordinary keyboards. Through wincontrol under Windows, I can simulate the back key to the key on the xbox, and even simulate the key combination. Unfortunately, the program does not allow me to simulate the xbox key, which makes it impossible for me to simulate xbox+A to quickly start the QAM menu.

pastaq, 2023-06-11:
> By default, the left and right keys on the back correspond to PRTSC and PAUSE respectively. And their desktop mode under SteamOS also takes effect. Their key values are the same as ordinary [...]

PRTSC is not a valid key in the Linux event code map here: https://github.com/torvalds/linux/blob/master/include/uapi/linux/input-event-codes.h

The pause button is detected correctly in this program already, so we just need to verify the event code for the left button.

I'll need the event codes from evtest as described in my previous post.