ShadowBlip/HandyGCCS #108: Right Menu button change button mappings

theolderbear, 2023-07-31:
I installed holoiso on Ayaneo air on an sd card one 10 days ago and all buttons were working as a charm. Yesterday I installed on ssd and I have issue. Basically when I press the button to open the Right menu (the one with two lines on the right of ayaneo button), the menu pops up but the mapping of all button is messed up, B does not work, down goes left and ayaneo button is A.
If I press again the two lines button everything goes back to normal. Any suggestion? Should I format and install from scratch?

[Button Map]
button1 = SCR
button2 = QAM
button3 = ESC
button4 = OSK
button5 = MODE
button6 = OPEN_CHIMERA
button7 = TOGGLE_PERFORMANCE
button8 = MODE
button9 = TOGGLE_MOUSE
button10 = ALT_TAB
button11 = KILL
button12 = TOGGLE_GYRO
power_button = SUSPEND

pastaq, 2023-07-31:
That's... Odd. What version of the air?

theolderbear, 2023-07-31:
Yes, is weird. It is an Ayaneo air pre-pro, it is the ticker model, with improved fan but without the bigger battery. I even tried to restore the previous conf file but did not solve the issue.

theolderbear, 2023-07-31:
I work as a developer since 20 years, maybe if you give me some hints I could try to find out the issue

pastaq, 2023-07-31:
I've experienced something like that before by adding BTN_BASE as an enabled event. Steam input seems to use the BTN_* event index in the list of capabilities rather than actually matching the string name. Pressing that button shouldn't be adding any event but you could compare capabilities using `evtest` before/after pressing it. You could also confirm that it's shifting the events.

pastaq, 2023-11-06:
@theolderbear were you able to confirm which events are activated on  your device?