ShadowBlip/HandyGCCS #117: Suspend issue on Ayaneo air pro

theolderbear, 2023-08-24:
Hi, I installed this version of holoiso on Ayaneo air pro nad I have this issue:
Changing handygccs config for power_button does not solve the problem.

Step to reproduce:
1. Turn on console in game mode until steam ui is visible
2. Press power button
3. Animation is shown, screen go black and sticks leds start flashing (this happen when console is on sleep mode)
4. Press power button
5. Wait around 5 sec, screen turn up, steam ui is visible and sticks leds stop flashing
6. Press power button
7. Animation is shown, screen go black but sticks leds do not start flashing
8. Console is frozen and unresponsive to any button
9. Pressing power button multiple times cause console to restart, sometimes steam ui get visible for some seconds before restart
10. After restart, sometimes, no animation is visible and the screen stays black, pressing power button turns the screen on and show steam ui

Maybe handygccs is not the problem here, but maybe I could get some suggestion on how to solve this.

pastaq, 2023-08-26:
This doesn't seem to be a HandyGCCS issue. I would post this issue on they HoloISO GitHub, attach a dmesg and journalctl log as well.

ggrensteiner, 2023-08-31:
This may be the same issue as #120 @theolderbear can you get a strace of the handycon service binary and check if it's in an infinite loop rescanning all of the devices rather than waiting for events?

closed 2023-08-31 by pastaq