ShadowBlip/HandyGCCS #120: handycon doesn't survive suspend/resume

ggrensteiner, 2023-08-31:
The current build gets caught in a loop after resume on the ROG Ally. After resume the device stops accepting most inputs but every once in a while a few key presses will make it in after several seconds of delay.

This is seen using kernel 6.5 and this stack trace is a loop that just keeps going infinitely even without any event traffic at all. I think it's in a loop looking for the event inputs that it was monitoring before sleep.

[handyconstrace.txt](https://github.com/ShadowBlip/HandyGCCS/files/12489505/handyconstrace.txt)

ggrensteiner, 2023-08-31:
Seems to be an artifact of whatever mode the N-Key device gets into if it doesn't re-appear after two resumes. It's randomly started coming back every other resume again.

pastaq, 2023-08-31:
This is mostly expected behavior. When HandyGCCS loses access to an expected input device it will poll for the missing device. This might have an amplified effect on the ROG ally though considering it is losing multiple devices at the same time. I haven't been able to reproduce the extreme slowdowns others have reported dung that polling state though.

fennectech, 2023-10-25:
is this fixable?

fennectech, 2023-10-25:
I’d like to put a 25 USD bounty on a fix for this.  This issue is the only thing holding me back from using Linux default on my ally

pastaq, 2023-10-26:
The issue isn't with HandyGCCS, but with the Ally having proper suspend support in Linux. This will need to be resolved through other means.

closed 2023-10-26 by pastaq