ShadowBlip/HandyGCCS #121: [bug] Back buttons on GPD Win 4 not functional in gamepadui

aarron-lee, 2023-09-03:
Device: GPD Win 4 6800u
Version: ChimeraOS 43-1
Bios: v3.06

after updating to the latest handycon via pikaur, the back buttons are still not functional on gamepadui.

handycon does seem to be properly registering the power button. Before, a power button tap would immediately go into suspend, but now it shows the Steam Deck suspend animation before actually suspending.

note that on desktop mode, the back-left button registers as a screenshot button, but it turns out that this is probably the default behavior; even when I had handycon fully disabled, it worked as a screenshot button on the desktop only.

created this issue after original discussion in https://github.com/ChimeraOS/chimeraos/issues/728

pastaq, 2023-09-03:
Can you run the system profiler from [OpenGamepadUI](https://github.com/ShadowBlip/OpenGamepadUI/blob/main/rootfs/usr/share/opengamepadui/scripts/system_profiler.py) and post the resulting file? Please ensure `handycon.service` and `opengamepadui` are both not running before conducting the test. When it asks for the key presses, press the left back button then the right back button before hitting `ctrl+c` on the keyboard to end the script. This must be run with `sudo` permissions.

Thanks.

aarron-lee, 2023-09-04:
Generated file is here: [G1618-04_system_profile.txt](https://github.com/ShadowBlip/HandyGCCS/files/12508265/G1618-04_system_profile.txt)

Also, not sure if this helps, but when I tried to manually map the back buttons via [input-remapper](https://github.com/sezanzeb/input-remapper), it turns out that the back buttons are registered as a mouse. `Pause` was the back right button, `Screenshot` was the back left button.

[image]

Unfortunately, input-remapper's systemd service seems to fail to initialize in gamepadui on boot, but it does properly work when I tested it in desktop mode BPM.

pastaq, 2023-09-04:
Since there are two`  mouse for windows` devices it's possible we're just grabbing the wrong one. Do you know if the renters trigger on the device attached to
`usb-0000:73:00.3-4.2/input0` or the one attached to `usb-0000:73:00.3-4.2/input1`

aarron-lee, 2023-09-05:
hrm, so I tried building handycon after updating `gpd_gen3.py` with the `input1` ID, and noticed the following after a reboot:

- the back buttons seem to no longer register in the `input-mapper` app, which I think means that handycon now is controlling the back buttons.
  - The back-left button had also previously been a PrintScreen button by default on Desktop mode without handycon
    - it no longer is registering as a printscreen button on the desktop
- however, when I try to use the back buttons in game mode, they don't do anything.
  - it seems that the buttons are now being controlled, but aren't actually registered to do anything.

I'll try to change around the code in handycon, and try to figure out what's happening.

aarron-lee, 2023-09-05:
Going to investigate some more tomorrow, but did some basic logging in `gpd_gen3.py`.

Added some additional logging.

When pressing the back-left button, the following values were logged:

```
seed_event event at 1693883873.070568, code 04, type 04, val 458822 
 active_keys [99] 
 button_on 458822 
---
seed_event event at 1693883873.070568, code 99, type 01, val 01 
 active_keys [99] 
 button_on 1 
---
seed_event event at 1693883873.070568, code 00, type 00, val 00 
 active_keys [99] 
 button_on 0 
---
seed_event event at 1693883873.336501, code 99, type 01, val 02 
 active_keys [99] 
 button_on 2 
---
seed_event event at 1693883873.336501, code 00, type 00, val 01 
 active_keys [99] 
 button_on 1 
---
seed_event event at 1693883873.366600, code 04, type 04, val 458822 
 active_keys [] 
 button_on 458822 
---
seed_event event at 1693883873.366600, code 99, type 01, val 00 
 active_keys [] 
 button_on 0 
---
seed_event event at 1693883873.366600, code 00, type 00, val 00 
 active_keys [] 
 button_on 0 
---
```

When I pressed the back-right button, got the following logs:

```
seed_event event at 1693883892.136303, code 04, type 04, val 458824 
 active_keys [119] 
 button_on 458824 
---
seed_event event at 1693883892.136303, code 119, type 01, val 01 
 active_keys [119] 
 button_on 1 
 ---
seed_event event at 1693883892.136303, code 00, type 00, val 00 
 active_keys [119] 
 button_on 0 
 button2 [[1, 316], [1, 304]]
---
seed_event event at 1693883892.402902, code 119, type 01, val 02 
 active_keys [119] 
 button_on 2 
---
seed_event event at 1693883892.402902, code 00, type 00, val 01 
 active_keys [119] 
 button_on 1 
---
seed_event event at 1693883892.432318, code 04, type 04, val 458824 
 active_keys [] 
 button_on 458824 
---
seed_event event at 1693883892.432318, code 119, type 01, val 00 
 active_keys [] 
 button_on 0 
---
seed_event event at 1693883892.432318, code 00, type 00, val 00 
 active_keys [] 
 button_on 0 
---
```

pastaq, 2023-09-05:
I just looked at the code. It's super wrong for the key up events...

pastaq, 2023-09-05:
Line 41 should have just [119] for seed_event.code and 47 should have 99

pastaq, 2023-09-05:
Actually all the GPD devices have this bug

pastaq, 2023-09-05:
Just pushed a fix

aarron-lee, 2023-09-05:
Thanks, I can confirm that this almost fixed the issue. The last remaining change is to change from `input0` to `input1`.

https://github.com/ShadowBlip/HandyGCCS/blob/80e86e94021e5e7c476343bcd4f4a5ff461e5862/src/handycon/handhelds/gpd_gen3.py#L21

After updating to input1 on the latest handygccs, I can confirm that the back-left button is working as QAM, back-right for screenshot.

aarron-lee, 2023-09-05:
Just to clarify, I had to do one more manual code change, which was updating the `KEYBOARD_ADDRESS` code in `gpd_gen3.py` from `input0` to `input1`. Worked perfectly afterwards.

I'm not seeing the code change on the main branch, so I'm assuming it's a change that'll be pushed later.

Anyways, thanks for the help with troubleshooting this issue! :)

pastaq, 2023-09-06:
I misread what you wrote. I'll update that soon, thanks!

pastaq, 2023-09-06:
Updated. Please verify and we can close this issue.

aarron-lee, 2023-09-06:
just updated to the latest handygccs via pikaur, I can confirm that it all works now.

closed 2023-09-06 by aarron-lee