ShadowBlip/HandyGCCS #124: [Rog Ally] Way to use M1/M2 back buttons

dasshit, 2023-10-06:
Is there a way to use back buttons as ABXY-buttons? I'm trying to do so in my fork, but there two problems:
1. I dont know how to simulate AXBY-buttons
2. At process_event I'm catching identical events on both buttons. How to identify what button was pressed?

pastaq, 2023-10-07:
You would need to trigger changing to gamepad mode on the controller. There are some scripts floating around on the discord that do this but they stopped working on kernel 6.5 so we suspect a libusb change broke it. That's the only way to make the back buttons act individually. You lose all the other combos in that mode where the back button acts like a modifier key.

How are you planning on simulating four keys with just two keys?

If you want to add those keys as emitable events you'll need to map them in constants.py, I.E. EVENT_BTN_A. K at the BTN_MODE map as an example of a single key.

dasshit, 2023-10-07:
I expressed myself inaccurately, meaning that each of the back buttons would correspond to one of the ABXY :)

pastaq, 2023-10-07:
There are only two back buttons?

dasshit, 2023-10-08:
Yes, both come to process_event function as active_keys = [185]

[Buttons](https://www.google.com/search?q=asus+rog+ally+back+buttons&client=ms-android-google&sca_esv=571696387&tbm=isch&prmd=sivn&sxsrf=AM9HkKnt40iyw7gBTWrMzWuLoMZ3mP_sYg:1696756715770&source=lnms&sa=X&ved=2ahUKEwib7sCqj-aBAxX0JBAIHUZsC2AQ_AUoAnoECAMQAg&biw=412&bih=771&dpr=2.63#vhid=-pFYa7qovvenpM&vssid=3981:L5yR7YrWYmxXyM)

pastaq, 2023-10-08:
I know, I daily drive once if them. I was just confused how you want to map four buttons to two buttons but I think I understand.

dasshit, 2023-10-08:
Now I'm playing Armored Core VI, so i want to map back buttons as A and X buttons.

dasshit, 2023-10-08:
I was able to made back buttons as A button

```python

button_a = [
    [e.EV_KEY, e.BTN_A]
]

if active_keys == [185] \
        and button_on in [1, 2]:
    await handycon.emit_now(seed_event, button_a, 1)
elif active_keys == [] \
        and seed_event.code == 185 \
        and button_on == 0:
    await handycon.emit_now(seed_event, button_a, 0)
```

pastaq, 2023-10-08:
> I was able to made back buttons as A button with following code:
> 
> ```python
> button_a = [
>     [e.EV_KEY, e.BTN_A]
> ]
> 
> if active_keys == [185] \
>         and button_on in [1, 2]:
>     await handycon.emit_now(seed_event, button_a, 1)
> elif active_keys == [] \
>         and seed_event.code == 185 \
>         and button_on == 0:
>     await handycon.emit_now(seed_event, button_a, 0)
> ```

I don't recommend you do this. The problem is the back button's lose their state if another button is pressed when they are in the "default" mode, meaning if you were to press another button before releasing the back button, you wont get a key up event for the back button. This will leave you in a position where you need to press the button again to end the press. 

Additionally, this will emit an A down event and interfere with the logic of the macro keys mapped to the ABXY/Dpad and back button.

In the gamepad mode they retain their state and this isn't an issue, and they also have unique event ID's.

dasshit, 2023-10-08:
Can you share link for info about gamepad mode? I cant fined it on discord :(

dasshit, 2023-10-08:
> > I was able to made back buttons as A button with following code:
> > 
> > ```python
> > button_a = [
> >     [e.EV_KEY, e.BTN_A]
> > ]
> > 
> > if active_keys == [185] \
> >         and button_on in [1, 2]:
> >     await handycon.emit_now(seed_event, button_a, 1)
> > elif active_keys == [] \
> >         and seed_event.code == 185 \
> >         and button_on == 0:
> >     await handycon.emit_now(seed_event, button_a, 0)
> > ```
> 
> I don't recommend you do this. The problem is the back button's lose their state if another button is pressed when they are in the "default" mode, meaning if you were to press another button before releasing the back button, you wont get a key up event for the back button. This will leave you in a position where you need to press the button again to end the press. 
> 
> Additionally, this will emit an A down event and interfere with the logic of the macro keys mapped to the ABXY/Dpad and back buttons.
> 
> In the gamepad mode they retain their state and this isn't an issue, and they also have unique event ID's.  

I see it now, when one or both of back buttons being hold - Y button starts to act as A button. After this hold state of A button is broken.

pastaq, 2023-10-08:
https://discord.com/channels/560281725175988233/1118341089355636776/1153034882075861032

dasshit, 2023-10-09:
Thanks, totally what i was looking for :) Swapped "Toggle performance" to A and X with M1/M2 buttons

```python
def init_handheld(...):
    ...
    with usb1.USBContext() as usb_context:
        dev = open_dev(0x0b05, 0x1abe, usb_context)
        if dev.kernelDriverActive(2) == 1:
            dev.detachKernelDriver(2)
            dev.claimInterface(2)
            dev.resetDevice()
            replay(dev)
            dev.releaseInterface(2)
            dev.attachKernelDriver(2)
```

closed 2023-11-06 by pastaq