ShadowBlip/HandyGCCS #125: Handycon can't detect keyboard 2 on Ally

Janer1, 2023-10-13:
Systemctl status handycon keeps giving me a log saying it's unable to be detected. Tested on both bios version 330 and 323. Using Bazzite

pastaq, 2023-10-13:
I'm working on adding support for BIOS 330 as that update changes some pcie addresses. Did you disable the SD card reader in BIOS by chance? That can also change the pcie addresses.

Janer1, 2023-10-13:
Nope never disabled the sd card reader

philmmanjaro, 2023-10-16:
most likely fixed via: https://github.com/honjow/HandyGCCS/commit/6a5a2c868d9e696c9cd59d2a8a68e6b93535d455

pastaq, 2023-10-16:
> most likely fixed via: [honjow@6a5a2c8](https://github.com/honjow/HandyGCCS/commit/6a5a2c868d9e696c9cd59d2a8a68e6b93535d455)

I disagree if the issue exists in BIOS 323.

philmmanjaro, 2023-10-21:
@Janer1 which kernel was used and is [this patch](https://github.com/ChimeraOS/linux/commit/975eca4580a3a4fbdfe3b24ee6523bc5aa9f2c95) applied?

closed 2023-11-06 by pastaq