ShadowBlip/HandyGCCS #126: ONEXPLAYER 2 PRO Support

NightHammer1000, 2023-10-18:
Device Name: ONEXPLAYER 2 PRO ARP23P

Capture:
```
System ID:
ONEXPLAYER 2 PRO ARP23P

X-Box 360 Device:
Microsoft X-Box 360 pad | usb-0000:64:00.3-4/input0 | bustype: 3 vendor: 1118 product: 654 version: 6144

keyboard Device:
AT Translated Set 2 keyboard | isa0060/serio0/input0 | bustype: 17 vendor: 1 product: 1 version: 43907

All Devices:
Microsoft X-Box 360 pad | usb-0000:64:00.3-4/input0 | bustype: 3 vendor: 1118 product: 654 version: 6144
HD-Audio Generic Headphone | ALSA | bustype: 0 vendor: 0 product: 0 version: 0
HD-Audio Generic Mic | ALSA | bustype: 0 vendor: 0 product: 0 version: 0
HD-Audio Generic HDMI/DP,pcm=7 | ALSA | bustype: 0 vendor: 0 product: 0 version: 0
HD-Audio Generic HDMI/DP,pcm=3 | ALSA | bustype: 0 vendor: 0 product: 0 version: 0
Video Bus | LNXVIDEO/video/input0 | bustype: 25 vendor: 0 product: 6 version: 0
GXTP7385:00 27C6:011A Keyboard | i2c-GXTP7385:00 | bustype: 24 vendor: 10182 product: 282 version: 256
GXTP7385:00 27C6:011A UNKNOWN | i2c-GXTP7385:00 | bustype: 24 vendor: 10182 product: 282 version: 256
GXTP7385:00 27C6:011A Stylus | i2c-GXTP7385:00 | bustype: 24 vendor: 10182 product: 282 version: 256
GXTP7385:00 27C6:011A | i2c-GXTP7385:00 | bustype: 24 vendor: 10182 product: 282 version: 256
AT Translated Set 2 keyboard | isa0060/serio0/input0 | bustype: 17 vendor: 1 product: 1 version: 43907
Sleep Button | PNP0C0E/button/input0 | bustype: 25 vendor: 0 product: 3 version: 0
Lid Switch | PNP0C0D/button/input0 | bustype: 25 vendor: 0 product: 5 version: 0
Power Button | PNP0C0C/button/input0 | bustype: 25 vendor: 0 product: 1 version: 0

Captured Keymaps:
[311]
[125]
[32, 125]
[29]
[29, 125]
[24, 29, 125]
[316]
[314]
[315]
[310]
[304]
[307]
[308]
[305]
[318]
[317]

```

My Notes:
```
VOL- (X1)
[125]
[32, 125]

VOL+ (X2)
[29]
[29, 125]
[24, 29, 125]

Orange Button on Left Controller:
[316]

Turbo Button on Top does not Produce a code.

Select:
[314]

Start:
[315]

Left Shoulder Button
[310]

Right Shoulder Button
[311]

Mode Swith Button does not Produce a code.
LED Button does not Produce a code
```

pastaq, 2023-10-18:
Can you use `evtest` to determine which device each of the buttons comes from?

pastaq, 2023-10-18:
Also, do the shoulder, start, select, and mode(orange) buttons not work by default?

Is the orange button momentary(instant on press)or triggered only on release?

pastaq, 2023-10-20:
> Sorry of the Oversight. Controller works normally without any work because it reports as a standart X360 Controller. Even Guide Button works.
> 
> For the Turbo and X1 and X2 Buttons. I solved that mystery as Well.
> 
> Its controlled through EC to switch between the Power Mode and the Hotkey mode. After setting the EC Address 0xEB to 0x40 the Button sends CTRL+ALT+META (17, 18, 91) in Hotkey mode
> 
> Setting the same EC Address to 0xEB also switches the X1 and X2 Keys to Volume Control in addition to setting the Turbo to Hotkey mode.
> 
> Setting it to 0x00 puts it back to Turbo Mode.

This is excellent. We can add this to the OXP platform driver pretty easily.

Do you know what registers the fan speed output to? I think that's the last piece of the puzzle. 

EC 0x4A controls the Fan. Write 0x00 for Automatic, 0x01 for Manual.
EC 0x4B controls the Speed, we need to know the range (0-100/128/255)
And we need to know where the fan speed is read from. Usually it's two registers concatenated.

NightHammer1000, 2023-10-20:
These Values are all that changed when using RW Utility on the EC while changing Fan Speeds in OneXConsole.
OneXConsole being such a BAD Program  does not make it any better.
I'm still trying to figure out the Range. Seems quite cryptic to me. But it has been a long day. I will give it another spin tomorrow with a fresh head to think with.

pastaq, 2023-10-20:
I updated the oxp-sensors to support the oxp2 and pro in my branch. We can test if the fan speed register is the same with this.

https://gitlab.com/derekjohn.clark/oxp-sensors

Can you please compile and test this?

pastaq, 2023-10-20:
> Its controlled through EC to switch between the Power Mode and the Hotkey mode. After setting the EC Address 0xEB to 0x40 the Button sends CTRL+ALT+META (17, 18, 91) in Hotkey mode

These numbers are confusing to me, they don't line up with the [Input Event Codes](https://github.com/torvalds/linux/blob/master/include/uapi/linux/input-event-codes.h) or either left or right version of any of the three keys. Can you post a full `evtest` capture of this key when captured?

NightHammer1000, 2023-10-21:
These codes are what I got from using https://w3c.github.io/uievents/tools/key-event-viewer.html as I am on Windows currently while researching the Original OneXConsole.

While doing this I found this. This tool has both Models figured out already and everything seems to work, In Windows at least. But here are all the details needed for controlling the EC.

Seems like OneXPlayer 2 and 2 Pro use the Same Values for the EC.

https://github.com/Valkirie/HandheldCompanion/blob/ab82a64e481952b9ce81c044e7ef783d7c003b4b/HandheldCompanion/Devices/OneXPlayer/OneXPlayer2.cs

NightHammer1000, 2023-10-21:
Will setup a Linux SD Card later that weekend to evtest.

pastaq, 2023-10-21:
> These codes are what I got from using https://w3c.github.io/uievents/tools/key-event-viewer.html as I am on Windows currently while researching the Original OneXConsole.
> 
> While doing this I found this. This tool has both Models figured out already and everything seems to work, In Windows at least. But here are all the details needed for controlling the EC.
> 
> Seems like OneXPlayer 2 and 2 Pro use the Same Values for the EC.
> 
> https://github.com/Valkirie/HandheldCompanion/blob/ab82a64e481952b9ce81c044e7ef783d7c003b4b/HandheldCompanion/Devices/OneXPlayer/OneXPlayer2.cs
> 
> The Tool manages to change the Hotkey the Turbo Button outputs. Have not figured out how yet. My C# is a bit rusty.

This is pretty funny. I was just talking to their dev last night to flash out some missing details. I'll have a branch with OXP 2 support posted today that will rely on the driver being updated.

NightHammer1000, 2023-10-21:
Great! One Step Closer to the Deck experience!

pastaq, 2023-10-22:
@NightHammer1000 can you test this branch in conjunction with the updated oxp-sensors w/ OXP2 fixes? It should automatically take over the turbo button when it initializes and the turbo will open the QAM. Needs the driver support for turbo takeover.

NightHammer1000, 2023-10-25:
Got around to test it now, Compiled and installed both HandyGCCS and the opx-sensors and it seems to work as it should. 
Thanks!

Now on to figuring out why that damned Soundchip won't work.

pastaq, 2023-10-26:
> Got around to test it now, Compiled and installed both HandyGCCS and the opx-sensors and it seems to work as it should. Thanks!
> 
> Now on to figuring out why that damned Soundchip won't work.

Awesome, thanks for testing. I just learend about the M1/M2 buttons. Can i get a capture of those from `evtest` without handygccs running and the tt_toggle in takeover please?

NightHammer1000, 2023-10-26:
Setting the Turbo Button EC Address like I said above can switch it into Volume Control. 
Will do the evtest when I am home again on the Weekend.

pastaq, 2023-10-27:
> Setting the Turbo Button EC Address like I said above can switch it into Volume Control. Will do the evtest when I am home again on the Weekend with the Takeover

HandyGCCS should be setting the EC when it starts up. There was a bug in oxp-sensors that was preventing the attribute from being exposed so you'll need to update that and ensure it is loaded.

NightHammer1000, 2023-10-28:
I bring gifts!

evtest of X1 and X2

```
Input driver version is 1.0.1
Input device ID: bus 0x11 vendor 0x1 product 0x1 version 0xab83
Input device name: "AT Translated Set 2 keyboard"
Supported events:
  Event type 0 (EV_SYN)
  Event type 1 (EV_KEY)
    Event code 1 (KEY_ESC)
    Event code 2 (KEY_1)
    Event code 3 (KEY_2)
    Event code 4 (KEY_3)
    Event code 5 (KEY_4)
    Event code 6 (KEY_5)
    Event code 7 (KEY_6)
    Event code 8 (KEY_7)
    Event code 9 (KEY_8)
    Event code 10 (KEY_9)
    Event code 11 (KEY_0)
    Event code 12 (KEY_MINUS)
    Event code 13 (KEY_EQUAL)
    Event code 14 (KEY_BACKSPACE)
    Event code 15 (KEY_TAB)
    Event code 16 (KEY_Q)
    Event code 17 (KEY_W)
    Event code 18 (KEY_E)
    Event code 19 (KEY_R)
    Event code 20 (KEY_T)
    Event code 21 (KEY_Y)
    Event code 22 (KEY_U)
    Event code 23 (KEY_I)
    Event code 24 (KEY_O)
    Event code 25 (KEY_P)
    Event code 26 (KEY_LEFTBRACE)
    Event code 27 (KEY_RIGHTBRACE)
    Event code 28 (KEY_ENTER)
    Event code 29 (KEY_LEFTCTRL)
    Event code 30 (KEY_A)
    Event code 31 (KEY_S)
    Event code 32 (KEY_D)
    Event code 33 (KEY_F)
    Event code 34 (KEY_G)
    Event code 35 (KEY_H)
    Event code 36 (KEY_J)
    Event code 37 (KEY_K)
    Event code 38 (KEY_L)
    Event code 39 (KEY_SEMICOLON)
    Event code 40 (KEY_APOSTROPHE)
    Event code 41 (KEY_GRAVE)
    Event code 42 (KEY_LEFTSHIFT)
    Event code 43 (KEY_BACKSLASH)
    Event code 44 (KEY_Z)
    Event code 45 (KEY_X)
    Event code 46 (KEY_C)
    Event code 47 (KEY_V)
    Event code 48 (KEY_B)
    Event code 49 (KEY_N)
    Event code 50 (KEY_M)
    Event code 51 (KEY_COMMA)
    Event code 52 (KEY_DOT)
    Event code 53 (KEY_SLASH)
    Event code 54 (KEY_RIGHTSHIFT)
    Event code 55 (KEY_KPASTERISK)
    Event code 56 (KEY_LEFTALT)
    Event code 57 (KEY_SPACE)
    Event code 58 (KEY_CAPSLOCK)
    Event code 59 (KEY_F1)
[... 344 lines cut ...]
Event: time 1698534521.528000, type 1 (EV_KEY), code 29 (KEY_LEFTCTRL), value 0
Event: time 1698534521.528000, type 4 (EV_MSC), code 4 (MSC_SCAN), value 700e3
Event: time 1698534521.528000, type 1 (EV_KEY), code 125 (KEY_LEFTMETA), value 0
Event: time 1698534521.528000, type 4 (EV_MSC), code 4 (MSC_SCAN), value 70012
Event: time 1698534521.528000, type 1 (EV_KEY), code 24 (KEY_O), value 0

// LED KEY
Event: time 1698534521.528000, -------------- SYN_REPORT ------------
Event: time 1698534525.272619, type 4 (EV_MSC), code 4 (MSC_SCAN), value 700e0
Event: time 1698534525.272619, type 1 (EV_KEY), code 29 (KEY_LEFTCTRL), value 1
Event: time 1698534525.272619, type 4 (EV_MSC), code 4 (MSC_SCAN), value 700e2
Event: time 1698534525.272619, type 1 (EV_KEY), code 56 (KEY_LEFTALT), value 1
Event: time 1698534525.272619, type 4 (EV_MSC), code 4 (MSC_SCAN), value 7004c
Event: time 1698534525.272619, type 1 (EV_KEY), code 111 (KEY_DELETE), value 1
Event: time 1698534525.272619, -------------- SYN_REPORT ------------
Event: time 1698534525.279891, type 4 (EV_MSC), code 4 (MSC_SCAN), value 700e0
Event: time 1698534525.279891, type 1 (EV_KEY), code 29 (KEY_LEFTCTRL), value 0
Event: time 1698534525.279891, type 4 (EV_MSC), code 4 (MSC_SCAN), value 700e2
Event: time 1698534525.279891, type 1 (EV_KEY), code 56 (KEY_LEFTALT), value 0
Event: time 1698534525.279891, type 4 (EV_MSC), code 4 (MSC_SCAN), value 7004c
Event: time 1698534525.279891, type 1 (EV_KEY), code 111 (KEY_DELETE), value 0
Event: time 1698534525.279891, -------------- SYN_REPORT ------------
^C
[liveuser@CachyOS ~]$ 
```

pastaq, 2023-10-29:
Something we've discovered is the X1/X2 are being mode switched by the onexplayer software in windows. I can  remap the functions they output to the volume buttons but I'd rather have the native events and me able to mode switch the volume keys in the driver to open up more functionality if possible. @NightHammer1000 do you think you could investigate what changes in the EC for those keys?

pastaq, 2023-10-29:
When is the `One KeyP-Mouse` present?
Can I get a copy of `cat /proc/bus/input/devices` when it is available?

pastaq, 2023-10-29:
MODE and X2 appear to have the same code. That is unfortunate if accurate.

NightHammer1000, 2023-10-29:
Oh. So here is the Deal. Did not notice the Controller was in Desktop mode.
One KeyP-Mouse is present when the Controller is in Desktop Mode. (Holding Mode Key)
If it's not these Keys do nothing.

My bad. Can be Ignored.

closed 2023-11-06 by pastaq