ShadowBlip/HandyGCCS #138: Add GPD WIN Mini

tcdude, 2024-01-07:
Please add the GPD WIN Mini's controller.

Information about the device:
output of `cat /proc/bus/input/devices`:

```
I: Bus=0019 Vendor=0000 Product=0001 Version=0000
N: Name="Power Button"
P: Phys=PNP0C0C/button/input0
S: Sysfs=/devices/LNXSYSTM:00/LNXSYBUS:00/PNP0C0C:00/input/input0
U: Uniq=
H: Handlers=kbd event0
B: PROP=0
B: EV=3
B: KEY=10000000000000 0

I: Bus=0019 Vendor=0000 Product=0005 Version=0000
N: Name="Lid Switch"
P: Phys=PNP0C0D/button/input0
S: Sysfs=/devices/LNXSYSTM:00/LNXSYBUS:00/PNP0C0D:00/input/input1
U: Uniq=
H: Handlers=event1
B: PROP=0
B: EV=21
B: SW=1

I: Bus=0019 Vendor=0000 Product=0006 Version=0000
N: Name="Video Bus"
P: Phys=LNXVIDEO/video/input0
S: Sysfs=/devices/LNXSYSTM:00/LNXSYBUS:00/PNP0A08:00/device:24/LNXVIDEO:00/input/input2
U: Uniq=
H: Handlers=kbd event2
B: PROP=0
B: EV=3
B: KEY=3e000b00000000 0 0 0

I: Bus=0011 Vendor=0001 Product=0001 Version=ab83
N: Name="AT Translated Set 2 keyboard"
P: Phys=isa0060/serio0/input0
S: Sysfs=/devices/platform/i8042/serio0/input/input3
U: Uniq=
H: Handlers=sysrq kbd leds event3
B: PROP=0
B: EV=120013
B: KEY=402000000 3803078f800d001 feffffdfffefffff fffffffffffffffe
B: MSC=10
B: LED=7

I: Bus=0003 Vendor=2f24 Product=0135 Version=0110
N: Name="  Mouse for Windows"
P: Phys=usb-0000:63:00.3-3/input0
S: Sysfs=/devices/pci0000:00/0000:00:08.1/0000:63:00.3/usb1/1-3/1-3:1.0/0003:2F24:0135.0001/input/input4
U: Uniq=
H: Handlers=event4 mouse0
B: PROP=0
B: EV=17
B: KEY=1f0000 0 0 0 0
B: REL=903
B: MSC=10

I: Bus=0003 Vendor=2f24 Product=0135 Version=0110
N: Name="  Mouse for Windows"
P: Phys=usb-0000:63:00.3-3/input1
S: Sysfs=/devices/pci0000:00/0000:00:08.1/0000:63:00.3/usb1/1-3/1-3:1.1/0003:2F24:0135.0002/input/input5
U: Uniq=
H: Handlers=sysrq kbd leds event5
B: PROP=0
B: EV=120013
B: KEY=1000000000007 ff9f207ac14057ff febeffdfffefffff fffffffffffffffe
B: MSC=10
B: LED=7

I: Bus=0010 Vendor=001f Product=0001 Version=0100
N: Name="PC Speaker"
P: Phys=isa0061/input0
S: Sysfs=/devices/platform/pcspkr/input/input6
U: Uniq=
H: Handlers=kbd event6
B: PROP=0
B: EV=40001
B: SND=6

I: Bus=0018 Vendor=222a Product=fff1 Version=0100
N: Name="ILTP7807:00 222A:FFF1"
P: Phys=i2c-ILTP7807:00
S: Sysfs=/devices/platform/AMDI0010:01/i2c-1/i2c-ILTP7807:00/0018:222A:FFF1.0005/input/input9
U: Uniq=
H: Handlers=event7 mouse1
B: PROP=2
B: EV=b
B: KEY=400 0 0 0 0 0
B: ABS=260800000000003

I: Bus=0018 Vendor=222a Product=fff1 Version=0100
N: Name="ILTP7807:00 222A:FFF1 Keyboard"
P: Phys=i2c-ILTP7807:00
S: Sysfs=/devices/platform/AMDI0010:01/i2c-1/i2c-ILTP7807:00/0018:222A:FFF1.0005/input/input11
U: Uniq=
H: Handlers=kbd event8
B: PROP=0
B: EV=100013
B: KEY=2000000000000000 0
B: MSC=10

I: Bus=0018 Vendor=093a Product=0255 Version=0100
N: Name="PNP0C50:00 093A:0255 Mouse"
P: Phys=i2c-PNP0C50:00
S: Sysfs=/devices/platform/AMDI0010:00/i2c-0/i2c-PNP0C50:00/0018:093A:0255.0004/input/input12
U: Uniq=
H: Handlers=event9 mouse2
B: PROP=0
B: EV=17
B: KEY=30000 0 0 0 0
B: REL=1943
B: MSC=10

I: Bus=0018 Vendor=093a Product=0255 Version=0100
N: Name="PNP0C50:00 093A:0255 Touchpad"
P: Phys=i2c-PNP0C50:00
S: Sysfs=/devices/platform/AMDI0010:00/i2c-0/i2c-PNP0C50:00/0018:093A:0255.0004/input/input13
U: Uniq=
H: Handlers=event10 mouse3
B: PROP=5
B: EV=1b
B: KEY=e520 30000 0 0 0 0
B: ABS=2e0800000000003
B: MSC=20

I: Bus=0000 Vendor=0000 Product=0000 Version=0000
N: Name="HD-Audio Generic HDMI/DP,pcm=3"
P: Phys=ALSA
S: Sysfs=/devices/pci0000:00/0000:00:08.1/0000:63:00.1/sound/card0/input14
U: Uniq=
H: Handlers=event11
B: PROP=0
B: EV=21
B: SW=140

I: Bus=0000 Vendor=0000 Product=0000 Version=0000
N: Name="HD-Audio Generic HDMI/DP,pcm=7"
P: Phys=ALSA
S: Sysfs=/devices/pci0000:00/0000:00:08.1/0000:63:00.1/sound/card0/input15
U: Uniq=
H: Handlers=event12
B: PROP=0
B: EV=21
B: SW=140

I: Bus=0000 Vendor=0000 Product=0000 Version=0000
N: Name="HD-Audio Generic HDMI/DP,pcm=8"
P: Phys=ALSA
S: Sysfs=/devices/pci0000:00/0000:00:08.1/0000:63:00.1/sound/card0/input16
U: Uniq=
H: Handlers=event13
B: PROP=0
B: EV=21
B: SW=140

I: Bus=0000 Vendor=0000 Product=0000 Version=0000
N: Name="HD-Audio Generic HDMI/DP,pcm=9"
P: Phys=ALSA
S: Sysfs=/devices/pci0000:00/0000:00:08.1/0000:63:00.1/sound/card0/input17
U: Uniq=
H: Handlers=event14
B: PROP=0
B: EV=21
B: SW=140

I: Bus=0000 Vendor=0000 Product=0000 Version=0000
N: Name="HD-Audio Generic Mic"
P: Phys=ALSA
S: Sysfs=/devices/pci0000:00/0000:00:08.1/0000:63:00.6/sound/card1/input18
U: Uniq=
H: Handlers=event15
B: PROP=0
B: EV=21
B: SW=10

I: Bus=0000 Vendor=0000 Product=0000 Version=0000
N: Name="HD-Audio Generic Headphone"
P: Phys=ALSA
S: Sysfs=/devices/pci0000:00/0000:00:08.1/0000:63:00.6/sound/card1/input19
U: Uniq=
H: Handlers=event16
B: PROP=0
B: EV=21
B: SW=4

I: Bus=0003 Vendor=0fac Product=0ade Version=0000
N: Name="keyd virtual keyboard"
P: Phys=
S: Sysfs=/devices/virtual/input/input20
U: Uniq=
H: Handlers=sysrq kbd event17 rfkill
B: PROP=0
B: EV=100003
B: KEY=10000000000000 0 7fffffffffffff0f fffbffffffffffff ffffffffffffffff fffffffffffffffe

I: Bus=0003 Vendor=0fac Product=1ade Version=0000
N: Name="keyd virtual pointer"
P: Phys=
S: Sysfs=/devices/virtual/input/input21
U: Uniq=
H: Handlers=event18 mouse4 js0
B: PROP=0
B: EV=f
B: KEY=ff0000 0 0 0 0
B: REL=147
B: ABS=3
```

`evtest` shows the generic XBox360 controller is `event17`
`evtest` shows L4/R4 are on `event5` named `Mouse for Windows`:
output of pressing those buttons:
```
Event: time 1704674696.978778, type 4 (EV_MSC), code 4 (MSC_SCAN), value 70046
Event: time 1704674696.978778, type 1 (EV_KEY), code 99 (KEY_SYSRQ), value 1
Event: time 1704674696.978778, -------------- SYN_REPORT ------------
Event: time 1704674697.233251, type 1 (EV_KEY), code 99 (KEY_SYSRQ), value 2
Event: time 1704674697.233251, -------------- SYN_REPORT ------------
Event: time 1704674697.267389, type 1 (EV_KEY), code 99 (KEY_SYSRQ), value 2
Event: time 1704674697.267389, -------------- SYN_REPORT ------------
Event: time 1704674697.275329, type 4 (EV_MSC), code 4 (MSC_SCAN), value 70046
Event: time 1704674697.275329, type 1 (EV_KEY), code 99 (KEY_SYSRQ), value 0
Event: time 1704674697.275329, -------------- SYN_REPORT ------------
Event: time 1704674698.502985, type 4 (EV_MSC), code 4 (MSC_SCAN), value 70048
Event: time 1704674698.502985, type 1 (EV_KEY), code 119 (KEY_PAUSE), value 1
Event: time 1704674698.502985, -------------- SYN_REPORT ------------
Event: time 1704674698.753393, type 1 (EV_KEY), code 119 (KEY_PAUSE), value 2
Event: time 1704674698.753393, -------------- SYN_REPORT ------------
Event: time 1704674698.787365, type 1 (EV_KEY), code 119 (KEY_PAUSE), value 2
Event: time 1704674698.787365, -------------- SYN_REPORT ------------
Event: time 1704674698.798621, type 4 (EV_MSC), code 4 (MSC_SCAN), value 70048
Event: time 1704674698.798621, type 1 (EV_KEY), code 119 (KEY_PAUSE), value 0
Event: time 1704674698.798621, -------------- SYN_REPORT ------------
```

contents of `dmi_data` after running `for f in /sys/class/dmi/id/*; do echo "$f:"; sudo cat $f; done > dmi_data`:
```
/sys/class/dmi/id/bios_date:
12/02/2023
/sys/class/dmi/id/bios_release:
5.29
/sys/class/dmi/id/bios_vendor:
American Megatrends International, LLC.
/sys/class/dmi/id/bios_version:
2.18
/sys/class/dmi/id/board_asset_tag:
Default string
/sys/class/dmi/id/board_name:
G1617-01
/sys/class/dmi/id/board_serial:
Default string
/sys/class/dmi/id/board_vendor:
GPD
/sys/class/dmi/id/board_version:
Default string
/sys/class/dmi/id/chassis_asset_tag:
Default string
/sys/class/dmi/id/chassis_serial:
Default string
/sys/class/dmi/id/chassis_type:
10
/sys/class/dmi/id/chassis_vendor:
Default string
/sys/class/dmi/id/chassis_version:
Default string
/sys/class/dmi/id/ec_firmware_release:
2.13
/sys/class/dmi/id/modalias:
dmi:bvnAmericanMegatrendsInternational,LLC.:bvr2.18:bd12/02/2023:br5.29:efr2.13:svnGPD:pnG1617-01:pvrDefaultstring:rvnGPD:rnG1617-01:rvrDefaultstring:cvnDefaultstring:ct10:cvrDefaultstring:skuDefaultstring:
/sys/class/dmi/id/power:
/sys/class/dmi/id/product_family:
Default string
/sys/class/dmi/id/product_name:
G1617-01
/sys/class/dmi/id/product_serial:
Default string
/sys/class/dmi/id/product_sku:
Default string
/sys/class/dmi/id/product_uuid:
e5a47400-9044-11ee-a3e3-1a265c028737
/sys/class/dmi/id/product_version:
Default string
/sys/class/dmi/id/subsystem:
/sys/class/dmi/id/sys_vendor:
GPD
/sys/class/dmi/id/uevent:
MODALIAS=dmi:bvnAmericanMegatrendsInternational,LLC.:bvr2.18:bd12/02/2023:br5.29:efr2.13:svnGPD:pnG1617-01:pvrDefaultstring:rvnGPD:rnG1617-01:rvrDefaultstring:cvnDefaultstring:ct10:cvrDefaultstring:skuDefaultstring:
```

pastaq, 2024-01-10:
> Please add the GPD WIN Mini's controller.
> 
> Information about the device: output of `cat /proc/bus/input/devices`:
> 
> ```
> I: Bus=0019 Vendor=0000 Product=0001 Version=0000
> N: Name="Power Button"
> P: Phys=PNP0C0C/button/input0
> S: Sysfs=/devices/LNXSYSTM:00/LNXSYBUS:00/PNP0C0C:00/input/input0
> U: Uniq=
> H: Handlers=kbd event0
> B: PROP=0
> B: EV=3
> B: KEY=10000000000000 0
> 
> I: Bus=0019 Vendor=0000 Product=0005 Version=0000
> N: Name="Lid Switch"
> P: Phys=PNP0C0D/button/input0
> S: Sysfs=/devices/LNXSYSTM:00/LNXSYBUS:00/PNP0C0D:00/input/input1
> U: Uniq=
> H: Handlers=event1
> B: PROP=0
> B: EV=21
> B: SW=1
> 
> I: Bus=0019 Vendor=0000 Product=0006 Version=0000
> N: Name="Video Bus"
> P: Phys=LNXVIDEO/video/input0
> S: Sysfs=/devices/LNXSYSTM:00/LNXSYBUS:00/PNP0A08:00/device:24/LNXVIDEO:00/input/input2
> U: Uniq=
> H: Handlers=kbd event2
> B: PROP=0
> B: EV=3
> B: KEY=3e000b00000000 0 0 0
> 
> I: Bus=0011 Vendor=0001 Product=0001 Version=ab83
> N: Name="AT Translated Set 2 keyboard"
> P: Phys=isa0060/serio0/input0
> S: Sysfs=/devices/platform/i8042/serio0/input/input3
> U: Uniq=
> H: Handlers=sysrq kbd leds event3
> B: PROP=0
> B: EV=120013
> B: KEY=402000000 3803078f800d001 feffffdfffefffff fffffffffffffffe
> B: MSC=10
> B: LED=7
> 
> I: Bus=0003 Vendor=2f24 Product=0135 Version=0110
> N: Name="  Mouse for Windows"
> P: Phys=usb-0000:63:00.3-3/input0
> S: Sysfs=/devices/pci0000:00/0000:00:08.1/0000:63:00.3/usb1/1-3/1-3:1.0/0003:2F24:0135.0001/input/input4
> U: Uniq=
> H: Handlers=event4 mouse0
> B: PROP=0
> B: EV=17
> B: KEY=1f0000 0 0 0 0
> B: REL=903
> B: MSC=10
> 
> I: Bus=0003 Vendor=2f24 Product=0135 Version=0110
> N: Name="  Mouse for Windows"
> P: Phys=usb-0000:63:00.3-3/input1
> S: Sysfs=/devices/pci0000:00/0000:00:08.1/0000:63:00.3/usb1/1-3/1-3:1.1/0003:2F24:0135.0002/input/input5
> U: Uniq=
> H: Handlers=sysrq kbd leds event5
> B: PROP=0
> B: EV=120013
> B: KEY=1000000000007 ff9f207ac14057ff febeffdfffefffff fffffffffffffffe
> B: MSC=10
> B: LED=7
[... 198 lines cut ...]
> /sys/class/dmi/id/modalias:
> dmi:bvnAmericanMegatrendsInternational,LLC.:bvr2.18:bd12/02/2023:br5.29:efr2.13:svnGPD:pnG1617-01:pvrDefaultstring:rvnGPD:rnG1617-01:rvrDefaultstring:cvnDefaultstring:ct10:cvrDefaultstring:skuDefaultstring:
> /sys/class/dmi/id/power:
> /sys/class/dmi/id/product_family:
> Default string
> /sys/class/dmi/id/product_name:
> G1617-01
> /sys/class/dmi/id/product_serial:
> Default string
> /sys/class/dmi/id/product_sku:
> Default string
> /sys/class/dmi/id/product_uuid:
> e5a47400-9044-11ee-a3e3-1a265c028737
> /sys/class/dmi/id/product_version:
> Default string
> /sys/class/dmi/id/subsystem:
> /sys/class/dmi/id/sys_vendor:
> GPD
> /sys/class/dmi/id/uevent:
> MODALIAS=dmi:bvnAmericanMegatrendsInternational,LLC.:bvr2.18:bd12/02/2023:br5.29:efr2.13:svnGPD:pnG1617-01:pvrDefaultstring:rvnGPD:rnG1617-01:rvrDefaultstring:cvnDefaultstring:ct10:cvrDefaultstring:skuDefaultstring:
> ```
> 
> If I forgot some information, please let me know

The input_devices list doesn't appear to contain the xb360 controller. were you in mouse mode or running some other input emulation when you captured this?

tcdude, 2024-01-10:
sorry.. very likely that either keyd and/or mouse mode interfered with that grab. here should be the correct one:
```
I: Bus=0019 Vendor=0000 Product=0001 Version=0000
N: Name="Power Button"
P: Phys=PNP0C0C/button/input0
S: Sysfs=/devices/LNXSYSTM:00/LNXSYBUS:00/PNP0C0C:00/input/input0
U: Uniq=
H: Handlers=kbd event0
B: PROP=0
B: EV=3
B: KEY=10000000000000 0

I: Bus=0019 Vendor=0000 Product=0005 Version=0000
N: Name="Lid Switch"
P: Phys=PNP0C0D/button/input0
S: Sysfs=/devices/LNXSYSTM:00/LNXSYBUS:00/PNP0C0D:00/input/input1
U: Uniq=
H: Handlers=event1
B: PROP=0
B: EV=21
B: SW=1

I: Bus=0019 Vendor=0000 Product=0006 Version=0000
N: Name="Video Bus"
P: Phys=LNXVIDEO/video/input0
S: Sysfs=/devices/LNXSYSTM:00/LNXSYBUS:00/PNP0A08:00/device:24/LNXVIDEO:00/input/input2
U: Uniq=
H: Handlers=kbd event2
B: PROP=0
B: EV=3
B: KEY=3e000b00000000 0 0 0

I: Bus=0011 Vendor=0001 Product=0001 Version=ab83
N: Name="AT Translated Set 2 keyboard"
P: Phys=isa0060/serio0/input0
S: Sysfs=/devices/platform/i8042/serio0/input/input3
U: Uniq=
H: Handlers=sysrq kbd leds event3
B: PROP=0
B: EV=120013
B: KEY=402000000 3803078f800d001 feffffdfffefffff fffffffffffffffe
B: MSC=10
B: LED=7

I: Bus=0003 Vendor=2f24 Product=0135 Version=0110
N: Name="  Mouse for Windows"
P: Phys=usb-0000:63:00.3-3/input0
S: Sysfs=/devices/pci0000:00/0000:00:08.1/0000:63:00.3/usb1/1-3/1-3:1.0/0003:2F24:0135.0001/input/input4
U: Uniq=
H: Handlers=event4 mouse0
B: PROP=0
B: EV=17
B: KEY=1f0000 0 0 0 0
B: REL=903
B: MSC=10

I: Bus=0003 Vendor=2f24 Product=0135 Version=0110
N: Name="  Mouse for Windows"
P: Phys=usb-0000:63:00.3-3/input1
S: Sysfs=/devices/pci0000:00/0000:00:08.1/0000:63:00.3/usb1/1-3/1-3:1.1/0003:2F24:0135.0002/input/input5
U: Uniq=
H: Handlers=sysrq kbd leds event5
B: PROP=0
B: EV=120013
B: KEY=1000000000007 ff9f207ac14057ff febeffdfffefffff fffffffffffffffe
B: MSC=10
B: LED=7

I: Bus=0010 Vendor=001f Product=0001 Version=0100
N: Name="PC Speaker"
P: Phys=isa0061/input0
S: Sysfs=/devices/platform/pcspkr/input/input6
U: Uniq=
H: Handlers=kbd event6
B: PROP=0
B: EV=40001
B: SND=6

I: Bus=0018 Vendor=222a Product=fff1 Version=0100
N: Name="ILTP7807:00 222A:FFF1"
P: Phys=i2c-ILTP7807:00
S: Sysfs=/devices/platform/AMDI0010:01/i2c-1/i2c-ILTP7807:00/0018:222A:FFF1.0005/input/input9
U: Uniq=
H: Handlers=event7 mouse1
B: PROP=2
B: EV=b
B: KEY=400 0 0 0 0 0
B: ABS=260800000000003

I: Bus=0018 Vendor=222a Product=fff1 Version=0100
N: Name="ILTP7807:00 222A:FFF1 Keyboard"
P: Phys=i2c-ILTP7807:00
S: Sysfs=/devices/platform/AMDI0010:01/i2c-1/i2c-ILTP7807:00/0018:222A:FFF1.0005/input/input11
U: Uniq=
H: Handlers=kbd event8
B: PROP=0
B: EV=100013
B: KEY=2000000000000000 0
B: MSC=10

I: Bus=0018 Vendor=093a Product=0255 Version=0100
N: Name="PNP0C50:00 093A:0255 Mouse"
P: Phys=i2c-PNP0C50:00
S: Sysfs=/devices/platform/AMDI0010:00/i2c-0/i2c-PNP0C50:00/0018:093A:0255.0004/input/input12
U: Uniq=
H: Handlers=event9 mouse2
B: PROP=0
B: EV=17
B: KEY=30000 0 0 0 0
B: REL=1943
B: MSC=10

I: Bus=0018 Vendor=093a Product=0255 Version=0100
N: Name="PNP0C50:00 093A:0255 Touchpad"
P: Phys=i2c-PNP0C50:00
S: Sysfs=/devices/platform/AMDI0010:00/i2c-0/i2c-PNP0C50:00/0018:093A:0255.0004/input/input13
U: Uniq=
H: Handlers=event10 mouse3
B: PROP=5
B: EV=1b
B: KEY=e520 30000 0 0 0 0
B: ABS=2e0800000000003
B: MSC=20

I: Bus=0000 Vendor=0000 Product=0000 Version=0000
N: Name="HD-Audio Generic HDMI/DP,pcm=3"
P: Phys=ALSA
S: Sysfs=/devices/pci0000:00/0000:00:08.1/0000:63:00.1/sound/card0/input14
U: Uniq=
H: Handlers=event11
B: PROP=0
B: EV=21
B: SW=140

I: Bus=0000 Vendor=0000 Product=0000 Version=0000
N: Name="HD-Audio Generic HDMI/DP,pcm=7"
P: Phys=ALSA
S: Sysfs=/devices/pci0000:00/0000:00:08.1/0000:63:00.1/sound/card0/input15
U: Uniq=
H: Handlers=event12
B: PROP=0
B: EV=21
B: SW=140

I: Bus=0000 Vendor=0000 Product=0000 Version=0000
N: Name="HD-Audio Generic HDMI/DP,pcm=8"
P: Phys=ALSA
S: Sysfs=/devices/pci0000:00/0000:00:08.1/0000:63:00.1/sound/card0/input16
U: Uniq=
H: Handlers=event13
B: PROP=0
B: EV=21
B: SW=140

I: Bus=0000 Vendor=0000 Product=0000 Version=0000
N: Name="HD-Audio Generic HDMI/DP,pcm=9"
P: Phys=ALSA
S: Sysfs=/devices/pci0000:00/0000:00:08.1/0000:63:00.1/sound/card0/input17
U: Uniq=
H: Handlers=event14
B: PROP=0
B: EV=21
B: SW=140

I: Bus=0000 Vendor=0000 Product=0000 Version=0000
N: Name="HD-Audio Generic Mic"
P: Phys=ALSA
S: Sysfs=/devices/pci0000:00/0000:00:08.1/0000:63:00.6/sound/card1/input18
U: Uniq=
H: Handlers=event15
B: PROP=0
B: EV=21
B: SW=10

I: Bus=0000 Vendor=0000 Product=0000 Version=0000
N: Name="HD-Audio Generic Headphone"
P: Phys=ALSA
S: Sysfs=/devices/pci0000:00/0000:00:08.1/0000:63:00.6/sound/card1/input19
U: Uniq=
H: Handlers=event16
B: PROP=0
B: EV=21
B: SW=4

I: Bus=0003 Vendor=045e Product=028e Version=0101
N: Name="Microsoft X-Box 360 pad"
P: Phys=usb-0000:63:00.3-5/input0
S: Sysfs=/devices/pci0000:00/0000:00:08.1/0000:63:00.3/usb1/1-5/1-5:1.0/input/input22
U: Uniq=
H: Handlers=event19 js1
B: PROP=0
B: EV=20000b
B: KEY=7cdb000000000000 0 0 0 0
B: ABS=3003f
B: FF=107030000 0
```

pastaq, 2024-01-10:
When you get the chance please checkout this branch and install from source to test. 

https://github.com/ShadowBlip/HandyGCCS/tree/win_mini

tcdude, 2024-01-10:
So I cloned, checked out `win_mini` and installed that. I started gamescope-session-plus and played around with it. I see the Xbox 360 Controller that works as expected and L4 opens QAM and R4 takes a screenshot..

Not sure what else I should be testing to confirm that everything works as expected. LMK what (if any) other tests I should run

pastaq, 2024-01-11:
That's all that is expected. Thanks for helping out with this.

closed 2024-01-11 by pastaq